moveit/moveit #2668: Is there an standard mechanism to build a collision object from an URDF?

rafaelrojasmiliani, 2021-05-16:
**Problem**
I would like to automatically load collision objects described by their URDF into a planning scene. These URDF have several links attached by fixed joints and I would like to publish their TF. I would like that the position of these frames are automatically updated when objects are attached to a robot.  This problem brings me to the following questions

1. Is there a standard way to convert URDF (which only contains fixed joints) to a collision object message?
2. Is there a standard mechanism to publish the frames of the collision objects? 

**Alternatives I have considered**
To solve 1 I'm currently implementing an URDF parser which takes the links and fixed joints from an URDF and build a collision_object message. There is a thread [here](https://answers.ros.org/question/281826/add-collison-object-from-a-urdf-file-in-moveit/), but I did not found more information about how to implement it.

To publish the object frames I would use a robot_state publisher for each piece and program a node that listen to the planning scene topic/service and publish the  base frame of each piece or all the frames of the collision object primitives. 

**Additional context**
I'm building an automatic assembly system where different components are defined as URDF with several links. Some of these links are sockets, and the assembly of two components is done by bringing their sockets frames to the same pose. I would like to update the positions of the pieces taken by the robot with moveit by attaching the collision objects to the robot.

v4hn, 2021-05-16:
Hi there,

I have some thought on this, but there's no silver bullet (as of yet).

- There is currently some discussion among maintainers on a better PlanningScene implementation.
  That one would likely also support collision objects descried in URDF, but it's only discussion up to now

- If you have a `RobotState` instance for a URDF (you might generate one via `RobotModelLoader`), there are [methods for creating `Marker` from a RobotState](https://github.com/ros-planning/moveit/blob/master/moveit_core/robot_state/include/moveit/robot_state/robot_state.h#L1743-L1767).
  The format is quite similar for `CollisionObject`, so we might as well add a method for that.

- You can get transforms from the current PlanningScene published via the `tf_publisher_capability`, but I'm not sure that's what you want to do?

- There is always the obvious fallback to include the object *as part of your `robot_description`*. If you attach it to the base frame via a FloatingJoint, you can(and have to) update its position via TF.

> I'm currently implementing an URDF parser which takes the links and fixed joints from an URDF and build a collision_object message.

Sounds useful in general! I would prefer a method in RobotState though because that could account for articulated joints as well.

rafaelrojasmiliani, 2021-05-16:
Thank you @v4hn for your comment. I also thought in using the `RobotModelLoader` to create a robot model from an URDF. Then create a planning scene with the object that I want to displace and some how add this planning scene to the planning scene of the robot. However, I believe that this method will require to create also a SRDF and other stuff created by the configuration manager... or not?

I didn't know the `tf_publisher_capability`, its new in Noetic!. I would try to port it to Melodic. As I can see [here](https://github.com/ros-planning/moveit/blob/47af01bac8b5c37e2af87e4f1bd580ed8378535d/moveit_ros/move_group/src/default_capabilities/tf_publisher_capability.cpp#L91) it publishes the pose of the first shape primitive, this would be perfect for the solution using the robot states publisher.

Do you have a link to the discussion about the new PlanningScene class?

felixvd, 2021-05-16:
The discussion is in https://github.com/ros-planning/moveit/issues/202

A URDF->CollisionObject converter would be very useful. I currently store CollisionObjects as YAML files myself.

Also, what you're doing for the assembly frames sounds a lot like what we did for the World Robot Summit 2018 Assembly Challenge [here](https://github.com/o2as/ur-o2as). You will be interested in [this section](https://github.com/o2as/ur-o2as/tree/master/catkin_ws/src/o2as_parts_description/scripts) where we create the assembly tree.

> Is there a standard mechanism to publish the frames of the collision objects?

I'm not sure what frames you refer to exactly, but the `tf_publisher_capability` is probably (part of) what you were looking for. Note that [subframes](https://github.com/ros-planning/moveit/pull/1439) were intended for use in this context, and you can use them in planning requests. Also take note of [this PR](https://github.com/ros-planning/moveit/pull/2037) which should make the subframes definition a lot easier.

v4hn, 2021-05-17:
> I also thought in using the `RobotModelLoader` to create a robot model from an URDF.
> Then create a planning scene with the object that I want to displace and somehow add this planning scene to the planning scene of the robot.

That's not what I proposed. Currently a `PlanningScene` can only ever contain *one* RobotState. So it's impossible to merge scene objects.
Instead, one thing you can probably do without too much new code is create a `RobotState` object from the URDF, which describes your collision object, and extract a `CollisionObject` representation from it. Afterwards you could just add the generated set of CollisionObjects to the `PlanningScene` of your real robot. As I said we already have a method to generate Markers. A similar method for CollisionObjects could probably mimic parts of it.

rafaelrojasmiliani, 2021-07-07:
@v4hn , Thank you. I did not understand very well the Planning Scene.

I have come with a first version of a urdf to collision object converter. My solution is a single function which takes as an input a reference to `urdf::Model` and returns a `moveit_msgs::CollisionObject`. This first version in in my personal [moveit tutorial] (https://github.com/rafaelrojasmiliani/moveit_tutorial/blob/noetic/09_manipulation/src/pick_place/c_pick_and_place/src/urdf_to_collision_object.cpp). I would like to do a pull request, but I don't know where. I would propose to place this function somewhere in `moveit/moveit_core/collision_detection/src/`. Can some one indicate a where this should be?

This converter set the relative position between the collision objects. However, I have understood that the actual moveit implementation  interprest `CollisionObjec::primitive_poses` as the positions with respect to the header's frame. To fix this, I implemented the function `update_frames` available in the same file with the link above. I have understood that [this pull requeest](https://github.com/ros-planning/moveit/pull/2037) will make that `update_frames` function unnecessary. 

@felixvd , Thank you very much for your message. The project is very interesting. What do you mean with "assembly tree"? This question may be off-topic for this issue, I would be possible to contact you via another channel?

v4hn, 2021-07-08:
@rafaelrojasmiliani It sounds like this might have significant overlap with @captain-yoshi's new [scene_parser](https://github.com/captain-yoshi/scene_parser) that he works on as part of his GSoC project to define benchmark scenes.
I'm unsure how big the overlap is because you have very different use cases in mind.

In general, I could imagine a method in the `PlanningScene` class to load a URDF into the scene or (in your case) add another field `urdf` to the `CollisionObject` message. A more general alternative might be to use a `RobotModelLoader` to build a `RobotModel`/`State` from a URDF and have a method in the `PlanningScene` to add a representation of that robot (might be all fixed joints) to the scene as `CollisionObject`s. One issue with adding a field to the message would be that, in contrast to all other fields, this field would only ever be read because the PlanningScene would split up the URDF to generate individual collision shapes instead.