moveit/moveit #2669: Wrong namespace for /get_planning_scene service

gecastro, 2021-05-18:
### Description

I'm having issues accessing the /get_planning_scene service from a nodelet (It is tested and works properly as ROS node).

Given that I'm using the planning scene interface from a nodelet which is part of the camera group, I see the following message:

[ INFO] [1621315389.645441396, 0.780000000]: waitForService: Service [/camera/get_planning_scene] has not been advertised, waiting...

Although the service is declared at the global level /get_planning_scene

I tried to pass the namespace to the interface, but it doesn't have any effect with "/":

moveit::planning_interface::PlanningSceneInterface planning_scene_interface("/")

Any ideas?

Would it be advisable to run whole moveit within a group, such as "/robot1"?

NOTE: I did create the moveit_config package with the setup assistant, so the namespaces for moveit are the default values.

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04
* Binary build 1.0.7-1bionic.20210414.235849

### Steps to reproduce
Using the moveit::planning_interface::PlanningSceneInterfac withing a nodelet

### Expected behaviour
It should look for /get_planning_scene

### Actual behaviour
It is looking for /camera/get_planning_scene

### Backtrace or Console output

waitForService: Service [/camera/get_planning_scene] has not been advertised, waiting...

v4hn, 2021-05-18:
Hi Germán,

this is an issue in the core ROS lib [we discussed before](https://github.com/ros-planning/moveit/issues/2364) and there are two workarounds.
As you could not find this issue anywhere when you looked for information, please propose a place where we can add it more prominently, so that you would have found it.

gecastro, 2021-05-18:
Thanks for your answer Michael.

It's probably fine. I didn't make the association that the problem would be more general at the level of the NodeHandle.

closed 2021-05-18 by v4hn