moveit/moveit #2675: Support FollowCartesianTrajectory

gautz, 2021-05-19:
**Is your feature request related to a problem? Please describe.**
Now that UR support FollowCartesianTrajectory, is there any plan to integrate it in MoveIt? Or will it only be implemented in MoveIt2 ? https://github.com/fzi-forschungszentrum-informatik/cartesian_ros_control/issues

**Describe the solution you'd like**
Basically a planning pipeline working entirely in Cartesian space.

v4hn, 2021-05-19:
Who does collision checking (and self-collision checking) with this action?

As far as I can see [the action interface](https://github.com/fzi-forschungszentrum-informatik/cartesian_ros_control/blob/master/cartesian_control_msgs/action/FollowCartesianTrajectory.action) does not pass the intended joint trajectory back to MoveIt, so MoveIt's whole scene representation cannot be meaningfully used in this action.

I'm not sure what you would expect MoveIt-based interface for this action to be, but I could imagine an action server supporting this interface on top of moveit_servo. But of course that interface would not at all look at action servers offered by robot drivers and it sounds like that is what you would want?

gautz, 2021-05-20:
I'm lacking a deeper understanding of the whole planning pipeline but here are the elements that I identified:
- OMPL supports planning in Cartesian space, in particular Cartesian Constrained planning https://github.com/ros-planning/moveit/pull/2273 https://github.com/ros-planning/moveit2/pull/377 
- MoveIt by defaults plans in Cartesian space when Cartesian constraints are given
- Industrial manufacturing processes are usually defined in Cartesian space, and Industrial robot controllers provide such interface. So does UR now too.
- But as far as I understand only a JointSpaceTrajectory can be output from the MoveIt planning pipeline.

The action interface can implicitly contain the joint trajectory through the posture definition: https://github.com/fzi-forschungszentrum-informatik/cartesian_ros_control/blob/master/cartesian_control_msgs/msg/CartesianPosture.msg

v4hn, 2021-05-20:
Don't misunderstand, I do see the use-case for `FollowCartesianTrajectory`. It's just that a useful interface with MoveIt would look very different from what you seem to be interested in. I.e., MoveIt would provide a *server* for it but there is not much sense in it acting as a client.
@tylerjw and @AndyZe might be interested to add such an action server interface to moveit_servo.

> - OMPL supports planning in Cartesian space, in particular Cartesian Constrained planning

Yes, but the whole point of this kind of planning is to generate a *JointTrajectory* solution that fulfills the Cartesian constraints.
For example it can generate a joint trajectory that will stay on a Cartesian line, but might go back and forth on it to pass through a joint singularity and reach further in the end.
I expect this is not what you would want from `FollowCartesianTrajectory`.

> - MoveIt by defaults plans in Cartesian space when Cartesian constraints are given

Yes and no. It will *sample* in Cartesian space but will still project samples to joint space via IK requests. If we only knew about the Cartesian pose of a link we could not do collision checking for the whole robot (the poses of all other joints are missing).

> - Industrial manufacturing processes are usually defined in Cartesian space, and Industrial robot controllers provide such interface. So does UR now too.

Yes. This is a standard interface robot users need. Most robot drivers also integrate some kind of self-collision checking in such calls to make sure they will not hit other parts of the same robot. It has intrinsic shortcomings though if you work with more than one robot component because separate components will usually not know about each other unless the companies collaborate and explicitly provide support.
So the driver-supported Cartesian controller for a UR arm can't possibly account for self-collisions between the arm and an attached custom gripper unless you provide the whole collision geometry to the driver. Also there is an incentive not to do such generic collision checking because it's much harder to guarantee cycle times if you need to check arbitrary collision geometries. This is a big part of what MoveIt addresses.

> - But as far as I understand only a JointSpaceTrajectory can be output from the MoveIt planning pipeline.

Yes, of course this implicitly provides a Cartesian trajectory as well.

> The action interface can implicitly contain the joint trajectory through the posture definition: https://github.com/fzi-forschungszentrum-informatik/cartesian_ros_control/blob/master/cartesian_control_msgs/msg/CartesianPosture.msg

That looks useful! So you could generate a joint trajectory and fill in the whole joint trajectory in these fields.
Then you could *additionally* provide Cartesian limits/tolerances/allowed interaction forces for each waypoint and send it.
You get a collision-free trajectory the controller can execute with Cartesian specifications.
Such an approach would basically only require [a new controller interface](https://github.com/ros-planning/moveit/tree/master/moveit_plugins/moveit_simple_controller_manager/include/moveit_simple_controller_manager) and is quite orthogonal to planning.

Coming up with non-zero Cartesian tolerances that guarantee collision-free execution is non-trivial, especially close to singularities.
One other aspect that bothers me is that the `CartesianPosture` does not seem to specify how much the controller can deviate from the specified joint posture, so what enforces that the Cartesian controller will actually pick the correct joint angle for a redundant joint when leaving a singularity?

gautz, 2021-05-20:
No misunderstanding, thanks for the feedback. I am also aware of the many steps that need to happen in the configuration space. Trying to reformulate my concern.
What would be the most efficient way to get a CartesianTrajectory.msg out of MoveIt? I see two options:
- Write some kind of parser (a planner adapter?) that re-generates the Cartesian info by applying FK on the JointTrajectory.
- Propagate the Cartesian info within MoveIt from the planned solution (e.g. OMPL output) down to the output Trajectory message by e.g. adding a moveit_msgs/CartesianTrajectory[] field in the moveit_msgs MotionPlanResponse.msg (similar as in moveit_msgs MotionPlanRequest.msg)

Maybe then it would be the occasion to upgrade the moveit_msgs/CartesianTrajectory[] to fit the proposed https://github.com/fzi-forschungszentrum-informatik/cartesian_ros_control/blob/beta-testing/cartesian_control_msgs/msg/CartesianTrajectory.msg
which is also used by SWRI https://github.com/marrts/cartesian_trajectory_msgs/blob/master/msg/CartesianTrajectory.msg
@marrts @Levi-Armstrong 

> Such an approach would basically only require [a new controller [...]

So you mean some planners not only give you a collision-free path but also collision-free tolerances around this path in joint space, good to know :). Then maybe the CartesianTrajectory.msg should include an additional field to describe collision-free joint position deviations, e.g.: ``float64[] posture_joint_tolerance`` which should be close to zero close to singularity.

I kind of get your last points about control, but I think I would leave that part to @stefanscherzinger

v4hn, 2021-05-20:
> What would be the most efficient way to get a CartesianTrajectory.msg out of MoveIt? I see two options:
> - Write some kind of parser (a planner adapter?) that re-generates the Cartesian info by applying FK on the JointTrajectory.

As `FollowCartesianTrajectory` is supposed to be a direct alternative to `FollowJointTrajectory`, I would implement this approach in a controller interface as mentioned in my earlier comment.

Simple methods to provide the "parser" you propose could be added to `moveit_core/trajectory_processing` and I would propose to have methods that take C++ `RobotTrajectory` objects to generate the CartesianTrajectory message.

If you generate a linear Cartesian motion, using e.g. `computeCartesianPath` or the pilz planner, the resulting joint trajectory will currently spline-interpolate in joint space through Cartesian waypoints. So if you resample the trajectory at a higher resolution the eef waypoints it will not remain linear in Cartesian space anymore. That's a shortcoming in the current system and that's what a `CartesianTrajectory` message can in theory help with.
But the same is true the other way around: If you generate a spline-based joint trajectory and you convert it to the CartesianTrajectory, sampling at a higher rate (while ignoring the exact posture information of course which is optional) it actually changes the represented trajectory.

> - Propagate the Cartesian info within MoveIt from the planned solution (e.g. OMPL output) down to the output Trajectory message by e.g. adding a moveit_msgs/CartesianTrajectory[] field in the moveit_msgs MotionPlanResponse.msg (similar as in moveit_msgs MotionPlanRequest.msg)

At the moment I'm not at all convinced it makes much sense to add it there. It would make sense to add it to `GetCartesianPath.srv` at least.
And especially w.r.t. the Cartesian constraint support in OMPL the question of how to account for all these things in MotionPlanRequest/Response is justified.

> Maybe then it would be the occasion to upgrade the moveit_msgs/CartesianTrajectory[] to fit the proposed [...]

Assuming there is a good standard message package to rely on, that clearly makes sense.
@bmagyar introduced these messages in MoveIt two years ago for seed support in CHOMP et. al.

> > Coming up with non-zero Cartesian tolerances that guarantee collision-free execution is non-trivial, especially close to singularities.
> 
> So you mean some planners not only give you a collision-free path but also collision-free tolerances around this path in joint space, good to know :).

I did not say that. I would love to have such planners, but I'm not aware of any computationally efficient way to provide such tolerances.
Basically you can compute Cartesian distances to collision for each waypoint. But there is still a big nonlinear step to go from that information to joint tolerances (and you have the same problem when trying to provide Cartesian tolerances **in a specific frame**, as the distances are w.r.t. different frames).

> Then maybe the CartesianTrajectory.msg should include an additional field to describe collision-free joint position deviations, e.g.: ``float64[] posture_joint_tolerance`` which should be close to zero close to singularity.

I'm clearly missing something like that in the action anyway because otherwise the action server has no criteria to abort the (Cartesian) trajectory execution if unexpected things happen.
But filling in these tolerances is also non-trivial if you actually want to allow for big deviations away from singularities.