moveit/moveit #2676: pilz industrial motion planner not using the updated allowed collision matrix

robertjbush, 2021-05-20:
### Description

Asked by @felixvd to raise an issue ticket ([see this ROS answers question](https://answers.ros.org/question/378602/pilz-industrial-motion-planner-not-using-the-updated-allowed-collision-matrix/))

### My environment
* ROS Noetic in a docker container with Ubuntu 20.04
* Moveit! version 1.1.2
* ros-industrial-client with motoman driver
* motoman gp8 robot
* binary build

### Problem
* Using MoveGroup commander (Python API) and Pilz Industrial Motion Planner, changes to the allowed collision matrix (ACM) are published to the planning scene but do not propagate to the planning interface.

### Action Taken So Far
* Tested with an identical stack except with OMPL planner in the pipeline. Behaviour is not reproducible, changes propagate to the planning interface (perhaps through pre/post-processing plugins?).
* Tested with planning directly in rviz. Behaviour is not reproducible, changes propagate to the planning interface.

### Steps to reproduce

My base application is a docker container and that is detailed below.
```
ARG ROS_DISTRO=noetic
ARG BASE_IMAGE=ros:${ROS_DISTRO}-ros-base
ARG UID=0
ARG GID=0
ARG HOME=/root

FROM ${BASE_IMAGE} AS base

RUN apt-get update && apt-get upgrade -y && \
  apt-get install -y --no-install-recommends \
  clang-format \
  clang-tidy \
  cppcheck \
  git-core \
  libgl1-mesa-dri \
  libgl1-mesa-glx \
  libmuparser-dev \
  libopenblas-dev \
  libxml2-utils \
  nodejs \
  npm \
  python3-pip \
  python3-psutil \
  python3-vcstool \
  python-is-python3 \
  ros-${ROS_DISTRO}-controller-interface \
  ros-${ROS_DISTRO}-controller-manager \
  ros-${ROS_DISTRO}-control-toolbox \
  ros-${ROS_DISTRO}-diagnostic-updater \
  ros-${ROS_DISTRO}-eigen-conversions \
  ros-${ROS_DISTRO}-filters \
  ros-${ROS_DISTRO}-hardware-interface \
  ros-${ROS_DISTRO}-joint-limits-interface \
  ros-${ROS_DISTRO}-joint-state-controller \
  ros-${ROS_DISTRO}-joint-state-publisher \
  ros-${ROS_DISTRO}-moveit-core \
  ros-${ROS_DISTRO}-moveit-commander \
  ros-${ROS_DISTRO}-moveit-fake-controller-manager \
  ros-${ROS_DISTRO}-moveit-planners-ompl \
  ros-${ROS_DISTRO}-moveit-ros-move-group \
  ros-${ROS_DISTRO}-moveit-ros-planning \
  ros-${ROS_DISTRO}-moveit-ros-planning-interface \
  ros-${ROS_DISTRO}-moveit-ros-visualization \
  ros-${ROS_DISTRO}-moveit-simple-controller-manager \
  ros-${ROS_DISTRO}-pilz-industrial-motion-planner \
  ros-${ROS_DISTRO}-plotjuggler \
  ros-${ROS_DISTRO}-robot-state-publisher \
  ros-${ROS_DISTRO}-ros-controllers \
[... 417 lines cut ...]

        if not self._reached_pose():
            raise MoveFailed("The robot did not reach it's target pose")

    @require_trajectory
    def _reached_pose(self) -> bool:
        target_joint_values: List[float] = cast(List[float], self._target_joint_values)

        distance = [
            abs(target - current)
            for target, current in zip(target_joint_values, self.get_joint_positions().values())
        ]

        if [i for i in distance if i >= self.JOINT_ERROR_THRESHOLD]:
            return False

        return sum(distance) / len(target_joint_values) < self.JOINT_MEAN_ERROR_THRESHOLD
```
The rest of my code is standard implementation and calls to the functions above.

### Expected behaviour
* Successful plan

### Actual behaviour
* IK solution no found error

gavanderhoorn, 2021-05-20:
Could you add the full error message, instead of paraphrasing it?

robertjbush, 2021-05-20:
Edited issue to include error message (see section _"Actual Behaviour"_)

robertjbush, 2021-05-20:
I have some extra debug messages to share. Below are two traces for identical plans. One however was initiated from the `MoveGroupCommander` and the other from `rviz` GUI. Notice that the `pipeline_id` is set from `rviz` but not `MoveGroupCommander` and the planning request is fully populated from `rviz` yet predominately empty from `MoveGroupCommander`.

**MoveGroup Commander**
```
[ INFO] [1621513504.298925721] [/fake/move_group]: Received event 'stop'
[DEBUG] [1621513504.299080820] [/fake/move_group]: The action server has received a new goal request
[DEBUG] [1621513504.299305419] [/fake/move_group]: A new goal has been recieved by the single goal action server
[DEBUG] [1621513504.299422718] [/fake/move_group]: Accepting a new goal
[DEBUG] [1621513504.299489517] [/fake/move_group]: Accepting goal, id: /fake/move_group_commander_wrappers_1621513484971512745-1-1621513504.298822822, stamp: 1621513504.30
[DEBUG] [1621513504.299549717] [/fake/move_group]: Publishing feedback for goal, id: /fake/move_group_commander_wrappers_1621513484971512745-1-1621513504.298822822, stamp: 1621513504.30
[DEBUG] [1621513504.299584417] [/fake/move_group]: Publishing feedback for goal with id: /fake/move_group_commander_wrappers_1621513484971512745-1-1621513504.298822822 and stamp: 1621513504.30
[DEBUG] [1621513504.299618716] [/fake/move_group]: Trying to publish message of type [moveit_msgs/MoveGroupActionFeedback/12232ef97486c7962f264c105aae2958] on a publisher with type [moveit_msgs/MoveGroupActionFeedback/12232ef97486c7962f264c105aae2958]
[DEBUG] [1621513504.299670216] [/fake/move_group]: sync robot state to: 4.300s
[ INFO] [1621513504.393283161] [/fake/move_group]: Planning request received for MoveGroup action. Forwarding to planning pipeline.
[DEBUG] [1621513504.393360960] [/fake/move_group]: Running 'Fix Workspace Bounds'
[DEBUG] [1621513504.393391560] [/fake/move_group]: It looks like the planning volume was not specified. Using default values.
[DEBUG] [1621513504.393415960] [/fake/move_group]: Running 'Fix Start State Bounds'
[DEBUG] [1621513504.393437060] [/fake/move_group]: Running 'Fix Start State In Collision'
[DEBUG] [1621513504.393815757] [/fake/move_group]: Start state is valid with respect to group gp8_toolchange
[DEBUG] [1621513504.393859257] [/fake/move_group]: Running 'Fix Start State Path Constraints'
[DEBUG] [1621513504.394245154] [/fake/move_group]: Path constraints are OK. Running usual motion plan.
[DEBUG] [1621513504.394376153] [/fake/move_group]: Loading PlanningContext for request
<request>
workspace_parameters:
  header:
    seq: 0
    stamp: 0.000000000
    frame_id:
  min_corner:
    x: -5
    y: -5
    z: -5
  max_corner:
    x: 5
    y: 5
    z: 5
start_state:
  joint_state:
    header:
      seq: 0
      stamp: 0.000000000
      frame_id:
    name[]
    position[]
    velocity[]
    effort[]
  multi_dof_joint_state:
    header:
      seq: 0
      stamp: 0.000000000
      frame_id:
    joint_names[]
    transforms[]
    twist[]
    wrench[]
  attached_collision_objects[]
  is_diff: 1
goal_constraints[]
  goal_constraints[0]:
    name:
    joint_constraints[]
    position_constraints[]
      position_constraints[0]:
            header:
          seq: 0
          stamp: 0.000000000
          frame_id: world
        link_name: axia80_mate
        target_point_offset:
          x: 0
[... 219 lines cut ...]
</request>
[ INFO] [1621513379.275331993] [/fake/move_group]: Initialized Point-to-Point Trajectory Generator.
[DEBUG] [1621513379.275374993] [/fake/move_group]: Found planning context loader for PTP group:gp8_toolchange
[ INFO] [1621513379.275439292] [/fake/move_group]: Generating PTP trajectory...
[DEBUG] [1621513379.275748390] [/fake/move_group]: Running 'Add Time Parameterization'
[DEBUG] [1621513379.276085388] [/fake/move_group]: Motion planner reported a solution path with 26 states
[DEBUG] [1621513379.279943861] [/fake/move_group]: Planned path was found to be valid when rechecked
[DEBUG] [1621513379.280052860] [/fake/move_group]: Trying to publish message of type [moveit_msgs/DisplayTrajectory/41936b74e168ba754279ae683ce3f121] on a publisher with type [moveit_msgs/DisplayTrajectory/41936b74e168ba754279ae683ce3f121]
[DEBUG] [1621513379.280157259] [/fake/move_group]: Setting the current goal as succeeded
[DEBUG] [1621513379.280239759] [/fake/move_group]: Setting status to succeeded on goal, id: /fake/rviz-1-1621513379.216763503, stamp: 1621513379.22
[DEBUG] [1621513379.280288258] [/fake/move_group]: Publishing result for goal with id: /fake/rviz-1-1621513379.216763503 and stamp: 1621513379.22
[DEBUG] [1621513379.280304158] [/fake/move_group]: Trying to publish message of type [moveit_msgs/MoveGroupActionResult/e8f0259d9c66fae7820b6818cc47d807] on a publisher with type [moveit_msgs/MoveGroupActionResult/e8f0259d9c66fae7820b6818cc47d807]
[DEBUG] [1621513379.280329658] [/fake/move_group]: Publishing feedback for goal, id: /fake/rviz-1-1621513379.216763503, stamp: 1621513379.22
[DEBUG] [1621513379.280338158] [/fake/move_group]: Publishing feedback for goal with id: /fake/rviz-1-1621513379.216763503 and stamp: 1621513379.22
[DEBUG] [1621513379.510386647] [/fake/move_group]: Published full planning scene: ''
[DEBUG] [1621513379.510453947] [/fake/move_group]: scene update 9.51044 robot stamp: 0
[DEBUG] [1621513379.510473747] [/fake/move_group]: Setting new planning scene: ''
[DEBUG] [1621513379.510554746] [/fake/rviz]: scene update 9.51052 robot stamp: 0
[DEBUG] [1621513379.510612046] [/fake/rviz]: Setting new planning scene: ''
[DEBUG] [1621513379.760481996] [/fake/move_group]: Published full planning scene: ''
[DEBUG] [1621513379.760583195] [/fake/move_group]: scene update 9.76057 robot stamp: 0
[DEBUG] [1621513379.760604095] [/fake/move_group]: Setting new planning scene: ''
[DEBUG] [1621513379.760648395] [/fake/rviz]: scene update 9.76063 robot stamp: 0
[DEBUG] [1621513379.760670795] [/fake/rviz]: Setting new planning scene: ''
```

robertjbush, 2021-05-26:
Now using MoveIt 1.1.4. Could the author of the new API function `get_planning_pipeline(self, planning_pipeline)`, @felixvd explain what type/ object 'planning_pipeline' is? Im still stuck with this issue and Im trying to get a better understanding of the API, but can't get the above function to return.

felixvd, 2021-05-26:
Sorry for not getting back to you. I can't test this at the moment, and it doesn't look like you posted the lines that you are actually executing. Your classes seem to make sense (although I didn't read it all), but what is the order of operations?

The `planning_pipeline_id` and `planner_id` are both strings that need to match an active planning pipeline and a planner inside that pipeline. You can check the available ones by looking at the terminal when your move_group starts up, or by checking the dropdown menu in the Rviz MotionPlanning plugin. These lines should work for ompl and the pilz planner:

```
group = moveit_commander.MoveGroupCommander("my_robot")
group.set_planning_pipeline_id("pilz_industrial_motion_planner")
group.set_planner_id("LIN")
[...]
group.set_planning_pipeline_id("ompl")
group.set_planner_id("RRTConnect")
```

If you 1) change the allowed collision matrix as shown in your code, 2) publish the change, and 3) try planning with both planners from `MoveGroupCommander`, I don't see how the plans could be using different ACMs. The Rviz MotionPlanning plugin should also use the same ACM. Please post the misbehaving lines if you can reproduce the problem.

> I have some extra debug messages to share.

This one makes me think that you may have called `set_planner_id` but not `set_planning_pipeline_id`.

> My `planning_pipeline.launch` is standard as below:

Are you starting up both the `ompl` and `pilz_industrial_motion_planner` pipelines in your move_group? It looks like your `moveit_config` package is out of date. If you regenerate it with the setup assistant, your file should look like [this](https://github.com/ros-planning/panda_moveit_config/blob/melodic-devel/launch/planning_pipeline.launch.xml), and the `move_group.launch` should contain a section like this:

```
  <!-- Planning Pipelines -->
  <group ns="move_group/planning_pipelines">

    <!-- OMPL -->
    <include ns="ompl" file="$(find o2ac_moveit_config)/launch/planning_pipeline.launch.xml">
      <arg name="pipeline" value="ompl" />
    </include>

    <!-- CHOMP -->
    <include ns="chomp" file="$(find o2ac_moveit_config)/launch/planning_pipeline.launch.xml">
      <arg name="pipeline" value="chomp" />
    </include>

    <!-- Pilz Industrial Motion-->
    <include ns="pilz_industrial_motion_planner" file="$(find moveit_resources_prbt_moveit_config)/launch/planning_pipeline.launch.xml">
      <arg name="pipeline" value="pilz_industrial_motion_planner" />
    </include>

    <!-- Support custom planning pipeline -->
    <include if="$(eval arg('pipeline') not in ['ompl', 'chomp', 'pilz_industrial_motion_planner'])"
	     file="$(find o2ac_moveit_config)/launch/planning_pipeline.launch.xml">
      <arg name="pipeline" value="$(arg pipeline)" />
    </include>
  </group>
``` 

If there was anything unclear in the tutorials, please note it now while it's hot on your mind and feel free to submit a PR.

robertjbush, 2021-05-26:
Thank you for your reply, my config is exactly as you describe above and since updating to 1.1.4 the `set_planning_pipeline_id` eradicates one of my concerns. However, I now apologise in advance for this question, as I feel I'm missing something very obvious and being a bit dumb. The command below does not return:

`group.get_planning_pipeline_id("pilz_industrial_motion_planner")`

The error is:
```
ArgumentError: Python argument types in
    MoveGroupInterface.get_planning_pipeline_id(MoveGroupInterface, str)
did not match C++ signature:
    get_planning_pipeline_id(moveit::planning_interface::MoveGroupInterfaceWrapper {lvalue})
```

I feel like I should be passing it a `planning_pipeline` **object** but I don't currently create one. Perhaps my oversight here is something so obvious to you guys that it isn't particularly documented? If so I apologise for the dumb question, but could you tell me how to create this object.

robertjbush, 2021-05-26:
I am trying to "get" the id; "set" works fine. As per the [code](https://github.com/ros-planning/moveit/blob/31f9a75f4095dec0ccfe8ba05eea5f473327d7c1/moveit_commander/src/moveit_commander/move_group.py#L540) you wrote, the docstring says to pass `get` the `planning_pipeline`.

felixvd, 2021-05-27:
My bad, I will double check/fix that soon. 

As a workaround for now, I would just set both the planning pipeline and planner to whatever you need before planning.

robertjbush, 2021-05-27:
> My bad, I will double check/fix that soon.

I'm just glad it wasn't me being dumb! 😃

> it doesn't look like you posted the lines that you are actually executing. Your classes seem to make sense (although I didn't read it all), but what is the order of operations?

Video below showing my order of operations and showing the `OMPL` vs. `Pilz` issue from `MoveGroupCommander` (using your new `set` function 🚀 ). Remember Pilz does actually work as expected when planned from `rviz` which I don't understand 😕. Would really appreciate some guidance as I'm blocked at the moment. 

https://user-images.githubusercontent.com/44445100/119825731-2dd00200-beef-11eb-8501-ff489862d2f0.mp4

felixvd, 2021-05-27:
1. Long shot, but are you sure that the OMPL plan for target B with collisions disabled is the same motion that the PTP plan should return? You can check the trajectory in the plan to see if there's more than one joint goal.

2. @jschleicher @ct2034 In my own system I've gotten spurious IK failures that often work out when I attempt the plan multiple times. Could this be related?

robertjbush, 2021-05-27:
>  You can check the trajectory in the plan to see if there's more than one joint goal.

I hope the below is what you were after?

Last point in `_trajectory` for `OMPL` target b (collisions disabled via **MGI manager**)    

- 
      positions: [1.2297643327024208, 0.8866434517297366, -0.13153589567446222, 0.3947528471556745, 1.0533737209939544, -0.20323410382787466]
      velocities: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
      accelerations: [0.07811556058867675, -0.129385950679334, -0.3886550505995608, -0.8402842448804891, 0.2427570865380844, 0.09535058874000003]
      effort: []
      time_from_start: 
        secs: 6
        nsecs: 658554343

Last point in `_trajectory` for `pilz` target b (collisions disabled via **srdf**)

- 
      positions: [1.2297754365406206, 0.8868086455407083, -0.1315005379184601, 0.39480578386032017, 1.0535808297249376, -0.20323103154208577]
      velocities: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
      accelerations: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
      effort: []
      time_from_start: 
        secs: 4
        nsecs: 781501524

robertjbush, 2021-05-27:
@felixvd I have a breakthrough of sorts from your debug suggestion! 👊 Planning to a joint position rather than pose works as expected. The command is below for reference. Of course for my application I need to supply poses, not joint positions but perhaps now this will help you guys pin point the root cause? 🤞
`jogger.moveit.plan_joint_position([1.2297754365406206, 0.8868086455407083, -0.1315005379184601, 0.39480578386032017, 1.0535808297249376, -0.20323103154208577],0.5, 0.5).`

_For reference:_ I did then change IK solver from IKFast to KDL but that doesn't make a difference

robertjbush, 2021-06-02:
@jschleicher @ct2034 would you be able to help identify the root cause? We are more than happy to contribute to a fix but we didn't write the library, so it would be far more efficient if Pilz engineers pointed us in the right direction.

robertjbush, 2021-06-07:
I made a workaround for `PTP` planner_id, which is to convert pose to joint positions under the hood before planning. Can't do this for `LIN`, as the source code checks in cartesian points, see [here](https://github.com/ros-planning/moveit/blob/bab87f9aa32df22273674fbd007d495d7c455cf1/moveit_planners/pilz_industrial_motion_planner/src/trajectory_generator_lin.cpp#L147). I believe the root cause is somewhere in the `isStateColliding` function, [see here](https://github.com/ros-planning/moveit/blob/3361b2d1b6b2feabc2d3e93c75653f5a00e87fa4/moveit_planners/pilz_industrial_motion_planner/src/trajectory_functions.cpp#L558). Please help with some guidance as to where I go from here. I'm solely using the python API.

jschleicher, 2021-06-08:
@robertjbush I'm not into the collision system and have no clue, why the ACM could not be updated. The code you point to, just uses the planning scene to check for collisions. Maybe @felixvd can give a hint, what's wrong in the usage or needs to be done to update the collsiion matrix?
https://github.com/ros-planning/moveit/blob/3361b2d1b6b2feabc2d3e93c75653f5a00e87fa4/moveit_planners/pilz_industrial_motion_planner/src/trajectory_functions.cpp#L571-L574

felixvd, 2021-06-29:
@robertjbush Have you encountered this problem since? If so, is there a minimal example to reproduce the problem?

robertjbush, 2021-06-29:
Hi @felixvd, Ive stopped using `Pilz` as a result of this bug. Is the code and information I specified not sufficient to replicate? Have you not encountered this issue with Pilz?

On Thursday we have a new software starting and this bug will be one of his tasks.

felixvd, 2021-06-30:
The gold standard would be a .test file using one of the robots in `moveit_resources`, containing one plan that succeeds with OMPL and another that fails with a Pilz planner :) 

But it looks like we were able to reproduce a similar (or the same) bug on our system. Using OMPL after disabling collisions for a part of the scene (actually the robot) works, but the LIN planning request fails. I'll look into it.

cambel, 2021-06-30:
Hi, I am checking this problem too. 

I was able to reproduce the issue. In our scenario, there is an object define directly from the robot's URDF so it is considered as part of self-collision, however we can disable the collision checking `planning_scene_interface.py->allow_collisions`. The thing is, this works correctly for the OMPL planners but does not work for the Pilz ->LIN/CIRC.

After reading the code, I think that the problem is that Pilz planner's check if there is self collision when computing an IK solution for the robot. Whether this is the right approach or not, I don't know. The solution would be to compute an IK solution without self collision validation and then let the PlannerScene validate the found plan, OMPL seems to be doing this (wild guess).

From the code: 
The RobotState computeIK is called with the extra `isStateColliding`
https://github.com/ros-planning/moveit/blob/3361b2d1b6b2feabc2d3e93c75653f5a00e87fa4/moveit_planners/pilz_industrial_motion_planner/src/trajectory_functions.cpp#L75-L80

Without this extra check, the behavior is as expected. The Pilz planners can generate a plan over an obstacle that has collision allowed but fail otherwise with the error message:

> Found a contact between 'a_bot_right_inner_finger_pad' (type 'Robot link') and 'tray' (type 'Robot link'), which constitutes a collision. Contact information is not stored.

I tested this by just returning `true` (thus not self collision, weird naming though) here:
https://github.com/ros-planning/moveit/blob/3361b2d1b6b2feabc2d3e93c75653f5a00e87fa4/moveit_planners/pilz_industrial_motion_planner/src/trajectory_functions.cpp#L576

robertjbush, 2021-06-30:
We're going to test @cambel's approach tomorrow or Friday. Thanks for sharing. I'll report back with whether it fixes our bug asap.

aa-tom, 2021-07-02:
Hey all, I'm also taking a look at this in place of @robertjbush. I'm a little new to MoveIt and pilz so please forgive any silly questions!

@cambel I'd be interested in learning how you tested your modified code. Are there unit tests within pilz that could be updated to verify new functionality, or would we have to make the modifications, compile from source, then swap the standard pilz package out with the modified one?

It sounds like MoveIt will apply collision detection at the appropriate part of the planning pipeline, so I'm wondering if the collision detection logic in pilz can be removed completely?

cambel, 2021-07-02:
Hi @aa-tom, we have installed moveit from source, so I just modified the code, compiled and test. I don't know if there is a unit test for it.

As far as I understand, moveit checks collisions for the plan generated by any planner. The extra check of self collision for each IK solution sounds like a good idea if it's limited to the robot's link involved and if it can consider the update collision matrix. So, removing the self collision entirely from the pilz is one option but I guess it'd be better to have the self collision check focus on the parts of the robot that matter.

v4hn, 2021-07-02:
> As far as I understand, moveit checks collisions for the plan generated by any planner.

Yes and no. It's [a feature that can be disabled](https://github.com/ros-planning/moveit/blob/master/moveit_ros/planning/planning_pipeline/include/moveit/planning_pipeline/planning_pipeline.h#L100-L102), though it's enabled by default.
We usually expect planners to do collision checking on their own.

Pilz's approach is quite unsatisfying to me in that regard anyway, because they only check for self-collisions and only for LIN (not for PTP) planning.
So LIN trajectories will stop before self-collisions, but PTP trajectories will not.
Neither will stop before environment collisions because the planner straight ignores the `PlanningScene` it receives together with the planning request and instead [creates an empty one for collision checking](https://github.com/ros-planning/moveit/blob/master/moveit_planners/pilz_industrial_motion_planner/src/trajectory_functions.cpp#L574).
I'm not sure whether there are reasons for this behavior, maybe @jschleicher can comment on that.

So all in all, at the moment the feature *has* to be enabled for MoveIt anyway because of environment collisions, but it's still reasonable to pass a collision checker to the IK plugin (which could in theory still ignore it if that's what you want).

aa-tom, 2021-07-05:
@v4hn @jschleicher Do you know where about in the code the `PlanningScene` would be received from MoveIt? I'm thinking of passing it down to the `isStateColliding()` function and using it as the basis for collision detection rather than creating a new one from scratch, which I'm hoping will propagate the collision matrix

v4hn, 2021-07-05:
[this is the top-level entry point from MoveIt to the planner](https://github.com/ros-planning/moveit/blob/master/moveit_planners/pilz_industrial_motion_planner/src/pilz_industrial_motion_planner.cpp#L121) where the correct scene is passed.

For the service/action interface for sequence planning it's [here instead](https://github.com/ros-planning/moveit/blob/master/moveit_planners/pilz_industrial_motion_planner/src/move_group_sequence_service.cpp#L74).

aa-tom, 2021-07-05:
Excellent, thanks. Based on a look at the code it seems like the `PlanningContext` class is responsible for holding all relevant state required by a particular planner? My current idea is:
1. Modify `PlanningContextBase` to contain a reference to the `PlanningScene`
2. Modify `PlanningContextLoader::loadContext() to populate this reference with the scene`
2. Pass the scene reference from the `PlanningContextPtr` down the method calls `TrajectoryGenerator::generate() -> TrajectoryGenerator::plan() -> pilz_industrial_motion_planner::generateJointTrajectory() -> pilz_industrial_motion_planner::computePoseIK() -> pilz_industrial_motion_planner::isStateColliding()`
3. Use this scene to create the collision detection copy using the goal pose of the robot

Does this sound sensible?

aa-tom, 2021-07-06:
@v4hn I've implemented a fix based on the above approach. What's the process for raising a PR?

v4hn, 2021-07-06:
> What's the process for raising a PR?

I don't get the question. Create a pull-request on github, make sure CI succeeds, and in the best case add a test that is fixed by the PR.

robertjbush, 2021-08-18:
This has been fixed by @aa-tom (with help from @cambel), but is waiting for @jschleicher to review #2803.

closed 2021-08-18 by rhaschke