moveit/moveit #2680: The goals of failed plan requests should be visualized

felixvd, 2021-05-21:
It is an everyday occurrence common for plans to fail, and one of the most common fail cases seems to be a mistake in a goal pose or joint configuration. If the goal was visualized, it would be much easier to evaluate these errors. It would have helped to see at first glance what was wrong in #2677, too.

To implement this, I would add a MarkerArray publisher to the `move_group_capability` base class and visualize the goals of failed plans at the end of `move_action_capability.cpp` and `plan_action_capability.cpp`.

Cartesian goals should be displayed as a frame (and maybe the end effector) at the goal pose. Joint goals can be visualized by setting a `RobotState` to the goal configuration and using `getRobotMarkers`. `pick_place.cpp` already uses a MarkerArray to visualize grasps with the end effector [here](https://github.com/ros-planning/moveit/blob/master/moveit_ros/manipulation/pick_place/src/pick_place.cpp#L175). Some code in `constraint_sampler_tools.cpp` ([here](https://github.com/ros-planning/moveit/blob/master/moveit_core/constraint_samplers/src/constraint_sampler_tools.cpp#L82)) may also be a reference. 

The MarkerArrays should be published to `/move_group/failed_goals` and separated by namespace into `joint_goals` and `pose_goals`.

This is a feasible task for a new contributor. You only need to know how to deal with visualization markers to start working on this. It would be extremely helpful, in my opinion.

rhaschke, 2021-05-21:
Use [MTC](https://github.com/ros-planning/moveit_task_constructor) and its visualization capabilities ;-)
Why should we invest time to implement the same feature on a lower level again?

felixvd, 2021-05-25:
Are you proposing to wrap the `go` and `plan` functions in a Task (possibly optionally)? The point is to make the main convenience interface introspectable. I guess that's an option, I never thought of it that way. How do you envision it, in case someone wants to pick up this issue?

rhaschke, 2021-05-28:
I propose to use the MTC framework for any planning problems - _instead_ of the low-level functions of MoveIt.