moveit/moveit #2681: MoveGroupInterface (python) not reading from /joint_states

christian-rauch, 2021-05-21:
### Description

I am trying to read joint values via the `MoveGroupInterface`'s `get_current_joint_values` in python.

### Your environment
* ROS Distro: Noetic
* OS Version: Ubuntu 20.04
* Source or Binary build? binary
* If binary, which release version? latest

### Steps to reproduce
1. start a move_group
2. run the following example program:
```python
#!/usr/bin/env python3
import rospy
from moveit_ros_planning_interface._moveit_move_group_interface import MoveGroupInterface
from moveit_commander import roscpp_initialize
import sys

class ReadJoints:
    def __init__(self) -> None:
        rospy.init_node("joint_read_test")
        roscpp_initialize(sys.argv)

        self.cmd_head = MoveGroupInterface("head", "robot_description", rospy.get_namespace())

        ret = self.cmd_head.start_state_monitor(5.0)
        print("started?", ret)

        rospy.Timer(rospy.Duration(1), self.process, reset = True)

    def run(self):
        rospy.spin()

    def process(self, event=None):
        state = self.cmd_head.get_current_joint_values()
        print("state", state)

if __name__ == '__main__':
    ReadJoints().run()
```

### Expected behaviour
After `start_state_monitor` the node `joint_read_test` should connect to some `/joint_states` topic and print the joint values once per second.

### Actual behaviour
The node calls `start_state_monitor` successfully (returns true), but `rosnode info /joint_read_test` only lists the `/clock` topic (I am running via Gazebo) and getting the joint values fails with `Failed to fetch current robot state`.

I adapted `roscpp_initialize` from https://github.com/ros-planning/moveit/issues/1187#issuecomment-508891103.

It also prints `Didn't received robot state (joint angles) with recent timestamp within 1 seconds.` on stdout but I don't see any parameter to change this timeout from 1s to something higher.

Btw, there are 2 typos in this one message :-)
1. "Didn't received" -> "Didn't receive"
2. "1 seconds" -> "1 second"

v4hn, 2021-05-30:
> Expected behaviour
> After start_state_monitor the node joint_read_test should connect to some /joint_states topic

It sounds unintuitive, but no, this is not to be expected.
`moveit_commander` uses C++ bindings, so the subscribers will be created from the C++ ROS node that is started via `roscpp_initialize` (for my test just now `/move_group_commander_wrappers_1622405841309722262`.
Note that any python process that uses moveit_commander is effectively **two** ROS nodes. The python node and a C++ node that is initialized for the bindings.

Aside from that I cannot reproduce your issue with `moveit_resources_panda_moveit_config`'s `demo.launch`, so it's likely related to your Gazebo setup.

> It also prints Didn't received robot state (joint angles) with recent timestamp within 1 seconds. on stdout

This hints at your actual problem. I suspect gazebo might take more than a second [WallTime](https://github.com/ros-planning/moveit/blob/master/moveit_ros/planning/planning_scene_monitor/src/current_state_monitor.cpp#L219) to publish joint states for the currently published clock.

> but I don't see any parameter to change this timeout from 1s to something higher.

The constant is the second parameter's default for `getCurrentState` [not specified in the respective call](https://github.com/ros-planning/moveit/blob/master/moveit_ros/planning_interface/move_group_interface/src/move_group_interface.cpp#L2085).
The `wait` parameter is thus also not forwarded to the Python-API (not even in `get_current_state`).
Feel free to write a patch and we'll merge it.

> Btw, there are 2 typos in this one message :-)

I fixed one of them [a few days before in master](https://github.com/ros-planning/moveit/blame/master/moveit_ros/planning/planning_scene_monitor/src/current_state_monitor.cpp#L229) :-)
The other one would require logic to get the special case right and I just don't care...

AswinkarthikeyenAK, 2022-02-20:
@christian-rauch ,
Were you able to fix this? 
Is so, can you please let me know how to get values from get_current_pose().pose or from get_current_joint_values() without running into Failed to fetch current robot state using the Python API?

Thanks