moveit/moveit #2696: CI broken

rhaschke, 2021-05-28:
We still have a serious issue in our GHA config:
While the CI checks for PR #2691 (as well as the first master build after merging it) succeeded:
* https://github.com/ros-planning/moveit/actions/runs/879728413
* https://github.com/ros-planning/moveit/actions/runs/881887578

now builds are failing due to the `-pedantic` flag:
https://github.com/ros-planning/moveit/actions/runs/885022060

I think the workspace caching isn't correctly working yet: In the succeeding master build, the `target_ws` was restored from cache, in the failing one not (for a reason I don't understand).

I'm afraid we need to remove the -pedantic flag for now because we cannot quickly adapt `prbt_ikfast_manipulator_plugin` in `moveit_resources`. In any case, we should understand the errornous caching. @tylerjw, do you have any insight?

v4hn, 2021-05-28:
> I'm afraid we need to remove the -pedantic flag for now

Fair enough. Please go ahead I don't have the time right now.
I will work on fixes later.

rhaschke, 2021-06-07:
The caching issue is hard to grasp... Closed for now.

closed 2021-06-07 by rhaschke