moveit/moveit #2699: PSM ros params wrongly needs namespace prefix

v4hn, 2021-05-31:
Quoting @mintar from [here](https://github.com/ros-planning/moveit/issues/1330#issuecomment-851349546):

> Just a further note for everyone else who stumbles upon this: I'm using a prefix (mobipick) for my robot. It used to work exactly how they described:
> 
> mobipick/move_group/robot_description_planning/default_attached_padding: 0.0
> 
> In the current ROS Noetic version of MoveIt (1.1.5), there's a second mobipick necessary to make it work:
> 
> mobipick/move_group/mobipick/robot_description_planning/default_attached_padding: 0.0

Looking into it a bit this is probably a bug introduced in Henning's efforts to have `move_group` use `MoveItCpp` for support of multiple planning pipelines.
Specifically, [this commit](https://github.com/ros-planning/moveit/pull/2127/commits/8df40df56ad58a6358834129b09b3a8b2c35a02c) replaces `move_group`'s own PlanningSceneMonitor with the one initialized by MoveItCpp.

Please have a look @henningkayser .

mintar, 2021-05-31:
I investigated further, and it turns out that this behavior is there since forever, so @henningkayser's commits are not to blame here. Sorry about the confusion.

I've tested the following configurations:

* ROS Noetic, Ubuntu 20.04, Moveit 1.1.5
* ROS Noetic, Ubuntu 20.04, Moveit 1.1.3
* ROS Noetic, Ubuntu 20.04, Moveit 1.1.1
* ROS Kinetic, Ubuntu 16.04, Moveit 1.1.1~6
* ROS Kinetic, Ubuntu 16.04, Moveit 0.9.18

They all show the same behavior:

Parameter path without a prefix:

`/move_group/robot_description_planning/default_attached_padding: 0.0`

Parameter path with prefix `mobipick` (note the second "mobipick"):

`/mobipick/move_group/mobipick/robot_description_planning/default_attached_padding: 0.0`

By the way, there are a couple of other parameters read from `robot_description_planning`. These are also consistent between ROS Kinetic/Ubuntu 16.04/Moveit 0.9.18 and ROS Noetic/Ubuntu 20.04/Moveit 1.1.5.

`/mobipick/robot_description_planning/cartesian_limits` (only Noetic)
`/mobipick/robot_description_planning/default_collision_operations`
`/mobipick/robot_description_planning/joint_limits`
`/mobipick/robot_description_planning/shape_transform_cache_lookup_wait_time`

So **the real bug here** seems to be that the padding parameters are read from a **private namespace** (`/mobipick/move_group/`), while the others are not. This behavior was there forever. Perhaps should be changed for consistency reasons, along with a nice warning when the old parameters are present.

v4hn, 2021-06-02:
Looks like you found a broken commit 7 years ago from @davetcoleman .
https://github.com/ros-planning/moveit/blame/master/moveit_ros/planning/planning_scene_monitor/src/planning_scene_monitor.cpp#L1425-L1427
The comment does not make any sense to me. A leading slash is no "bad param server address").

I believe the problem should be gone if you simply remove the conditional that removes the leading `/`.
Could you please verify and potentially create a pull request?

mintar, 2021-06-02:
>  Could you please verify and potentially create a pull request?

Here you go: #2706

PR #2706 (2021-06-05): PSM: Don't read padding parameters from private namespace
Before this commit, the padding parameters were read from the private namespace of the move_group node (unlike all the other PSM parameters).

Fixes #2699.
commit bab87f9a (2021-06-05): PSM: Don't read padding parameters from private namespace (#2706)

* PSM: Don't read padding parameters from private namespace

Before this commit, the padding parameters were read from the private
namespace of the move_group node (unlike all the other PSM parameters).

Fixes #2699.
closed 2021-06-05 by v4hn