moveit/moveit #2701: Collision checking only includes some links from a EEF group with multiple fingertips

Levaru, 2021-05-31:
### Description

I'm trying to perform a collision check between my end-effector group and a collision object in the planning scene. The eef is a simple two finger parallel gripper, with the following tree:
```
                |-> gripper_left_link -> finger_left_link
tool_base_link--|
                |-> gripper_right_link -> finger_right_link
```
The joints of the left side mimic the joints of the right side.

The eef group is defined in the SRDF like this:
```
<group name="robot_tool">
            <link name="tool_base_link" />
            
            <link name="gripper_left_link" />
            <link name="finger_left_link" />
            <link name="gripper_right_link" />
            <link name="finger_right_link" />
</group> 
```

When performing a collision check via `isStateColliding` or `collision_detection::CollisionRequest` and specifying the group name of the end-effector, it will only check collisions for one link-chain of the end-effector. In my case the links are `tool_base_link->gripper_right_link->finger_right_link`.

If I don't specify a group name, it will check all links correctly. 

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04
* Binary build, latest release version for melodic

### Expected behaviour
Collision checking should include all links that are specified in the SRDF group.

### Actual behaviour
The collision check will only include the link-chain of one finger tip.

Levaru, 2021-05-31:
I just modified the URDF of the gripper by removing the `<mimic/>` tag from the left-side joints. This fixed the issue, so it seems that link-chains which include a mimic joint will be ignored when performing collision checks. 

I would like to keep the mimic joint, is there some other way to include the links when performing a collision check?

v4hn, 2021-06-01:
Mimic joints in a group should of course not be ignored during collision checking. So it looks like you found a bug.

Could you provide a minimal example (or possibly a test) to reproduce this? It would help with debugging the issue.

Levaru, 2021-06-01:
@v4hn I was not able to reproduce this bug in a minimal example package. While comparing the URDF's between my robot and the one used for the example, I couldn't spot any difference. 

I randomly decided to swap the mimic tag from the left joint to the right one and for some reason this fixed the issue. I did the same thing in my minimal example package and was able to reproduce the bug. What would be the best way to provide you my example package?

v4hn, 2021-06-01:
> I randomly decided to swap the mimic tag from the left joint to the right one and for some reason this fixed the issue.

O.o

> I did the same thing in my minimal example package and was able to reproduce the bug.
> What would be the best way to provide you my example package?

A git repository(+rosinstall file if you need dependencies) would be nice.
A tarball with a (or a few) ROS packages inside works as well.

Levaru, 2021-06-01:
@v4hn Here is the tarball with the packages:
[collision_test.tar.gz](https://github.com/ros-planning/moveit/files/6576261/collision_test.tar.gz)

You will need some additional ROS and MoveIt packages that should be available through the standard ROS repository for Melodic and Ubuntu 18.04.

I have prepared two different URDF files for the end-effector called `hand_with_bug.xacro` and `hand_without_bug.xacro`. To test one or the other you should follow these steps:

1. Change the commented lines inside `panda_arm_hand.urdf.xacro` (package: `franka_description`) to load the desired version.
2. Change the commented lines inside `collision_test.cpp` (package: `collision_test`) to the correct `finger_joint_name`.
3. Run `catkin build`
4. Run `roslaunch panda_moveit_config demo.launch` in a terminal and wait for it to load
5. Run `roslaunch collision_test collision_test.launch` in another terminal
6. Start the collision test with the Next button of the RvizVisualToolsGui and keep clicking it to progress through the test

The collision_test will first spawn a collision object inside the gripper. Then it will gradually bring the gripper-fingers closer together and performing a collision check everytime the Next button is pressed. It will first perform a collision check for the whole robot (no planning group name set) and then repeat the process for the end-effector planning group.

What you should see happening:
(Without the bug) For `req.group_name = "";` `req.group_name = "hand";` the collision check will report a collision between the collision object and both fingers.

(With the bug) For `req.group_name = "";` the collision check will report a collision between the collision object and both fingers but for `req.group_name = "hand";` it will only be one finger.

Levaru, 2021-06-01:
@v4hn Oops, i'm sorry that tarball is the wrong version. Here is the correct one:
[collision_test.tar.gz](https://github.com/ros-planning/moveit/files/6576547/collision_test.tar.gz)

simonschmeisser, 2021-06-01:
Please try calling `robot_state.update()` after setting the joint positions

Levaru, 2021-06-01:
@simonschmeisser Updating the robot state didn't help.

Levaru, 2021-06-02:
@simonschmeisser @v4hn 

I think I found the root of this bug. If a joint mimics another joint but has a higher name if sorted alphabetically, then the child link-chain of that joint will be ignored in collision checks (maybe also in other places, idk).

Simple example to illustrate this:
```
<joint name="joint_2" type="prismatic">
    <!-- joint params -->
</joint>

<joint name="joint_1" type="prismatic">
    <!-- joint params -->
    <mimic joint="joint_2"/>
</joint>
```
This config will cause the link-chain after `joint_1` to be ignored. Both `joint_1` and `joint_2` have almost the same name, with the number being the only difference. If sorted alphabetically, `joint_1` comes before `joint_2` but since `joint_1` has the mimic tag, it breaks for some reason.

This config will work though:
```
<joint name="joint_1" type="prismatic">
    <!-- joint params -->
</joint>

<joint name="joint_2" type="prismatic">
    <!-- joint params -->
    <mimic joint="joint_1"/>
</joint>
```

v4hn, 2021-06-02:
> (With the bug) For req.group_name = ""; the collision check will report a collision between the collision object and both fingers but for req.group_name = "hand"; it will only be one finger.

I can't easily reproduce the behavior with the tarball on a current `master` branch build.
Please check whether building the `master` branch (works on melodic) fixes the problem for you.
Also your test is quite nice, but still quite far from a minimal example. :)
Could you possibly simplify it to a "compile, run, see "THIS IS WRONG" being printed because a conditional/assert fails if you can reproduce with `master`?

v4hn, 2021-06-02:
Also please note that setting the value of a mimic joint explicitly will not automatically update the joint it mimics.

If you are interested in such a feature, please provide a patch.

Levaru, 2021-06-02:
@v4hn 
> Could you possibly simplify it to a "compile, run, see "THIS IS WRONG" being printed because a conditional/assert fails if you can reproduce with `master`?

To be honest, I don't know how to simplify it further. I just tried compiling the MoveIt `master` branch but it requires some python3 tools that I can't install without completely breaking my current setup.

 > Also please note that setting the value of a mimic joint (a) explicitly will not automatically update the joint it mimics (b).
Instead, the next time (b) is set, your custom value for (a) will be overwritten.

I'm not setting the value for the mimic joint (a) only for the main joint (b). Otherwise I would just remove the mimic tag and set the positions for both joints. In my example with the bug `panda_finger_joint2` is the main joint and `panda_finger_joint1` is mimicking it and I only set the values for `panda_finger_joint2`.