moveit/moveit #2702: CHOMP doesn't read parameters from proper namespace

mintar, 2021-05-31:
### Description

Parameters read by `move_group` (`~` is `/move_group`):

```
# 1. Params that should be in a namespace, but aren't:
- ~/collision_clearence
- ~/collision_threshold
- ~/enable_failure_recovery
- ~/joint_update_limit
- ~/learning_rate
- ~/max_iterations
- ~/max_iterations_after_collision_free
- ~/max_recovery_attempts
- ~/obstacle_cost_weight

# 2. Params that are in a namespace, but shouldn't (or MSA templates should be adjusted)
- ~/planning_pipelines/chomp/jiggle_fraction
- ~/planning_pipelines/ompl/jiggle_fraction

# 3. Params that are attempted to be read, but are not in the config file (that's fine, just uses defaults then)
- ~/planning_pipelines/chomp/capabilities
- ~/planning_pipelines/chomp/default_workspace_bounds
- ~/planning_pipelines/chomp/max_sampling_attempts
- ~/planning_pipelines/chomp/start_state_max_dt

# 4. Params where config file and code agree (yay!)
- ~/planning_pipelines/chomp/planning_plugin
- ~/planning_pipelines/chomp/request_adapters
- ~/planning_pipelines/chomp/start_state_max_bounds_error
```

Parameters written by setup assistant:

```
# 1. 
- ~/planning_pipelines/chomp/collision_clearence
- ~/planning_pipelines/chomp/collision_threshold
- ~/planning_pipelines/chomp/enable_failure_recovery
- ~/planning_pipelines/chomp/joint_update_limit
- ~/planning_pipelines/chomp/learning_rate
- ~/planning_pipelines/chomp/max_iterations
- ~/planning_pipelines/chomp/max_iterations_after_collision_free
- ~/planning_pipelines/chomp/max_recovery_attempts
- ~/planning_pipelines/chomp/obstacle_cost_weight

# 2.
- ~/jiggle_fraction

# 3. Params that are written, but never read; perhaps this is fine (I didn't actually run CHOMP, so maybe they are only read in specific code execution paths)
- ~/planning_pipelines/chomp/planning_time_limit
- ~/planning_pipelines/chomp/pseudo_inverse_ridge_factor
- ~/planning_pipelines/chomp/ridge_factor
- ~/planning_pipelines/chomp/smoothness_cost_acceleration
- ~/planning_pipelines/chomp/smoothness_cost_jerk
- ~/planning_pipelines/chomp/smoothness_cost_velocity
- ~/planning_pipelines/chomp/smoothness_cost_weight
- ~/planning_pipelines/chomp/use_pseudo_inverse
- ~/planning_pipelines/chomp/use_stochastic_descent

# 4. Params where config file and code agree (yay!)
- ~/planning_pipelines/chomp/planning_plugin
- ~/planning_pipelines/chomp/request_adapters
- ~/planning_pipelines/chomp/start_state_max_bounds_error
```

By the way, is it to late to rename `collision_clearence` to `collision_clearance`?

### Environment

* ROS Distro: Noetic
* OS Version: Ubuntu 20.04
* Binary build, release version 1.1.5

### Steps to reproduce

* Install [rosautodoc](https://github.com/bponsler/rosautodoc)
* 1st terminal: `roscore`
* 2nd terminal: `mkdir /tmp/autodoc-output && cd /tmp/autodoc-output && rosautodoc`
* 3rd terminal: `export ROS_MASTER_URI=http://localhost:33133 && roslaunch <your_moveit_config> moveit_planning_execution.launch`
* wait until startup complete
* kill all processes and compare `/tmp/autodoc-output/*_move_group.md` (= parameters read attempts by move_group) and `/tmp/autodoc-output/roslaunch.md` (= parameters put on the parameter server by the launch file)

v4hn, 2021-06-02:
> # 2. Params that are in a namespace, but shouldn't (or MSA templates should be adjusted)
> - ~/planning_pipelines/<pipeline>/jiggle_fraction

The template needs to be adjusted here. The jiggle_fraction is a request adapter parameter and these might vary between pipelines.

> By the way, is it to late to rename collision_clearence to collision_clearance?

Can be renamed, but it would make sense to add a check for the old spelling and trigger a warning in this case.

> # 1. Params that should be in a namespace, but aren't:

Chomp [currently ignores the passed namespace argument](https://github.com/ros-planning/moveit/blob/master/moveit_planners/chomp/chomp_interface/src/chomp_plugin.cpp#L52) and thus reads all it's parameters from the wrong namespace.
This looks somewhat straightforward to fix, but requires some testing.

Any chance we can get these contributions from you @mintar ? :pleading_face:

mintar, 2021-06-02:
> Any chance we can get these contributions from you @mintar ?

Here you go: #2707. Apparently, you decided I should have a MoveIt day! ;-)

v4hn, 2021-06-04:
> Apparently, you decided I should have a MoveIt day! ;-)

Well, you spent the time surveying and debugging all inconsistent ROS-parameters, so I thought you might be interested to have them fixed as well. :-)
Thanks for your contributions!

PR #2707 (2021-06-10): CHOMP: Read parameters from proper namespace
This PR contains the following changes (see separate commits):

* CHOMP: Read parameters from proper namespace
* CHOMP: Rename param 'clearence' to 'clearance'
* MSA template: Put param jiggle_fraction into proper namespace

Fixes #2702.
commit 071d8544 (2021-06-10): CHOMP: Read parameters from proper namespace

Fixes #2702.
closed 2021-06-10 by v4hn