moveit/moveit #2704: Publishing CollisionObject message (via rospy) with more than one Plane crashes MoveIt

FSund, 2021-06-01:
### Description

If I try adding multiple shape_msgs/Plane via a single moveit_msgs/CollisionObject message (via rospy), MoveIt crashes. Adding a single plane works fine.

### Your environment
* ROS Distro: Noetic
* OS Version: Ubuntu 20.04 (Pop!_OS 20.04 LTS)
* Everything is installed via apt 

```
$ apt search ros-noetic-moveit
Sorting... Done
Full Text Search... Done
ros-noetic-moveit/focal,now 1.1.5-1focal.20210524.225952 amd64 [installed]
  Meta package that contains all essential package of MoveIt.
```

### Steps to reproduce
- Start `demo.launch`
- Publish a message via rospy on the `/collision_object` topic, with `planes` and `plane_poses` containing more than 1 object

### Expected behaviour
The planes should appear as a scene object in RViz.

### Actual behaviour
MoveIt crashes, with the following output:

    [move_group-5] process has died [pid 264641, exit code -11, cmd /opt/ros/noetic/lib/moveit_ros_move_group/move_group __name:=move_group __log:=/home/filip/.ros/log/def9fd40-c2d4-11eb-8518-c5a414c77dea/move_group-5.log].
    log file: /home/filip/.ros/log/def9fd40-c2d4-11eb-8518-c5a414c77dea/move_group-5*.log

If I try planning via RViz I get the following warning:

    [ WARN] [1622550530.487940504] [/rviz_pop_os_264611_545357757035741097]: move action server not connected

v4hn, 2021-06-02:
Can you add them as two separate `CollisionObject`s instead of making them one with two primitives?
Might be related to @felixvd's ongoing efforts for better support [with multiple collision shapes in one object](https://github.com/ros-planning/moveit/pull/2037).

felixvd, 2021-06-02:
I don't think that PR will fix this problem, but I suppose it is related

FSund, 2021-06-02:
> Can you add them as two separate `CollisionObject`s instead of making them one with two primitives?
> Might be related to @felixvd's ongoing efforts for better support [with multiple collision shapes in one object](https://github.com/ros-planning/moveit/pull/2037).

Yes, adding via separate CO's work fine, so I'm using that as a workaround. So not a huge issue for me, but thought it might be of interest, since it seems like it should work.