moveit/moveit #2709: max_range is not applied to "Free Voxels" in PointCloudOctomapUpdater

FSund, 2021-06-03:
### Description

The parameter `max_range` (typically set in `sensor_manager.launch.xml` or `sensors_3d.yaml`) seems to only apply to occupied voxels, not free voxels.

Is this the intended behaviour? It makes my computer very slow when I update the Octomap, using scan data that goes out to 10-15 meters, even if i use a max range of 3 meters. RViz also has a lot of trouble switching between "All Voxels", "Occupied Voxels" and "Free Voxels".

### Your environment
* ROS Distro: Noetic
* OS Version: Ubuntu 20.04 (Pop!_OS 20.04 LTS)
* Everything is installed via apt 

```
$ apt search ros-noetic-moveit
Sorting... Done
Full Text Search... Done
ros-noetic-moveit/focal,now 1.1.5-1focal.20210524.225952 amd64 [installed]
  Meta package that contains all essential package of MoveIt.
```

FSund, 2021-06-03:
Based on some quick testing, this seems to be the case for both `occupancy_map_monitor/DepthImageOctomapUpdater` and `occupancy_map_monitor/PointCloudOctomapUpdater`.

v4hn, 2021-06-04:
> The parameter `max_range` (typically set in `sensor_manager.launch.xml` or `sensors_3d.yaml`) seems to only apply to occupied voxels, not free voxels.
> 
> Is this the intended behaviour?

At first glance I would say no, but I'm not entirely sure what you are experiencing.
If everything further away from the center than 3m is empty I would not expect such a big performance drop.
After all this is what octomaps improve over voxel grids.

Could you post screenshots of the planning scene and scan input or even try to find a patch that improves behavior for your use case?

FSund, 2021-06-04:
In short, the "Free Voxels" appear much farther out than what I set as `max_range`, but "Occupied Voxels" only appear within the limit.

I will try to produce some screenshots this weekend.

v4hn, 2021-08-28:
> Similar to the occupied voxels image, shouldn't be too-far-away voxels just skipped completely? 

I'm not sure I follow. Do you mean [like this](https://github.com/ros-planning/moveit/pull/2848/files#diff-3668caa85ddb6fd526274703c836ab4edb57ab186875960234b6a6f806728c58L263-L264)?
That boils down to whether you consider a clipped beam to represent an obstacle too far away or a measurement error.
If you know clipping means there is free space in the entire range, and that's what the code currently assumes, obstacles have to be cleared that were present at this point in the map and thus the cells have to be considered.

> [...] resulting in much more voxels representing the free space than before. This will definitely impact efficiency!

Could you elaborate? this patch **reduces** the length of the traced beam for clipped cells and thus should reduce the number of cells in the voxel grid that are considered.

FSund, 2021-08-30:
> That boils down to whether you consider a clipped beam to represent an obstacle too far away or a measurement error.

That is a good point which I hadn't considered. It is possible to have both options by adding a `clip_distance` parameter, in addition to `max_range`. Setting `max_range` to 0 and `clip_distance` to something non-zero could give the behavior implemented in your PR.

v4hn, 2021-08-30:
> It is possible to have both options by adding a `clip_distance` parameter, in addition to `max_range`.
> Setting `max_range` to 0 and `clip_distance` to something non-zero could give the behavior implemented in your PR.

I dislike the approach because clipping should *never* mean that there is a different free space to consider. Either it's the same distance or `0.0`.

Looking at the code again, the whole block is already wrapped in a `!isnan` conditional. So assuming invalid measured points are indicated by nan (as is the convention in PCL afaik) we only add valid measurements that are too far away to `clip_cells`.

I'll wait for feedback from @rhaschke here.

rhaschke, 2021-08-30:
First of all, note that I never used the octomap feature - so correct me if I'm wrong.
As far as I understand, it explicitly represents both, occupied and free voxels separately. Currently, occupied voxels are clipped beyond a distance of `max_range`. Free voxels are not.
Given, the two renderings of free voxels before and after this PR:
|Before | After |
|----------|-----------|
|[image] |[image] |

I noticed that before free voxels are represented by "a few" large voxels, while after this PR, we see many small voxels along the `max_range` boundary. However, I agree that by clipping them away, in general, the number of voxels will be reduced.

> That boils down to whether you consider a clipped beam to represent an obstacle too far away or a measurement error.

I would consider voxels beyond the `max_range` threshold as irrelevant (don't care) for the application because I know the robot will only move within this range. Thus, clipping them - as you do right now - should be fine.

v4hn, 2021-08-30:
> First of all, note that I never used the octomap feature - so correct me if I'm wrong.
As far as I understand, it explicitly represents both, occupied and free voxels separately. 

These octomaps represent a voxelgrid by storing explicit voxel nodes with an occupancy log probability.
Every voxel not explicitly represented by such a node has no value assigned.
The key feature in the tree is pruning in (aligned) powers of two: If all nodes on a level in an aligned block of twice the current voxel size share the same log probability, they are pruned into their single "parent" node.

> I noticed that before free voxels are represented by "a few" large voxels, while after this PR, we see many small voxels along the max_range boundary.

That took me a while to think out. You are right. I believe raytracing through the whole volume of empty space will often result in pruned octomap nodes.
The `max_range` sphere does not align with any aligned grid boundary, so any explicit voxel (with a probability assigned) on the outer hull of the sphere will almost surely be represented at the highest resolution. That's suboptimal.
At the same time, there is a lot less volume to consider for which we create nodes at the moment (even though they are bigger).
It's a trade-off, and either implementation might be bad for an average cloud input (if such a thing exists).
I don't think there is a valid argument here though without solid benchmarks and I'm not willing to look into that.

To support my patch, I would simply require that `max_range` should behave in an intuitive way - by not adding explicit voxels anywhere outside the sphere, similar to my patch.

I'm not sure whether the octomap *additionally* restricts the maximum represented volume somewhere. Looking through the code I'm unsure how expanding volumes are handled. At least the [`max_range` parameter in our templates](https://github.com/ros-planning/moveit/blob/master/moveit_setup_assistant/templates/moveit_config_pkg_template/launch/sensor_manager.launch.xml#L11) is not used for it. (It does not seem to be used for *anything* right now :confused:)
I found that we can set a bounding box for the octomap and [tried to use it](https://github.com/v4hn/moveit/commit/d85e554ec0a1748b0b0dc0248da3c2c5b49265f1), but reading octomap's sources, this functionality is not supported by the `updateNode` method we use...

A side point is that [octomap's own `insertRay` method](https://github.com/OctoMap/octomap/blob/devel/octomap/include/octomap/OccupancyOcTreeBase.hxx#L867-L873), which we could actually use in our code if anyone would be interested to look into this, also truncates rays at a max_range.

xiaopgaxiaopg, 2022-06-06:
> EDIT: The first part of this issue has been solved. See [#2709 (comment)](https://github.com/ros-planning/moveit/issues/2709#issuecomment-908128055) and the followup answer for unaddressed parts of the issue.
> 
> ### Description
> The parameter `max_range` (typically set in `sensor_manager.launch.xml` or `sensors_3d.yaml`) seems to only apply to occupied voxels, not free voxels.
> 
> Is this the intended behaviour? It makes my computer very slow when I update the Octomap, using scan data that goes out to 10-15 meters, even if i use a max range of 3 meters. RViz also has a lot of trouble switching between "All Voxels", "Occupied Voxels" and "Free Voxels".
> 
> ### Your environment
> * ROS Distro: Noetic
> * OS Version: Ubuntu 20.04 (Pop!_OS 20.04 LTS)
> * Everything is installed via apt
> 
> ```
> $ apt search ros-noetic-moveit
> Sorting... Done
> Full Text Search... Done
> ros-noetic-moveit/focal,now 1.1.5-1focal.20210524.225952 amd64 [installed]
>   Meta package that contains all essential package of MoveIt.
> ```

> It does indeed look like your changes fixed the issue with free vs. occupied voxels. Great work!
> 
> The `max_range` parameter for the octomap does not seem to have any effect, but that is a separate issue, and not really a problem for my use case if the octomap updater is fixed.
> 
> Occupied Voxels: [image]
> 
> Free Voxels [image]
Dear sir, I have emailed you in hopes of getting help

xiaopgaxiaopg, 2022-06-06:
Dear sir, I have emailed you in hopes of getting help

xiaopgaxiaopg, 2022-06-06:
I used a D435 camera to simulate Otomap in a gazebo environment, there will be a wall-like thing, I personally think it is a point cloud, but I don't know how to remove this, I see that Mr. Simulation does not have this in the simulation map, so I asked for help。
[image]
[image]

PR #2848 (2021-08-30): clipped points should only be considered up to the max_range
Otherwise we add "free" voxels outside the max_range volumes for no reason.

I did not test it myself but @FSund posted screenshots in the issue that demonstrate the impact.

Fixes #2709