moveit/moveit #2710: eigen_stl_containers is not found

fjnn, 2021-06-03:
### Description

Trying to install the package on Noetic but `eigen_stl_containers` is not found.

### Your environment
* ROS Distro: [Noetic]
* OS Version: e.g. Ubuntu 20.04
* Source build
* noetic-devel

### Steps to reproduce
Fresh ROS Noetic installed with `sudo apt install ros-noetic-desktop-full`
Fresh moveit workspace created.
The package is clonned in noetic-devel

### Expected behaviour
run `catkin_make` and be everything compiled

### Actual behaviour
```
CMake Error at /opt/ros/noetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by
  "eigen_stl_containers" with any of the following names:

    eigen_stl_containersConfig.cmake
    eigen_stl_containers-config.cmake

```

### Additional
Environment variables:

```
printenv | grep -i ROS
ROS_VERSION=1
PKG_CONFIG_PATH=/opt/ros/noetic/lib/pkgconfig
ROS_PYTHON_VERSION=3
ROS_PACKAGE_PATH=/opt/ros/noetic/share
ROSLISP_PACKAGE_DIRECTORIES=
ROS_ETC_DIR=/opt/ros/noetic/etc/ros
CMAKE_PREFIX_PATH=/opt/ros/noetic
PYTHONPATH=/opt/ros/noetic/lib/python3/dist-packages
ROS_MASTER_URI=http://localhost:11311
LD_LIBRARY_PATH=/opt/ros/noetic/lib
PATH=/opt/ros/noetic/bin:/home/gizem/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/opt/mattermost-desktop-4.6.2-linux-x64
ROS_ROOT=/opt/ros/noetic/share/ros
ROS_DISTRO=noetic
```
Tried installing `rosdep install -r --from-path . --ignore-src --rosdistro eloquent -y`
```
rosdep install -r --from-path . --ignore-src --rosdistro eloquent -y
WARNING: given --rosdistro eloquent but ROS_DISTRO is "noetic". Ignoring environment.
WARNING: ROS_PYTHON_VERSION is unset. Defaulting to 3
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
moveit_ros_manipulation: Cannot locate rosdep definition for [moveit_msgs]
moveit_ros_planning: Cannot locate rosdep definition for [moveit_resources_panda_moveit_config]
moveit_servo: Cannot locate rosdep definition for [moveit_resources_panda_moveit_config]
moveit_kinematics: Cannot locate rosdep definition for [moveit_resources_panda_moveit_config]
moveit_core: Cannot locate rosdep definition for [orocos_kdl]
moveit_ros_warehouse: Cannot locate rosdep definition for [warehouse_ros]
pilz_industrial_motion_planner_testutils: Cannot locate rosdep definition for [moveit_msgs]
moveit_ros_move_group: Cannot locate rosdep definition for [moveit_resources_fanuc_moveit_config]
moveit_ros_occupancy_map_monitor: Cannot locate rosdep definition for [geometric_shapes]
moveit_setup_assistant: Cannot locate rosdep definition for [moveit_resources_panda_moveit_config]
moveit_planners_ompl: Cannot locate rosdep definition for [moveit_resources_panda_description]
moveit_commander: Cannot locate rosdep definition for [moveit_resources_fanuc_moveit_config]
moveit_ros_visualization: Cannot locate rosdep definition for [object_recognition_msgs]
moveit_ros_planning_interface: Cannot locate rosdep definition for [moveit_resources_panda_moveit_config]
moveit_ros_perception: Cannot locate rosdep definition for [object_recognition_msgs]
pilz_industrial_motion_planner: Cannot locate rosdep definition for [code_coverage]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully

```

In fact, I believe that I have the proper source lists: `grep -h ^deb /etc/apt/sources.list /etc/apt/sources.list.d/*`
```
deb http://no.archive.ubuntu.com/ubuntu/ focal main restricted
deb http://no.archive.ubuntu.com/ubuntu/ focal-updates main restricted
deb http://no.archive.ubuntu.com/ubuntu/ focal universe
deb http://no.archive.ubuntu.com/ubuntu/ focal-updates universe
deb http://no.archive.ubuntu.com/ubuntu/ focal multiverse
deb http://no.archive.ubuntu.com/ubuntu/ focal-updates multiverse
deb http://no.archive.ubuntu.com/ubuntu/ focal-backports main restricted universe multiverse
deb http://security.ubuntu.com/ubuntu focal-security main restricted
deb http://security.ubuntu.com/ubuntu focal-security universe
deb http://security.ubuntu.com/ubuntu focal-security multiverse
deb http://apt.insync.io/ubuntu focal non-free contrib
deb [arch=amd64 signed-by=/usr/share/keyrings/ros-archive-keyring.gpg] http://packages.ros.org/ros2/ubuntu focal main
deb [arch=amd64 signed-by=/usr/share/keyrings/ros-archive-keyring.gpg] http://packages.ros.org/ros2/ubuntu focal main
deb http://packages.ros.org/ros/ubuntu focal main
deb [arch=amd64] https://packages.microsoft.com/repos/ms-teams stable main
```

But even `sudo apt install ros-eloquent-eigen-stl-containers` doesn't work:
```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package ros-eloquent-eigen-stl-containers
```
Any advices?

fjnn, 2021-06-03:
When I tried binary install `sudo apt install ros-noetic-moveit`, the moveit packages are installed but building from source still doesn't work.

JafarAbdi, 2021-06-03:
> Tried installing rosdep install -r --from-path . --ignore-src --rosdistro eloquent -y

@fjnn You need to use `noetic` as a ROS distro rather than `eloquent`

`rosdep install -r --from-path . --ignore-src --rosdistro noetic -y`

JafarAbdi, 2021-07-01:
Closing due to inactivity, please reopen it if you still see this problem

closed 2021-07-01 by JafarAbdi