moveit/moveit #2711: Turtlebot3 with OM-X moveit move() doesn't wait for trajectory to complete before sending another one.

slavenpetkovic, 2021-06-03:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: [Melodic]
* OS Version: Ubuntu 18.04
* Source or Binary build?
* If binary, which release version?
* If source, which branch?

### Steps to reproduce
Tell us how to reproduce this issue. Attempt to provide a working demo, perhaps using Docker.

### Expected behaviour
Tell us what should happen

### Actual behaviour
Tell us what happens instead

### Backtrace or Console output

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

 While using Turtlebot3 with OM-X for pick and place function, I used MoveGroupInterface to move arm to desired position using this move_arm function (code below). First I tried with plan + move() and with plan + execute functions and in main i called that move_arm function a few times to move arm to position1 then to position2 etc.. 

But it doesn't wait for first trajectory to complete before sending another one. The definition of that move() function says that it should wait for trajectory to complete.
        
Then i tried to monitor what is going on with /move_group/feedback and /move_group/status topics and it turns out that robot publishes that the arm reached goal state before it actually reached it. (status = 3 or something like that immediately after giving command to move).

//Move to valid NAMED pose (used for arm)
void move_arm(moveit::planning_interface::MoveGroupInterface& move_group_interface, const std::string &name){
ros::NodeHandle node_handle;
move_group_interface.setNamedTarget(name);
moveit::planning_interface::MoveGroupInterface::Plan my_plan;
bool success = (move_group_interface.plan(my_plan) == moveit::planning_interface::MoveItErrorCode::SUCCESS);
ROS_INFO_NAMED("tutorial", "Planning (pose goal) %s", success ? "OK" : "FAILED");
bool result = (move_group_interface.execute(my_plan) == moveit::planning_interface::MoveItErrorCode::SUCCESS);
ROS_INFO_NAMED("tutorial", "MOVE_GROUP_RESULT : %s", result ? "SUCCESS" : "FAILED");
}

In main is something like this

ros::AsyncSpinner spinner(1);
spinner.start();

...

static const std::string PLANNING_GROUP_ARM = "arm";
moveit::planning_interface::MoveGroupInterface move_group_interface_arm(PLANNING_GROUP_ARM);
move_group_interface_arm.setPlannerId("RRTConnect");
move_group_interface_arm.setPlanningTime(5);
move_group_interface_arm.setNumPlanningAttempts(10);
const moveit::core::JointModelGroup* joint_model_group = move_group_interface_arm.getCurrentState()->getJointModelGroup(PLANNING_GROUP_ARM);

move_arm(move_group_interface_arm, "position1");
move_arm(move_group_interface_arm, "position2");
...

When i add ros::WallDuration(5.0).sleep(); line in between these two move_arm calls, it works, but that is not the right way to do it.

Any idea why this happes or suggestion how to solve this problem?
Is there anything else I should pass to MoveGroupInterface constructor except that planning group name string?

v4hn, 2021-06-04:
> Then i tried to monitor what is going on with /move_group/feedback and /move_group/status topics and it turns out that robot publishes that the arm reached goal state before it actually reached it.

Unless the execution actually returns an error (e.g. because execution tolerances were violated), first and foremost this is a bug in the `FollowJointTrajectory` action server of the arm driver.
It should not succeed before actually *reaching* the last waypoint. In the past some implementations returned after sending the last waypoint as target. Depending on the density of waypoints this can be a huge difference.

We have [a safety net for this in moveit](https://github.com/ros-planning/moveit/blob/master/moveit_ros/planning/trajectory_execution_manager/src/trajectory_execution_manager.cpp#L1302), but maybe it's [disabled in your case](https://github.com/ros-planning/moveit/blob/master/moveit_ros/planning/trajectory_execution_manager/src/trajectory_execution_manager.cpp#L1538)?

slavenpetkovic, 2021-06-04:
I can't find that file "trajectory_execution_manager.cpp", in include folder I just have .h file.

gavanderhoorn, 2021-06-04:
@v4hn only pointed you to the source which implements that check.

users configuration is done through ROS parameters.

See [How do I disable execution_duration_monitoring ?](https://answers.ros.org/question/196586) on ROS Answers.

Note: the question is whether you've disabled that already. It's not a suggestion to disable it.

Fundamentally: if the driver does not report execution progress correctly, there isn't really anything MoveIt can do. The driver is the single source of truth in this case.

v4hn, 2021-06-04:
Thanks for the clarification gijs!

> if the driver does not report execution progress correctly, there isn't really anything MoveIt can do.
> The driver is the single source of truth in this case.

There is nothing MoveIt **should** do, because the driver is supposed to know better. The safety net I mentioned actually looks at the `joint_states` topic (and thus at a different output channel of the driver), but of course this should not be needed. I'm actually puzzled why this additional check fails with the described setup.