moveit/moveit #2714: [Help wanted] Analyzing FCL collision distance performance

AndyZe, 2021-06-03:
As recommended to me in #2698, let's take a look at the performance of the FCL distanceCallback() function. I used [hotspot](https://github.com/KDAB/hotspot) and ran the attached simple benchmarking executable. I'm new to this, so input from others would be great!

[benchmark_collisions.cpp.txt](https://github.com/ros-planning/moveit/files/6593001/benchmark_collisions.cpp.txt)

AndyZe, 2021-06-03:
Here's the flame graph from `master` branch. I don't get a lot from this except almost all of the time is spent in fcl functions. I think that's a good thing because it means MoveIt is not the bottleneck.

[image]

AndyZe, 2021-06-03:
Now a look at cycles spent in each function.

`master` branch:
[image]

PR #2698:
[image]

AndyZe, 2021-06-03:
It looks like there may be a little bit of low-hanging fruit in #2698 if we can find al alternative to `fmin` (0.005% of cycles). Also, allocating a std::pair<string, string> is expensive (0.005% of cycles). I already knew that and spent a few hours trying to find an alternative. If anybody knows how to use `reserve` with a std::pair please let me know.   ;)

xqms, 2021-06-07:
While writing my comment for #2698 I just noticed that the logic in the `GLOBAL` case seems suboptimal:

https://github.com/ros-planning/moveit/blob/bab87f9aa32df22273674fbd007d495d7c455cf1/moveit_core/collision_detection_fcl/src/collision_common.cpp#L496-L522

The issue is that we first look up whether we already have a distance result for that pair of objects, and only if we found one then decrease the distance threshold. That means the first callback for a particular object pair always operates with the original `distance_threshold`, when it could use the minimum distance computed so far (`cdata->res->minimum_distance.distance`).

So I think the `if (cdata->req->type == DistanceRequestType::GLOBAL)` should be moved outside of the containing `if`.

xqms, 2021-06-07:
Along the same lines: we can also modify the `min_dist` parameter passed to `distanceCallback()` to indicate to FCL that the minimum distance it should look for during the broadphase has changed. That might result in fewer `distanceCallback()` calls in total.

And a final idea: we could use multithreading to do the `fcl::distance()` computations on the individual object pairs. That would mean collecting all `distanceCallback()` calls and then using a number of threads to do the `fcl::distance()` calls. I *think* FCL is thread-safe in that regard, but I haven't checked anything. This particular idea has the potential to reduce distance timings drastically on modern multi-core CPUs, but it's definitely more work to implement.

AndyZe, 2021-06-07:
@xqms Bravo, I'm seeing something like a 10X speed improvement when moving GLOBAL out of the containing `if`.

Maybe I shouldn't get too excited yet, though... still need to verify that this works reliably.

AndyZe, 2021-06-07:
Also, do you want to make a PR for this? It's really a significant improvement so I think you should get the credit.

AndyZe, 2021-06-07:
If I do this:

  // Global search: distance_threshold starts at the smallest distance between any pairs found so far
  if (cdata->req->type == DistanceRequestType::GLOBAL)
  {
    dist_threshold = cdata->res->minimum_distance.distance;
  }
  
  I wonder what `cdata->res->minimum_distance.distance` has been initialized to? If it was initialized to `0`, that could cause problems.

xqms, 2021-06-09:
> Also, do you want to make a PR for this? It's really a significant improvement so I think you should get the credit.

> Edit: I pushed the commit https://github.com/ros-planning/moveit/pull/2698/commits. Happy to change it later however you please.

Looks good to me! I'm fine with incorporating it in your PR.