moveit/moveit #2717: GSoC: Creation of a Benchmark Suite for Evaluating MoveIt Performance

captain-yoshi, 2021-06-11:
**mentors: @felixvd @mamoll @v4hn**

This issue will bundle all relevant information about the GSoC project. A Discord channel is also available to discuss on this project. I will try to add pertinent information from Discord to this issue.

## Motivation and goals
This GSoC project goal is to benchmark complete tests throughout the entire MoveIt pipeline and eventually the MoveIt Task Constructor (MTC). This will be done through a benchmark suite. Unit tests will also be benchmarked.
- Compare the impact of changes we consider integrating into MoveIt.
- Find regressions between MoveIt versions.
- Isolated tests or benchmarks don't necessarily reflect the entire MoveIt pipeline.
- Some algorithms do well in a particular environment but fall short in other contexts.

## Planning
#### Phase 1: Exploration and benchmarks (June)
- Create multiple tests with a few scenes (empty scene, solid primitives, meshes, cluttered) with some joint/pose goals, path constraints, FCL and Bullet, planners (OMPL, CHOMP, STOMP) using the `moveit_ros_benchmarks` package.
- Pick and place benchmark using MTC.
- Analyze tests and decide whether we want to use only self contained tests or use and expand the moveit_ros_benchmarks when benchmarking planners throughout the entire MoveIt stack.
- Add an Amazon Pick and Place Challenge (APC) scenario and do benchmarks on it.

#### Phase 2: Creation of the benchmark suite
- Implement the benchmark suite to run tests.
- Aggregate and plot results through Python.
- Add more tests and scenes.

## Documentation
- [Benchmark Tutorial](https://ros-planning.github.io/moveit_tutorials/doc/benchmarking/benchmarking_tutorial.html)
- [Benchmark Visualizer](https://planners-benchmarking.readthedocs.io/en/latest/user_guide/5_visualizing_the_results.html)
- [Planner Arena](http://plannerarena.org/)
## Links
- [Proposal](https://docs.google.com/document/d/10DHBipL6kuHAcFy03jLOLiPuMkW4cKVAcSG4gw6ixBU/edit?usp=sharing)
- GSoC log
- [Discord](https://discord.gg/KzEf7Ysy) #gsoc2021

captain-yoshi, 2021-06-11:
### Meeting notes 09/06/2021

#### Previous week summary
- Reading + tutorials.
- Create the box and block test (BBT) scenes from [YCB benchmarks](https://www.ycbbenchmarks.com/protocols-and-benchmarks/).

<p align="middle">
[image]
  [image]
  
</p>

#### Infrastructure
The benchmark suite should have benchmark tests configurable via a yaml file. The objects in the scenes should also be configurable (ex. which object and where). Debate whether we use the moveit_ros_benchmarks for some of the benchmarks or do it manually.

#### Cartesian planner
Benchmark pre-grasp approach between OMPL and Pilz Industrial planner.

#### Execution time
Would it make sense to benchmark the execution of a plan in demo mode (moveit fake controller) or just parse the trajectory to retrieve the trajectory duration?

#### General remarks
- Keep it simple and iterate.
- After the next meeting, we should have a more clearer planning and timeline of the project.
- It is easier to use MTC to select robot states for benchmarking. Moving the interactive markers in RViz is in my opinion difficult in a cluttered scene.
- BBT scene workspace not very restricted. Use something similar to the Amazon picking challenge problem for next scene. Example from @mamoll. 
<p align="center">
[image]
</p>

#### Next week goals
- Benchmark the BBT scene exhaustively (MoveIt + MTC).
- Create a report with BBT benchmarks as a starting point.
- Create an APC scenario if given enough time.

captain-yoshi, 2021-06-16:
### Meeting notes 16/06/2021
#### Previous week summary
- Benchmark MTC for pick and place in BBT scenes.
- Benchmark planners with constraints in BBT scenes.
- Explored and tested the moveit_ros_benchmarks package (Constraints + Visualization).
- Sadly the report is not finished yet...
- [PR](https://github.com/ros-planning/moveit_task_constructor/pull/275): Implement missing pregrasp property for RobotState.msg in MTC.
- [PR](https://github.com/ros-planning/moveit/pull/2718): Fix predefined poses benchmark example.

#### MoveIt Benchmark Suite
_This section will refer to some elements of the flowchart (see next comment below)._
**TODO add explanation**

#### Next week goals
- Finish the report with BBT benchmarks.
- Create a scene creator node where one can load collision objects from a configurable yaml file.

captain-yoshi, 2021-06-16:
### MoveIt Benchmark Suite Flowchart
<p align="center">
[image]
</p>

captain-yoshi, 2021-06-20:
**MP**
For the Motion Planning pipeline, I think there are only 3 ways to plan a motion:
- `PlanningContext`: solve()
- `PlanningPipeline`: generatePlan()
- `MoveGroupInterface`: plan()

Maybe 5 if you count MoveIt Commander Scripting and the Python MoveGroupInterface.

Is there other functions missing for planning?

felixvd, 2021-06-22:
Off the top of my head, some PRs for which it would be interesting to evaluate the performance effect (feel free to edit this post to add to the list):

- #2127 
- #2037 
- #2402

captain-yoshi, 2021-06-23:
### Meeting notes 23/06/2021
#### Previous week summary
- Benchmark [report](https://docs.google.com/spreadsheets/d/1PJA1e7gnr5tJn2Dz6su4CaZtYuYhJHdBqQYETJJPnms/edit?usp=sharing) (plots WIP).
- Benchmarked OMPL, CHOMP and STOMP planners with constraints in BBT scenes (see report).
- Not enough time to start on the scene creator.
- Lots of debugging.
- [PR](https://github.com/ros-industrial/stomp_ros/pull/35): Fix goal position tolerances from box primitive.
- [PR](https://github.com/ros-planning/moveit_resources/pull/86): Floating virtual joint causing weird behaviour.
- [Issue](https://github.com/ros-industrial/stomp_ros/issues/34): Related to previous PR.
- Issue: Fix header comment in `BenchmarkExecutor.h`.
- Issue:
- Issue:
- Issue:
- Issue:
- Issue:
- Issue:
- Issue:

#### MoveIt Benchmark Suite
_This section will refer to some elements of the [flowchart](https://github.com/ros-planning/moveit/issues/2717#issuecomment-862728937)._
The benchmark suite is divided into five sections: **Configuration**, **Pipeline**, **Metric**, **Agregate/Plot** and **Visualisation**.

captain-yoshi, 2021-06-30:
### Robowflex
Really glad we found this hidden gem called [robowflex](https://github.com/KavrakiLab/robowflex) before the end of GSoC! Essentially it's a benchmark suite and a toolbox (a big wrapper) around MoveIt, Tesseract, Dart and much more. The goals of this library are:
- Provide an easy to use interface to complex robotics libraries by brushing cruft and idiosyncrasies under the rug.
- Provide access to all underlying capabilities of the libraries so users are not hampered by Robowflex in any way.
- Remain untied as possible to the greater ROS system, and support all distros from Kinetic onward.

### Under the hood
Here are my thoughs after playing with it for roughly 10 hours.
- Has very good documentation.
- Good error messages.
- Multiple robots (see [robowflex_resource](https://github.com/KavrakiLab/robowflex_resources)).
- API easy to use and has a lot of examples (script folder).
- IO classes for Rosbag, TF2 Broadcaster, GNUPlot, ROS Paramter Server, Pluginlib, RViz and YAML.
- YAML: Decode and encode for all the essential messages: `geometry_msgs`, `shape_msgs`, `moveit_msgs`, `trajectory_msgs`...
- YAML: Handles the `PlanningScene` Mesh by **package name** or the default vertices soup.
- Live benchmark output to GNUPlot!
- Easy to setup tests programatically.
- Backend for the PlanningPippeline (similary to `moveit_ros_benchmarks`).
- Loads URDF, SRDF, kinematics, limits and planning programatically. This can be loaded from a YAML config.

### What's next?
I think it would make sense to use this library and add GSoC elements to it (STOMP, MTC, Agregate/Plot, Visualization in RViz for debugging trajectories). Of course assuming the author wants these features. So we should have a `moveit_benchmark_suite` that has prepared scenes and benchmarks ready to be executed, unit tests, and meshes for scenes that would depend on robowflex. Well talk about the next steps to take in tomorrows meeting.

@zkingston It would be great to have you're opinion on weather it would make sense to use you're library for benchmarking MoveIt Pipelines and if there are features that you would be interested to have in you're library. Note that this years GSoC is ~17 hours for 10 weeks.

captain-yoshi, 2021-06-30:
### Creating collision objects
Using YAML one can setup an open shelf in a scene pretty easily. The only downside is that YAML dosen't support math arithmetics (maybe spec 1.3 will add it) so we can't divide by 2 for example or concatenate strings. So there will be hardcoded values. Unless with create special tags that permits these arithmetic's.

We can of course create multiple solid primitives objects (table, shelf, chair) that we can call just like meshes, ex. `ressource: "package://ros_pkg/solid_primitive/shelf.yaml"`. For meshes we should probably have a minimum of 2 subframes: one for the centroid and one at the bottom center of the objects to put it easily on a flat surface. With a certain convention for axes.

Here is a quick example for a shelf and a can. This is just an example, and we should respect somewhat the CollisionObject message like [robowflex](https://github.com/KavrakiLab/robowflex/blob/master/robowflex_library/yaml/test.yml) does it.
```yaml
# YAML anchors
.: &root_frame world

.: &post_dimension
  - 0.035
  - 0.035
  - 1.25

.: &shelf_dimension
  - &dim_x 0.6
  - &dim_y 1.5
  - &dim_z 0.02

.: &shelf_gap 0.25

collision_objects:
  # Open shelf (f=front b=back, l=left, r=right)
  - id: post_bl
    type: box_primitive
    frame_id: *root_frame
    position: [-0.3, -0.75, 0.625]
    dimensions: *post_dimension
  - id: post_br
    type: box_primitive
    frame_id: post_bl
    position: [0.0, *dim_y, 0.0]
    dimensions: *post_dimension
  - id: post_fl
    type: box_primitive
    frame_id: post_bl
    position: [*dim_x, 0.0, 0.0]
    dimensions: *post_dimension
  - id: post_fr
    type: box_primitive
    frame_id: post_br
    position: [*dim_x, 0.0, 0.0]
    dimensions: *post_dimension

  - id: shelf_1
    type: box_primitive
    frame_id: *root_frame
    position: [0.0, 0.0, *shelf_gap]
    dimensions: *shelf_dimension
  - id: shelf_2
    type: box_primitive
    frame_id: shelf_1
    position: [0.0, 0.0, *shelf_gap]
    dimensions: *shelf_dimension
  - id: shelf_3
    type: box_primitive
    frame_id: shelf_2
    position: [0.0, 0.0, *shelf_gap]
    dimensions: *shelf_dimension
  - id: shelf_4
    type: box_primitive
    frame_id: shelf_3
    position: [0.0, 0.0, *shelf_gap]
    dimensions: *shelf_dimension
  - id: shelf_5
    type: box_primitive
    frame_id: shelf_4
    position: [0.0, 0.0, *shelf_gap]
    dimensions: *shelf_dimension

  # Can on the middle shelf
  - id: can
    type: mesh
    resource: "package://ros_pkg/mesh/can.stl"
    scaling: [1.0, 1.0, 1.0]
    frame_id: shelf_3
    position: [0.0, 0.0, -0.025]

```

We could easily create templates of Solid Primitives with Xacro and load them with arguments. I don't know if we want to go that way.

zkingston, 2021-06-30:
Glad to hear that you like robowflex! It was originally intended to be a tool for benchmarking and evaluating motion planners so I'm glad to see that you see its utility. I'm definitely open to some of the suggestions that you listed, such as supporting more planners, having stronger RViz visualization for debugging, and more plotting capabilities. However, I'd like to keep robowflex as a separate library from MoveIt proper. If you keep using the robowflex, let me know if you have any questions.

captain-yoshi, 2021-07-07:
### Meeting notes 30/06/2021
#### Previous week summary
- Scene creator using YAML file ([example](https://github.com/ros-planning/moveit/issues/2717#issuecomment-871477121)). Does not support math arithmetics.
- Exploration of robowflex repository (see [review](https://github.com/ros-planning/moveit/issues/2717#issuecomment-871076653))
#### Robowflex
The mentors are reluctant to use this library because of the repository maintenance. So I think we will reuse some of the code but not the library directly. Nonetheless there are a lot of cool features so check it out!
#### Next week goals
- Scene creator that creates MoveIt CollisionObjects from URDF.
- Backend for `PlanningPipeline` and `MoveitGroupInterface`.

captain-yoshi, 2021-07-07:
### URDF Parser for CollisionObjects
This [repo](https://github.com/captain-yoshi/scene_parser) parses collision objects from a URDF to a MoveIt CollisionObject message. By using Xacro one can construct shorter and more readable XML. Normally URDF represents a robot model but this time we use it to represent collision objects. We have access to the _urdf_to_graphiz_ tools which represents the objects in a graph scene and of course the URDF parser that will trigger an error message if the URDF is malformed. 

It's still a work in progress and feedback is welcomed!

Shelf example:

```xml
<robot name="scene" xmlns:xacro="http://www.ros.org/wiki/xacro">
  <xacro:include filename="$(find scene_parser)/config/utils.xacro" />
  <xacro:include filename="$(find scene_parser)/config/shelf.urdf.xacro" />

  <link name="world" />

  <!-- Create default shelf -->
  <xacro:shelf name="book_shelf"
               frame_id="world"
               xyz="0.8 0 0"/>

  <!-- Add cube on book-shelf level-1 -->
  <xacro:framebox name="box"
                  frame_id="book_shelf/level1"
                  xyz="0 0 0.005"
                  dimensions="0.03 0.03 0.03"/>

  <!-- Add can on book-shelf level-2 -->
  <xacro:framemesh name="can"
                   frame_id="book_shelf/level2"
                   scale="0.01 0.01 0.01"
                   resource="package://scene_parser/mesh/can.stl"/>
</robot>
```
<p align="middle">
[image]
  [image]
</p>

captain-yoshi, 2021-07-10:
### Meeting notes 07/07/2021
#### Previous week summary
- [Parser](https://github.com/ros-planning/moveit/issues/2717#issuecomment-875250058) for creating MoveIt CollisionObjects from URDF.
- Semi vacation!
#### MoveIt Benchmark Suite
I was struggling with the benchmark suite goal compared to other existing solutions like _robowflex_ and _moveit_ros_benchmarks_ on weather we create our own package or use and extend an existing one. Thanks to this week meeting, I have a better understanding of their goals and what they/will offer. The MP back-end will take inspiration from  _moveit_ros_benchmarks_ and _robowflex_. There is also this [benchmark project](https://github.com/JeroenDM/benchmark_ros_ws) from Jeroen which could be useful.

**_moveit_benchmark_suite_**: Benchmark suite that runs throughout the entire MoveIt pipeline.

- Adding/updating the planning scene
- MP: PLanningPipeline and MoveGroupInterface
- CMP: MTC and MoveGroupInterfacePickPlace
- MoveIt servo

**_moveit_ros_benchmarks_**: Motion planning benchmark for MoveIt planners (OMPL, CHOM, STOMP) using the `PipelinePlanner`. There are utilities to add scenes, constraints and queries into the database (warehouse_ros). It can be easily extended to change the default behaviour (storage, benchmark parameters).

**_robowflex_**: Wrapper and conveniance tool for benchmarking motion planning with multiple modules like MoveIt, Dart, Tesseract ...

#### Next week goals
- Backend for `MoveitGroupInterface` (MGI) example.
- Show preliminary numbers from the bbt scene with the MGI backend.
- Try Jeroen benchmark.

captain-yoshi, 2021-07-21:
### Meeting notes 07/14/2021
#### Previous week summary
- Backend for the `MoveGroupInterface` and the `PlanningPipeline`. 
- Strip down roboflex_library code.
- Preliminary numbers from the bbt scene with the MGI backend.
- Overhaul of the scene parser. Correctly parses robot URDF as collision objects.
#### Preliminary numbers
Did a quick benchmark between MGI and PP using RRT and STOMP with the BBT primitive scene and a joint constraint goal query. The PlanningPipeline is ~4X times faster then the MoveGroupInterface.
<p align="center">
[image]
</p>

#### Architecture
The benchmark suite uses the same architecture from [robowflex benchmark](https://github.com/KavrakiLab/robowflex/blob/master/.docs/markdown/benchmarking.md). I think it has a great workflow that combines simplicity and flexibility! The structure could be reused for other pipelines like MTC.

#### Next week goals
- Backend for MTC pick and place.

captain-yoshi, 2021-07-21:
### Meeting notes 21/07/2021
#### Previous week summary
- [MoveIt Benchmark Suite](https://github.com/captain-yoshi/moveit_benchmark_suite) repository.
- Simplify and unify `PlanningPipeline` and `MoveGroupInterface`. This makes it possible to benchmark both motion planning. 
- The MoveGroupInterface is now benchmarked against the real plan method (as opposed to being simulated).
- Added a bar graph in GNUPlot like the shadow_robot company did for success and correct metric.
- You can select the plot type for each metrics independently. Real time or at the end of the benchmark.
- Added ability to have multiple plots in the same GNUPlot window.
- Sorry for diverging from last week goals, but I think it was necessary to have a better motion planning then last week. Like @felixvd said: Small, finished work units are better than large, unfinished ones!

#### Motion planning
Example for plotting different metrics for multiple queries.
<p align="center">
[image]
</p>

#### General remarks
- Load scenes and queries in the launch file.
- Confusion around the word `backend`. A backend is simply a Unit Test. Some Unit Tests are simple and some are more complex (ex. motion planning).

#### Next week goals
- Load and use the benchmark options file for multiple scenes, collision detectors.
- Create open shelf and table-top scene with either few, simple objects (easy), or many complex objects (difficult) near/close the start/end goal.
- PlanningScene loading.
- Add the simple collision unit test.

captain-yoshi, 2021-07-28:
### Meeting notes 28/07/2021
#### Previous week summary
- Object mesh metadata [package](https://github.com/captain-yoshi/moveit_benchmark_suite/tree/master/resources)
- Load and use the benchmark options file for multiple scenes, collision detectors.
- Update SceneParser to use metadata for meshes
- Rework shelf xacro to include vertical boards + extras
- Add collision plugin
- PR: https://github.com/ros-planning/moveit/pull/2783

#### Next week goals
- Complete scene parser
- Fix plotting format when there are multiples scenes, interfaces and collision detectors.
- Show raw benchmark numbers + CPU info + Moveit version and commit ...
- Compile all the results (all the different tests) into one file that summarizes MoveIt's performance in a given state.
- Add the simple collision unit test (maybe add a config)
- Add more scenes (labware)