moveit/moveit #2719: Padding on Attached Objects

apupa, 2021-06-16:
### Description

Hi everybody,

I am currently working on a pick and place application and I am setting the padding in order to be sure that the planned trajectory is far from the obstacles.
I have been able to set the padding for all the links, however this parameter is not considered for the attached objects.

### Steps to reproduce
Set the robot padding, attach an object to the end effector and plan to a final position which is very near to an obstacle (distance lower than the padding).

### Expected behaviour
The planner should not be able to find a solution, since the robot must not be allowed to get so close to the obstacle.

### Actual behaviour
The robot reach the position avoiding the obstacle, but it is not compliant with the desired padding. 

### Code
I think this is because the CollisionRobotFCL class does not consider the scale and the padding when creating the collision geometry ([here](https://github.com/ros-planning/moveit/blob/dc5b4102a87de91fa41b2806db6655a3ec44c541/moveit_core/collision_detection_fcl/src/collision_robot_fcl.cpp#L88)).
Is this a bug?

Maybe the line could be replaced with this:
`FCLGeometryConstPtr co = createCollisionGeometry(shapes[i], getLinkScale(ab->getAttachedLinkName()), getLinkPadding(ab->getAttachedLinkName()), ab, i);`

This would lead the padding and scale to be exactly the same as the one of the link the object is attached to.

### Environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04
* Source: melodic-devel

v4hn, 2021-06-16:
It looks like this might be a bug and you did a really good job tracking it down to this line. :+1: 
I expect the updated line works for your use case?

Could you file a pull-request for this change?

apupa, 2021-06-16:
> It looks like this might be a bug and you did a really good job tracking it down to this line. +1
> I expect the updated line works for your use case?
> 
> Could you file a pull-request for this change?

Yes, with this change everything works as expected for my case.
I was not sure it was a bug. Now I will do the pull-request.

Thank you very much for the quick reply.

felixvd, 2021-08-16:
Fixed by #2721

closed 2021-08-16 by felixvd