moveit/moveit #2723: Using OMPL plugin

lepalom, 2021-06-17:
I would like to use the OMPL plugin, specifically to use this [code](https://github.com/GabrielKoenig/moveit_visualize_data). This code was developed patching moveit, melodic [version](https://github.com/GabrielKoenig/moveit/commit/fe09ffeb56dc73a41618b3daaf0c277442191682) .

Now, I'm trying to use it in noetic (or master) and I'm blocked and I would like to ask if you could guide me. The main problem is that I would need to obtain `ompl_interface::ModelBasedPlanningContextPtr` to use it. Following [this](https://github.com/ros-planning/moveit/pull/1428), I have understood why that functions have been deprecated but I have not found migration or alternative option.

There are several lines of [code](https://github.com/ros-planning/moveit/blob/master/moveit_planners/ompl/ompl_interface/src/ompl_planner_manager.cpp) in master that use that functionality and obviously are commented. So, my question is:

if I have a` planning_scene::PlanningSceneConstPtr&,` and a planning_group, how can I take the `ModelBasedPlanningContextPtr` to call the method `getOMPLSimpleSetup()->getSpaceInformation()` ?

Because, moveit had some functionality in melodic than now in noetic is missing or please could you explain how to upgrade this kind of code?

felixvd, 2021-08-07:
Maybe @JeroenDM @mamoll have an idea

v4hn, 2021-08-08:
> Because, moveit had some functionality in melodic than now in noetic is missing or please could you explain how to upgrade this kind of code?

There is no way to upgrade this functionality, it was just removed and the commented out code you refer to was commented out for a long time.
For your use case it makes a lot of sense to allow for it in some way again though and I would appreciate a patch.

> if I have a` planning_scene::PlanningSceneConstPtr&,` and a planning_group, how can I take the `ModelBasedPlanningContextPtr` to call the method `getOMPLSimpleSetup()->getSpaceInformation()` ?

Without looking into this any further, I would not expect that it is possible without modification of some API(s).