moveit/moveit #2724: How to properly link to moveit libraries so that local builds do not break after binary upgrade?

peci1, 2021-06-17:
### Description

Hi, I'm the developer of robot_body_filter, a package that links to `libmoveit_point_containment_filter.so` via the most simple and standard catkin commands (`find_package catkin REQUIRED COMPONENTS ... moveit_ros_perception ...` and `target_link_libraries(${PROJECT_NAME} ... ${catkin_LIBRARIES} ...)`).

Every time moveit is updated, my local build of robot_body_filter breaks because it does not link to the `libmoveit_point_containment_filter.so` development symlink, but to the release library `libmoveit_point_containment_filter.so.1.0.7`. When moveit got upgraded to 1.0.8, the 1.0.7 file ceases to exist and my library has broken linkage.

What should I do to keep the nice general ROS behavior that binary updates do not break ABI?

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04
* Source or Binary build? Binary
* If binary, which release version? 1.0.8

rhaschke, 2021-06-18:
We implicitly assume that MoveIt updates might break ABI (that's an experience from the past). For this reason, we explicitly increase the SOVERSION of all MoveIt libraries, explicitly enforcing people to rebuild. So, that's desired behaviour.
Without that, you would get weird crashes if ABI actually has changed, but downstream binaries like your `robot_body_filter` haven't been recompiled yet.

closed 2021-06-18 by rhaschke

gavanderhoorn, 2021-06-18:
To add to that:

> What should I do to keep the nice general ROS behavior that binary updates do not break ABI?

you must be aware @peci1 that ROS (1 at least) does not actually have nice behaviour in that it doesn't break ABI. ABI is broken all the time.

That's exactly the reason rebuilding your workspace after every `apt` update is (almost) a necessity, if not best practice to avoid undetected/able ABI incompatibilities.

MoveIt using sonumber increases to force people to do that is annoying, but at least makes it explicit.

peci1, 2021-06-18:
Thank for the explanation. I understand the reasons. What makes me still a bit uncomfortable is that just running `catkin_make` doesn't actually help and I have to manually clear CMake cache or delete the whole build directory to fix the linking (I assume that's because CMake caches the name of the library and it doesn't care the library ceased existing). Isn't there a way to at least delete the path from the cache automatically after the update to allow simpler rebuilds?

gavanderhoorn, 2021-06-18:
`apt` doesn't do anything with CMake, so I'm not sure I see a way to do that.

How would the package know where your workspace/`build`/`devel` are?

I believe the easiest way to "guarantee" ABI compatibility would be to release your package and only use released versions. The ROS buildfarm will rebuild all dependents after a dependency has seen a new release. Exactly to avoid ABI problems.

peci1, 2021-06-18:
The package is release, but of course as a developer I need to have a local workspace with the package built from source.

I don't expect apt to discover and invalidate workspaces. It could be more in line of some CMake find-script magic. What could e.g. happen:

1. I update apt packages
1. I call `catkin_make` in my workspace
2. it starts building robot_body_filter
3. in CMakeLists.txt, it reaches line `find_package(... moveit_ros_perception ...)`
4. The find script from moveit reads some cache variable, compares it to the version of the currently installed library, and if it doesn't match, it invalidates all cached paths of `moveit_ros_perception` and goes through standard first-time-like execution path

Does it seem silly to want this kind of behavior?

gavanderhoorn, 2021-06-18:
Ah, so that's all CMake. CMake caches that information.

We could try to break out of CMake's caching, but I don't feel that would be a good idea.

On the build tool side, Catkin does have a `--force-cmake` argument (and so does `catkin_tools` and Colcon), but I'm not sure that would help, as it would still probably lead to CMake figuring out it can just return cached information.