moveit/moveit #2728: melodic: 'param' unexpected in 'include' in move_group.launch

gavanderhoorn, 2021-06-21:
### Description

`roslaunch` complains about a `param` element inside an `include` element in `move_group.launch` with every MoveIt configuration package generated by the MSA on Melodic.

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04
* binary, `apt`: `1.0.8`

### Steps to reproduce

Generate a MoveIt config pkg using the current MSA on Melodic.

It'll include the following `move_group.launch`:

https://github.com/ros-planning/moveit/blob/dc5b4102a87de91fa41b2806db6655a3ec44c541/moveit_setup_assistant/templates/moveit_config_pkg_template/launch/move_group.launch#L48-L52

gavanderhoorn, 2021-06-21:
Reason I first open an issue is because this was introduced in #1893 introduced in `master` ([here](https://github.com/ros-planning/moveit/commit/cf218879dbc23aadf88dd56b8abe7970b7d61030#diff-c9666b8ddf7fb2b57118595db82fca3bb31b167fb07876b124ca6a6f94ba33df)), and then got overwritten by #2127 ([here](https://github.com/ros-planning/moveit/commit/b0a4173c72a9376b1b3a8c63e9bfcc26626184eb#diff-c9666b8ddf7fb2b57118595db82fca3bb31b167fb07876b124ca6a6f94ba33df)). As Noetic basically gets `master`, it's not a problem there.

#2127 moves those two `param`s back to the lines later on in `move_group.launch` (as children of `move_group` the `node`), which makes more sense to me.

Perhaps there was a reason they got moved to be `arg`s on the `include`d `planning_pipeline.launch.xml`, which I'm not aware of.

As `melodic-devel` doesn't have #2127, moving them back could lead to (further) regressions?

v4hn, 2021-06-21:
crap.

I guess this is related to the pilz_planner providing custom capabilities (namely [the `Sequence` capabilities](https://github.com/ros-planning/moveit/blob/master/moveit_planners/pilz_industrial_motion_planner/plugins/sequence_capability_plugin_description.xml).

However, these are now handled [in the `move_group` node now by looking for additional `capabilities` in the pipeline namespaces](https://github.com/ros-planning/moveit/blob/master/moveit_ros/move_group/src/move_group.cpp#L141-L152), so there is no reason to have the args with the planning pipeline in any way and they should be below with the `move_group` node.

Could you create a pull-request please?

gavanderhoorn, 2021-06-22:
I'm happy to move them back, but I don't see the same code in Melodic @v4hn:

https://github.com/ros-planning/moveit/blob/dc5b4102a87de91fa41b2806db6655a3ec44c541/moveit_ros/move_group/src/move_group.cpp#L110-L179

I believe that was also introduced in #2127, which Melodic doesn't have.

v4hn, 2021-06-22:
You are right, moving the parameters down is not enough to solve the issue, though I would be happier. :-)

The pilz planner PR moved the responsibility to set the capabilities parameter into the `*_planning_pipeline.launch.xml` and that's the current situation in melodic now. I don't like that at all, but I also did not review the original PR, so I can hardly complain...
Changing it once again is also unnecessary breakage. Users have to regenerate their configurations for noetic anyway and it's different there.
So I would say, let's turn the broken `<param>` tags you found into `<arg>`s ([which they were intended to be](https://github.com/ros-planning/moveit/blob/melodic-devel/moveit_setup_assistant/templates/moveit_config_pkg_template/launch/planning_pipeline.launch.xml#L9-L12)) and keep the logic that sets the capabilities parameter in melodic.

Does that work for you?

gavanderhoorn, 2021-06-22:
Yeah, I already did that locally to make things work again, so I can certainly PR that.

Things like this make me wonder whether any of the (active) maintainers actually use the MSA. It's not difficult to spot this, `roslaunch` is pretty vocal about it -- not to mention the capability management doesn't work correctly like this.

v4hn, 2021-06-22:
> Things like this make me wonder whether any of the (active) maintainers actually use the MSA.

Obviously you do. Last time I checked you were a maintainer and you are always active in comments. ;)
On a more serious note, the nature of the MSA is such that you won't use it in your everyday workflow.
For my own projects I use it something like twice a year, though I also get feedback at times from colleagues who use it.
Automated tests for the MSA (even just for the backend and without writing GUI tests) would help a lot with this, but somebody would have to write them.

> It's not difficult to spot this, `roslaunch` is pretty vocal about it

So thanks for bringing it up and counteracting [the Bystander effect](https://psycnet.apa.org/record/1968-08862-001) of FOSS projects!

felixvd, 2021-06-23:
> On a more serious note, the nature of the MSA is such that you won't use it in your everyday workflow. For my own projects I use it something like twice a year.

Same for me. It's easy to miss regressions like this in MSA, especially on legacy branches.

gavanderhoorn, 2021-06-23:
> It's easy to miss regressions like this in MSA, especially on legacy branches.

the regression was actually on `master`. It just got fixed almost accidentally by an unrelated follow-up PR.

gavanderhoorn, 2021-06-23:
Closing as #2729 for merged.

commit 5accdde3 (2021-06-22): msa: capabilities-related args are args in template (fix #2728) (#2729)

These two lines were moved here (from the `move_group` node section below) in #2507 as part of the Melodic backport of the Pilz planner, but were not changed to `arg`s.

`include` elements do not support `param` children.

Change these to `arg`s to make `roslaunch` pass them on to `planning_pipeline.launch.xml` as intended.

Note: this is not a problem on `master` (and consequently `noetic-devel`), as #2127 rearranged things again (to move capabilities management to the per-pipeline launch files) and fixed it.
closed 2021-06-23 by gavanderhoorn