moveit/moveit #2733: [noetic] Descartes Capability - Symbol lookup error with moveit 1.1.2 or newer, moveit 1.1.0 works fine

Hawxxer, 2021-06-23:
### Description

I try using descartes_capability with a moveit versionen newer than 1.1.1. While Moveit 1.1.1 works fine under ros noetic, since Moveit 1.1.2 I always get a symbol lookup error. I also opened a Issue at descartes_capability https://github.com/PickNikRobotics/descartes_capability/issues/11 but I can't really tell if it's a problem with moveit or descartes capability.

### Your environment
* ROS Distro: Noetic
* OS Version: Ubuntu 20.04.2
* 1.1.2, 1.1.0 from source, 1.1.5 binary 

### Steps to reproduce
start gazebo
start moveit
set new robot position
enable cartesian path
plan & execute

### Expected behaviour
descartes starts calculating new path

### Actual behaviour
moveit gets stuck

### Backtrace or Console output

[image]

´´´
[ INFO] [1624397880.308366634, 863.551000000]: MoveGroup context using planning plugin ompl_interface/OMPLPlanner
[ INFO] [1624397880.308388435, 863.551000000]: MoveGroup context initialization complete

You can start planning now!

[ INFO] [1624397881.319000126, 864.562000000]: Ready to take commands for planning group helene_arm.
[ INFO] [1624397881.319065229, 864.562000000]: Looking around: no
[ INFO] [1624397881.319090266, 864.562000000]: Replanning: no
[ INFO] [1624397885.824363451, 869.064000000]: Received request to compute Descartes path
/home/eric/rviz/moveit3/devel/lib/moveit_ros_move_group/move_group: symbol lookup error: /home/eric/habc/devel/lib/libdescartes_capability.so: undefined symbol: _ZN2tf14poseMsgToEigenERKN13geometry_msgs5Pose_ISaIvEEERN5Eigen9TransformIdLi3ELi1ELi0EEE
[move_group-3] process has died [pid 88747, exit code 127, cmd /home/eric/rviz/moveit3/devel/lib/moveit_ros_move_group/move_group --debug __name:=move_group __log:=/home/eric/.ros/log/190b5764-d3a0-11eb-b24b-335eae1200a5/move_group-3.log].
log file: /home/eric/.ros/log/190b5764-d3a0-11eb-b24b-335eae1200a5/move_group-3*.log
[ INFO] [1624397891.555577313, 874.791000000]: Stopping planning scene monitor
[rviz_eric_88728_9101102882021071447-4] process has finished cleanly
´´´

felixvd, 2021-06-23:
Did you try to clean and rebuild your workspace?

Hawxxer, 2021-06-23:
> Did you try to clean and rebuild your workspace?

Yes I always do
catkin clean
catkin build

JafarAbdi, 2021-06-23:
This's a bug in [descartes_capability](https://github.com/PickNikRobotics/descartes_capability) not moveit, this [commit](https://github.com/ros-planning/moveit/commit/b7711c134a8a1e173c9bf956f68ad8bbe27796c4) removed the dependency on `tf` but descartes_capability doesn't find/export it explicitly which should be the case (see [CMakeLists.txt](https://github.com/PickNikRobotics/descartes_capability/blob/master/CMakeLists.txt))

Do you mind adding `tf_eigen` to these lines [1](https://github.com/PickNikRobotics/descartes_capability/blob/master/CMakeLists.txt#L14) [2](https://github.com/PickNikRobotics/descartes_capability/blob/master/CMakeLists.txt#L36) [3](https://github.com/PickNikRobotics/descartes_capability/blob/master/package.xml) and see if it fixes the error you're seeing .?

Hawxxer, 2021-06-23:
Do you mean `eigen_conversions` ? I think there is no package `tf_eigen`. If yes, that works thank you very much!

JafarAbdi, 2021-06-23:
Sorry, yes, I think it was called `eigen_conversions` not `tf_eigen`

JafarAbdi, 2021-06-23:
Could you please open a PR the fixes it upstream .? thanks!

Hawxxer, 2021-06-23:
Will do, I will close the issue here.

closed 2021-06-23 by Hawxxer