moveit/moveit #2734: Enable AtlasStateSpace in OMPL Constrained planning interface

gautz, 2021-06-23:
The current PR for OMPL Constrained planning only allows to use ProjectedStateSpace for constrained planning
https://github.com/ros-planning/moveit/pull/2273

I would like to enable constrained planning with AtlasStateSpace or TangentBundleStateSpace. For this we need to:
- Add a parameter
- set the start and goal (joint) state from which the atlas will expand (anchorChart()).

I am able to set the start state in  moveit_planners/ompl/ompl_interface/src/model_based_planning_context.cpp by getting the robot current state:
- ``spec_.state_space_->copyToOMPLState(ompl_start_state.get(), getCompleteInitialRobotState());``
and setting it with anchorChart():
- ``spec_.constrained_state_space_->as<ob::AtlasStateSpace>()->anchorChart(ompl_start_state.get());``

But the goal state at this point is available only as a Cartesian pose. 
**Where in the OMPL-MoveIt interface can I get the goal state in Joint space and pass it to OMPL using anchorChart() ?**

see here for some WIP: https://github.com/gautz/moveit/commit/859164105e276cded2189f7aa00c58593e0784eb

JeroenDM, 2021-06-23:
Could you post the error message you get when you try planning without anchoring the goal state explicitly?

I think, in principle, this should be possible, but maybe it is not supported somewhere in the code base of OMPL or the MoveIt interface. The error message could give a pointer to where you need to look.

gautz, 2021-06-25:
Here is the error message, which makes sense, as I do not anchor the goal state yet:
```
[ INFO] [1624631784.419919369]: trying to anchor start states for constrained atlas state space.
[ERROR] [1624631784.423098131]: ompl::base::AtlasStateSpace::newChart(): Failed because manifold looks degenerate here.
[ERROR] [1624631784.423494292]: OMPL encountered an error: ompl::base::AtlasStateSpace::anchorChart(): Initial chart creation failed. Cannot proceed.
[ WARN] [1624631784.429139789]: Goal sampling thread never did any work. Space information not set up.
```

Btw I am not sure if ``spec_.constrained_state_space_->as<ob::AtlasStateSpace>()->anchorChart(ompl_start_state.get());`` is actually passing something.
I opened a PR on my fork to show what I tried: https://github.com/gautz/moveit/pull/2/files

Maybe @mamoll would have an idea on this?

gautz, 2021-07-05:
@zkingston would you have some idea on how/where to anchorState() in the MoveIt OMPL interface ?

zkingston, 2021-07-05:
Here are some quick thoughts: I haven't had a chance to look closely at your branch, but I would first try to anchor all the goals generated by the goal sampler that are not already controlled by a chart inside the sampler itself, assuming that the goal sampler is not dependent on the atlas. You might want to use something like a projection-based sampler as your goal sampler, since that would not be dependent on the atlas. Anchoring charts is required before planning to give an initial set of charts for the planner to start from, so if you have a planner that can grow only given a start state, only the start needs to be anchored, and if you have a bidirectional planner, then a chart at the goal is also needed.

However, I don't think that the atlas-based state space is thread safe with respect to chart creation so there will be issues since MoveIt uses OMPL's GoalLazySampler, which generates goal configurations asynchronously. There will probably need to be some mutexes to the atlas state space that lock chart creation and access for things to work properly if you are anchoring charts during planning.