moveit/moveit #2738: Windows maintainer?

tylerjw, 2021-06-24:
As several people have been graciously adding support for Windows there has been some frustration as PRs have gone stale or taken very long to merge that add support for building on Windows (and Mac).

One of the primary risks I see with adding Windows support is none of the current maintainers use these platforms in their normal development.  This means that in order to test issues that affect building on these other platforms it is a ton of work.  Even if we merge these various PRs there will inevitably be an issue in the future where some change from one of us will break Windows building.

Many potential users would be really excited about Windows support and will run into issues when they try to use moveit on windows and the current maintainer team will have a hard time helping these new users with their issues.

Thirdly as we add support for these platforms it makes sense to add CI configs for testing on these platforms.  This is yet another area where we will inevitably break this CI and will need help from someone with a good understanding of Windows especially.

I think the best path for Windows support would be if someone would volunteer to be a maintainer as our windows expert.  The primary things you'd be signing up for is this:

* Try to help new users who file issues who are trying to build on windows
* Contribute fixes (and tutorials) to keep the repos for Windows support
* Review fixes related to Windows support
* Attend maintainer meetings
* Help educate the rest of the maintainer team on the best practices for Windows support in the code
* Help other maintainers when they break the Windows CI

@lilustga thoughts?

@davetcoleman @henningkayser @rhaschke @v4hn thoughts?

rhaschke, 2021-06-25:
I very much agree with you, Tyler. Without a specific Windows-affine maintainer, it is hard to judge the open PRs and to maintain the reached state.

gavanderhoorn, 2021-06-25:
> a [..] Windows-affine maintainer

I like this :)

v4hn, 2021-06-25:
Yes please. Up to now we always said we will accept patches for cross-platform support if they look reasonable and don't break usage on linux. As none of the full maintainers are interested in these PRs though they tend to stale.. If anyone is interested in more than that, they should consider this job ad :sunglasses: 

Tagging @Tobias-Fischer and @wolfv as they initiated #2604 and offered to maintain the proposed CI job.

wolfv, 2021-06-26:
Hi yeah, I'd be happy to help out with CI or questions whenever needed for MoveIt or other related ROS projects. I've got quite a bit of Windows experience through conda-forge and our other cross-platform C++ projects. I also have a lot on my plate though, so not sure if I could fill in the role of a maintainer of MoveIt.

However, you can see some success story here on RoboStack + Windows + MoveIt: https://github.com/compas-dev/compas_fab/issues/85#issuecomment-809443514

Upstreaming our patches + enabling CI would really, really help the RoboStack project :) And wouldn't put a crazy burden or anything on the MoveIt project.

wolfv, 2021-06-26:
Also regarding Windows specific issues by new users: I think it would be completely fine for you to ignore them or close them immediately and say that there is "no official support" for Windows.
I don't think that merging a PR here and there that improves Windows support to a level where it "compiles" is really signalling strong Windows support from your end. But as mentioned before, it would really help the RoboStack effort. You could always refer users with Windows specific problems to our repository issue tracker, where we already have some open and closed issues for Windows & MacOS problems: https://github.com/RoboStack/ros-noetic/issues?q=is%3Aissue+is%3Aopen+sort%3Aupdated-desc

traversaro, 2021-06-26:
I am afraid I am not a regular user of MoveIt! enough to volunteer as a mantainer, however I guess @ooeygui (from Microsoft) may be interested in this discussion as well (and unless I am wrong I did not see him mentioned).

traversaro, 2021-06-26:
By the way, before/until a Windows mantainer is found, if the issue is not giving the impression to the users that MoveIt! supports Windows, perhaps it could make sense to add a notice/explanation/redirect to other issue tracked on this to https://moveit.ros.org/moveit/ros/microsoft/windows/2020/07/14/moveit-on-windows.html or https://moveit.ros.org/install/ ? For me, these are the first result that can be found by searching "Windows MoveIt"  on Google, and coming from the official MoveIt website, it may be giving false impressions.

traversaro, 2021-06-26:
I guess also @Ace314159 may be interested in this discussion as he/she has been working on several Windows-related PRs.

Tobias-Fischer, 2021-06-26:
Hiya - thanks for keeping us in the loop @v4hn! Unfortunately I don't have the capacity to become a MoveIt maintainer. RoboStack is keeping me busy enough ;). However, as mentioned in #2604, if there are things coming up related to MoveIt+Cross-CI, I'll be happily looking into those. Also fully agree with @wolfv's and @traversaro's comments.

ooeygui, 2021-06-28:
@traversaro Thank you for looping me into the conversation. I really appreciate all of the work the MoveIt maintainers and contributors are putting into these components. I also appreciate the work the RoboStack team is putting into CI & distribution.

I'm not able to commit to being a maintainer at the moment.  However, Lior and I have surfaced this ask internally. Due to internal timing, I will come back to this thread in August.