moveit/moveit #2739: Servo server JointTrajectory output incorrect when `use_gazebo` is true

turniphead, 2021-06-25:
### Description

Servo server trajectory output is incorrect when `use_gazebo` is set to true

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04
* Source or Binary build? binary
* If binary, which release version? 
* If source, which branch?

### Steps to reproduce
Launch servo server, configure `use_gazebo: true` and `command_out_type: trajectory_msgs/JointTrajectory`. 

### Expected behaviour
Output Trajectory should be useable with a ros controller

### Actual behaviour
I get this error from the controller: `Trajectory message contains waypoints that are not strictly increasing in time.`. Specifically, the second point of the trajectory is completely empty.

### Backtrace or Console output

console output: https://gist.github.com/turniphead/fe2d375848cddcea69ab2e02ec091bae
trajectory message output from servo server: https://gist.github.com/turniphead/046a5ec9282820ac4013aab71cc1e5a7

luisrayas3, 2021-06-25:
Suggested fix: https://github.com/ros-planning/moveit/pull/2740

JafarAbdi, 2021-06-26:
Fıxed by https://github.com/ros-planning/moveit/pull/2740

closed 2021-06-26 by JafarAbdi