moveit/moveit #2741: A list of minor issues

v4hn, 2021-06-26:
[Robert encouraged me](https://github.com/ros-planning/moveit/pull/2737#issuecomment-868504325) to post my list of minor issues on github and I don't see why I should keep it to myself. ;)

These are some minor issues I found while making a short screencast of the RViz GUI.
No guarantees on some of those as I use a somewhat wild mix of dependency versions that might induce some of them.
If you want to discuss any of them please create a new issue. Otherwise feel free to write patches to your heart's content :flying_saucer: 

- Planning an RViz trajectory with the PTP planner through an obstacle does not visibly indicate a failure (though plan validation fails) in the gui
- collision markers (red spheres published on a separate topic) do not vanish until new collisions happen
  It might make sense to publish an empty message to the topic on successful planning?
- `roslaunch moveit_resources_panda_moveit_config demo.launch`, add a `MotionPlanningDisplay`, save the config, quit, relaunch
  -> the Motion Planning panel is disabled, though the display is enabled. toggling the display makes the panel appear
- in demo mode the fake joint publisher can end up publishing no current robot states to rviz anymore during and after executing a trajectory - disabling+enabling the display shows the new robot state though! I have no idea how to reproduce this.
- pilz planner needs PTP as default (in RViz)
  There is currently no default there and users are required to selected one because planning fails without specifying one!
  Instead let's just default to PTP here
- RViz configs in MSA templates and the resources still use `XYOrbit` view
  This is totally unusable and `Orbit` view with the center at the center of the used workspace of the robot should be preferred.
- pilz should only complain about Multi-DOF joints when such a joint should actually be *planned for* (only shows before https://github.com/ros-planning/moveit_resources/pull/86)
- If you scroll through a planned trajectory with the slider (autoplay paused), and afterwards move the marker for the goal state again the robot state showing the trajectory waypoint will still be there and is relatively hard to get rid of. Moving the slider to the end/and or resuming play in the trajectory panel should probably disable the state to cleanup the screen when the user moved on.

Already done/PR created:

- geometric::TrajOpt is not implemented but in panda_moveit_config https://github.com/ros-planning/moveit_resources/pull/85
- on replay trajectory panel says "Play" even when the button stops #2737

rhaschke, 2021-06-28:
Why is the `XYOrbit` view "totally unusable"? I think that's the default in rviz.

v4hn, 2021-06-28:
> Why is the `XYOrbit` view "totally unusable"? I think that's the default in rviz.

Yes it is and it's a great choice for mobile robots. I'm not proposing to change it outside MoveIt.
For robotic manipulation though most people I know end up looking at the robot from the side instead of the top.
If you try to move the camera from a sideways view you end up a long way away from the real robot because of the 2d projection.
Also you can't zoom in on the end-effector which is usually the relevant part of the scene.

Of course this is less of a problem if you choose your ground plane to be a table surface, but it's still a bother.

felixvd, 2021-06-29:
I agree that `Orbit` view should be preferred over `XYOrbit` for the default configs.

jaskiratsingh2000, 2021-07-17:
@felixvd @rhaschke I am beginner starting with the code contribution and super interested to contribute to this repo. So please please redirect me to beginner-friendly issues.
I am looking forward to hearing from you.
Thanks

rhaschke, 2021-07-17:
@jaskiratsingh2000, good first-timer issues are labeled with [good first issue](https://github.com/ros-planning/moveit/labels/good%20first%20issue). From the above list, I suggest the following items:
6. The default planner is set [here](https://github.com/ros-planning/moveit/blob/5f486dd74adb867443393ee3afd36fa476b6c57c/moveit_ros/visualization/motion_planning_rviz_plugin/src/motion_planning_frame_planning.cpp#L380) called from [here](https://github.com/ros-planning/moveit/blob/53dacd10841dc093b97185ee7f011f3e1b6ecdc9/moveit_ros/visualization/motion_planning_rviz_plugin/src/motion_planning_frame.cpp#L402-L403)
7. Manually adapt the [rviz config in templates](https://github.com/ros-planning/moveit/blob/902b98a7aabd6fff49d933f0feb9934305e3f5b0/moveit_setup_assistant/templates/moveit_config_pkg_template/launch/moveit.rviz) based on some _real_ rviz config file.
11. Adapt the [`ProgressBarEditor`](https://github.com/ros-planning/moveit/blob/ef085558818e7e4fbc97e41f8dd12a1a2d6dbfa0/moveit_ros/visualization/motion_planning_rviz_plugin/include/moveit/motion_planning_rviz_plugin/motion_planning_frame_joints_widget.h#L190) to also show units.
10. Allow to drive currently selected joint with left / right keys. To this end, either augment the `JointsWidgetEventFilter` of the `QTreeView` or extend the widget itself (the former is probably easier).

AbhiramL, 2022-01-21:
Was issue 11 for progress editor bar fixed? If not can I help out, Im a beginner in github but I know a good amount of c++

rhaschke, 2022-01-22:
The original issue 11 _In the joints tab "Nullspace exploration:" is visible even when there is no slider below_ was fixed via #2944 as indicated by the PR id next to this line.
However, there is a related issue mentioned in https://github.com/ros-planning/moveit/issues/2741#issuecomment-881953482 that is still open:
Adapt the [`ProgressBarEditor`](https://github.com/ros-planning/moveit/blob/ef085558818e7e4fbc97e41f8dd12a1a2d6dbfa0/moveit_ros/visualization/motion_planning_rviz_plugin/include/moveit/motion_planning_rviz_plugin/motion_planning_frame_joints_widget.h#L190) to also show units.

gquey, 2022-01-26:
is number 15 still available on the list to be fixed? Looking to make a contribution. Thanks!

rhaschke, 2022-01-27:
Yes, so far we didn't receive a PR to fix the display of units in the joints tab. @AbhiramL, did you start working on this?

gquey, 2022-01-29:
@rhaschke  would it be okay if I started working on number 15 then if it is still available? Thanks

rhaschke, 2022-01-30:
@gquey, sure! I'm looking forward to your PR.

mralderman, 2022-09-16:
I am looking to contribute to this project. I was wondering if any of these issues are still avaliable?

rhaschke, 2022-09-16:
Sure. All issues not listed as done should be still available. Before starting, just check whether the latest source build still shows the mentioned problem.

7Arms, 2022-09-18:
Greetings @rhaschke , I want to solve one of these issues. Is it possible I could still work on any of the issues listed?

rhaschke, 2022-09-19:
As I wrote before: Just pick an issue, verify that it is still pending, mark the issue as being worked on and start. We are looking forward to your contribution! If you have questions, please open a new issue for them.

aaryanmurgunde, 2023-10-25:
Hi, I would like to work on the following issue, where can I find the relevant piece of code ?

> 14. In the "Joints" tab, joint values are given in degree. Some people think this is the correct way to show things to the user. But, in addition to this being a debatable choice, this makes it awful to set values from debugging output (in rad) to look at the robot state visually. As there are two clear opinions on this issue, I would propose to add a checkbox to switch units.

Jscastro197, 2023-12-11:
Can I still work on these issues? & do i need to create a new issue to find the relevant piece of code ?

OMar-Adelll, 2026-05-23:
Hi, I would like to work on this issue. This is my first time contributing, and I want to gain confidence. I believe it will push me to make more effective contributions.