moveit/moveit #2743: Pilz motion planner not loaded

kappa95, 2021-06-28:
### Description

Dear community,

I have created a very simple URDF and a Moveit! package with the setup assistant. I want to try the Pilz industrial motion and I have installed the packages: 

- `pilz_industrial_motion`
- `pilz_robot_programming`
- `pilz_robots`

 When I am launching the `demo.launch` I get these errors:
```bash
Planning scene monitors started.
[ERROR] [1624886667.050254776]: Exception while loading planner 'pilz_industrial_motion_planner::CommandPlanner': According to the loaded plugin descriptions the class pilz_industrial_motion_planner::CommandPlanner with base class type planning_interface::PlannerManager does not exist. Declared types are  ompl_interface/OMPLPlanner pilz::CommandPlanner
Available plugins: ompl_interface/OMPLPlanner, pilz::CommandPlanner
[ INFO] [1624886667.059809608]: Loading robot model 'robot'...
[ INFO] [1624886667.059939682]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1624886667.064689060]: rviz version 1.13.17
[ INFO] [1624886667.064803655]: compiled against Qt version 5.9.5
[ INFO] [1624886667.064825286]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1624886667.073215034]: Forcing OpenGl version 0.
[ INFO] [1624886667.086437718]: Set joints of group 'manipulator' to pose 'home'.
[ INFO] [1624886667.086881970]: Fake controller 'fake_manipulator_controller' with joints [ joint1 joint2 joint3 joint4 joint5 ]
[ INFO] [1624886667.089835383]: Returned 1 controllers in list
[ INFO] [1624886667.102665081]: Trajectory execution is managing controllers
Loading 'move_group/ApplyPlanningSceneService'...
Loading 'move_group/ClearOctomapService'...
Loading 'move_group/MoveGroupCartesianPathService'...
Loading 'move_group/MoveGroupExecuteTrajectoryAction'...
Loading 'move_group/MoveGroupGetPlanningSceneService'...
Loading 'move_group/MoveGroupKinematicsService'...
Loading 'move_group/MoveGroupMoveAction'...
Loading 'move_group/MoveGroupPickPlaceAction'...
Loading 'move_group/MoveGroupPlanService'...
Loading 'move_group/MoveGroupQueryPlannersService'...
Loading 'move_group/MoveGroupStateValidationService'...
Loading 'pilz_industrial_motion_planner/MoveGroupSequenceAction'...
[ERROR] [1624886667.169623312]: Exception while loading move_group capability 'pilz_industrial_motion_planner/MoveGroupSequenceAction': According to the loaded plugin descriptions the class pilz_industrial_motion_planner/MoveGroupSequenceAction with base class type move_group::MoveGroupCapability does not exist. Declared types are  move_group/ApplyPlanningSceneService move_group/ClearOctomapService move_group/MoveGroupCartesianPathService move_group/MoveGroupExecuteService move_group/MoveGroupExecuteTrajectoryAction move_group/MoveGroupGetPlanningSceneService move_group/MoveGroupKinematicsService move_group/MoveGroupMoveAction move_group/MoveGroupPickPlaceAction move_group/MoveGroupPlanService move_group/MoveGroupQueryPlannersService move_group/MoveGroupStateValidationService pilz_trajectory_generation/MoveGroupSequenceAction pilz_trajectory_generation/MoveGroupSequenceService
Loading 'pilz_industrial_motion_planner/MoveGroupSequenceService'...
[ERROR] [1624886667.169756011]: Exception while loading move_group capability 'pilz_industrial_motion_planner/MoveGroupSequenceService': According to the loaded plugin descriptions the class pilz_industrial_motion_planner/MoveGroupSequenceService with base class type move_group::MoveGroupCapability does not exist. Declared types are  move_group/ApplyPlanningSceneService move_group/ClearOctomapService move_group/MoveGroupCartesianPathService move_group/MoveGroupExecuteService move_group/MoveGroupExecuteTrajectoryAction move_group/MoveGroupGetPlanningSceneService move_group/MoveGroupKinematicsService move_group/MoveGroupMoveAction move_group/MoveGroupPickPlaceAction move_group/MoveGroupPlanService move_group/MoveGroupQueryPlannersService move_group/MoveGroupStateValidationService pilz_trajectory_generation/MoveGroupSequenceAction pilz_trajectory_generation/MoveGroupSequenceService

[ WARN] [1624886667.169944935]: MoveGroup running was unable to load pilz_industrial_motion_planner::CommandPlanner
```

### Environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04
* Binary version installed as Debian package

o0stinger0o, 2021-08-08:
The cause of this problem is that planner name has changed,it depend the  ros-melodic-pilz-industrial-motion-planner vision.

```
[ERROR] [1624886667.050254776]: Exception while loading planner 'pilz_industrial_motion_planner::CommandPlanner': According to the loaded plugin descriptions the class pilz_industrial_motion_planner::CommandPlanner with base class type planning_interface::PlannerManager does not exist. Declared types are  ompl_interface/OMPLPlanner pilz::CommandPlanner
Available plugins: ompl_interface/OMPLPlanner, pilz::CommandPlanner
```

you should modify 'pilz_industrial_motion_planner::CommandPlanner' to 'pilz::CommandPlanner'
it usually in    yourworkspace/xxx_moveit_config/launch/pilz_command_planner_planning_pipeline.launch.xml 
like this:

```
<launch>

  <!-- Pilz Command Planner Plugin for MoveIt! -->
  <arg name="planning_plugin" value="pilz_industrial_motion_planner::CommandPlanner" />

  <arg name="planning_adapters" value="
default_planner_request_adapters/AddTimeParameterization     
default_planner_request_adapters/FixWorkspaceBounds     
default_planner_request_adapters/FixStartStateBounds     
default_planner_request_adapters/FixStartStateCollision     
default_planner_request_adapters/FixStartStatePathConstraints" />

  <arg name="start_state_max_bounds_error" value="0.1" />

  <param name="planning_plugin" value="$(arg planning_plugin)" />
  <param name="request_adapters" value="$(arg planning_adapters)" />
  <param name="start_state_max_bounds_error" value="$(arg start_state_max_bounds_error)" />

</launch>
```

kappa95, 2021-08-20:
After the update the error vanishes. But now when I am executing this simple code:
```python

from pilz_robot_programming import *
import rospy
import math
from geometry_msgs.msg import Point

__REQUIRED_API_VERSION__ = "1"

def start_program():
    print("Executing " + __file__)

    r = Robot(__REQUIRED_API_VERSION__)
    # Simple ptp movement
    r.move(Ptp(goal=[0, 0.5, 0.5, 0, 0, 0], vel_scale=0.4))
    start_joint_values = r.get_current_joint_states()

if __name__ == '__main__':
    rospy.init_node('test')
    start_program()
```
I get:
```bash
[INFO] [1629473589.370637]: Waiting for connection to action server sequence_move_group...
[INFO] [1629473589.662082]: Waiting for connection to service /prbt/get_speed_override...
timeout exceeded while waiting for service /prbt/get_speed_override
```

kappa95, 2021-08-20:
Solved that error adding to the pilz launcher in moveit this line:

```xml
<include file="$(find prbt_hardware_support)/launch/fake_operation_mode_setup.launch" />
```

BeratTezer, 2024-02-21:
Hi, I use this code

    #!/usr/bin/env python

    from geometry_msgs.msg import Pose, Point
    from pilz_robot_programming import *
    import math
    import rospy

    __REQUIRED_API_VERSION__ = "1"  # API version
    __ROBOT_VELOCITY__ = 0.5        # velocity of the robot

    # main program
    def start_program():
        print(r.get_current_pose()) # print the current position of thr robot in the terminal

    if __name__ == "__main__":
        # init a rosnode
        rospy.init_node('robot_program_node')

        # initialisation
        r = Robot(__REQUIRED_API_VERSION__)  # instance of the robot

        # start the main program
        start_program()

But I cannot connect to the program. Do you have an any idea?