moveit/moveit #2746: Migrate moveit_opw_plugin to moveit_kinematics

v4hn, 2021-06-30:
As discussed [due to an issue with pilz planner tests](https://github.com/ros-planning/moveit/issues/2653#issuecomment-869067421) we would like to move [`moveit_opw_kinematics_plugin`](https://github.com/JeroenDM/moveit_opw_kinematics_plugin) into [`moveit_kinematics`](https://github.com/ros-planning/moveit/tree/master/moveit_kinematics) and want to migrate [our `fanuc` package in `moveit_resources`](https://github.com/ros-planning/moveit_resources/blob/master/fanuc_moveit_config/config/kinematics.yaml#L2) to use it in favor of the KDL solver.

This will also help users in general because @JeroenDM (current maintainer of the plugin) does not have enough time to provide support/do releases.

simonschmeisser, 2021-11-05:
Would I basically just copy paste a certain version in a PR? @JeroenDM  was working on a noetic version that uses the latest API of `opw``as a dependency instead of a git submodule.

v4hn, 2021-11-05:
> Would I basically just copy paste a certain version in a PR?

As stated we would want to take over maintainership for this plugin altogether, possibly adding JeroenDM to help maintain if he's interested.

> was working on a noetic version that uses the latest API of `opw``as a dependency instead of a git submodule.

That's the most critical part that should happen to add it here. I don't see us adding a submodule for a single optional plugin.
And I guess we would want to make it part of moveit_kinematics, so a bit more than c&p is involved.

gavanderhoorn, 2021-11-05:
> > was working on a noetic version that uses the latest API of `opw``as a dependency instead of a git submodule.
> 
> That's the most critical part that should happen to add it here. I don't see us adding a submodule for a single optional plugin.

Noetic has this released already: [status_page/ros_noetic_default.html?q=opw](http://repositories.ros.org/status_page/ros_noetic_default.html?q=opw).

We could ask @Levi-Armstrong to release for Melodic as well. I'm actually not sure why he didn't do that yet.

simonschmeisser, 2021-11-05:
I just noticed I wasn't up to date and there is now a `noetic-devel` branch as well. Having `opw` released for melodic would indeed be good as MoveIt master is supposed to be melodic compatible.

Levi-Armstrong, 2021-11-06:
I will make a release next week.

JeroenDM, 2021-11-07:
> Would I basically just copy paste a certain version in a PR? @JeroenDM was working on a noetic version that uses the latest API of `opw``as a dependency instead of a git submodule.

Both the melodic and noetic versions should be in a working state (and are released). Maybe we can sync some minor differences before copy-pasting them into moveit_kinematics?

gavanderhoorn, 2021-12-02:
@Levi-Armstrong released `opw_kinematics` for Melodic: https://github.com/ros/rosdistro/pull/31368.

@Levi-Armstrong: is this the exact same version as was released for Noetic?

Levi-Armstrong, 2021-12-02:
Yes