moveit/moveit #2757: use std::make_shared where applicable

v4hn, 2021-07-08:
MoveIt is sprinkled with `...SharedPtr::reset(new ...)` code and `SharedPtr(new ...)` constructors.
It is slightly (though in most cases not noticeably) faster and cleaner to use `std::shared_ptr<...>()` instead.

Anyone is welcome to help clean this up and create a pull-request along the lines of #2756 for wider parts of the code base. :muscle:

pvanlaar, 2021-07-08:
I like to help! @v4hn, could you assign me? regards Patrick

jaskiratsingh2000, 2021-07-17:
Hey  I am quite beginner and super interested to begin my contribution towards this repo and this issue seems interesting to me. Can you please let me know that where exactly do I have to make changes or add functionalities?
Any examples corresponding to this would be highly appreciable. Thanks!

jaskiratsingh2000, 2021-07-17:
Is this still open?

rhaschke, 2021-07-17:
As you can see a few lines above, this issue was tackled in #2762. However, because that PR is not yet merged (due to some open issues), this issue is not yet closed yet as well.

jaskiratsingh2000, 2021-07-17:
So @rhaschke Do you think that I should work on this?

jaskiratsingh2000, 2021-07-17:
@rhaschke Can you please assign this to me?

rhaschke, 2021-07-17:
I don't think there is anything you can do. I'm looking into the open issues right now.

jaskiratsingh2000, 2021-07-17:
@rhaschke Yes, please redirect me to super beginner friendly issue as I am supper interested to work under this repo. So I am looking forward to hearing from you

pvanlaar, 2021-07-19:
The PR is merged, so this issue can be closed.

rhaschke, 2021-07-19:
Closed via #2762.

closed 2021-07-19 by rhaschke