moveit/moveit #2760: Building fails for newest noetic-devel branch

udopa, 2021-07-09:
### Description
Building with `catkin build` fails for newest noetic-devel branch because of missing headers. The bug was probably introduced in #2604 with the move to cmake headers ([here](https://github.com/ros-planning/moveit/commit/7d621a3be08a6df71f669ccc9c0791b86f9e61fb)). As far as I can see, the master branch is probably affected by this as well.

I tried updating my dependencies with rosdep, but that didn't work.

### Your environment
* ROS Distro: Noetic
* OS Version: Ubuntu 20.04
* Source build
* If source, which branch? - noetic-devel [commit](https://github.com/ros-planning/moveit/commit/83c82e34282a5448a302e7dd1cfdff48fed95c1d)

### Steps to reproduce
Pull noetic-branch and build with `catkin build`

### Expected behaviour
Successful build for complete moveit package

### Actual behaviour
Build fails because of missing headers:
- ```#include "moveit_kinematics_base_export.h"``` in _moveit/moveit_core/kinematics_base/include/moveit/kinematics_base/kinematics_base.h_
- ```#include "moveit_planning_scene_export.h"``` in _moveit/moveit_core/planning_scene/include/moveit/planning_scene/planning_scene.h_

### Backtrace or Console output

```sh
/home/user/catkin_ws/src/moveit/moveit_core/kinematics_base/include/moveit/kinematics_base/kinematics_base.h:47:10: fatal error: moveit_kinematics_base_export.h: No such file or directory
```
The same problem would exist for the planning scene export, but building crashes before it reaches that point.

rhaschke, 2021-07-09:
Our CI builds show that MoveIt compiles without any issues: https://github.com/ros-planning/moveit/actions/runs/1011787937
Could you please describe your catkin workspace in more detail to understand the actual problem? Did you try a `catkin clean` before building?

udopa, 2021-07-09:
Yea, sure. This is my .rosinstall file for my workspace. 
```
- git:
    local-name: moveit_msgs
    uri: https://github.com/ros-planning/moveit_msgs.git
    version: master
- git:
    local-name: moveit
    uri: https://github.com/ros-planning/moveit.git
    version: noetic-devel
- git:
    local-name: universal_robot
    uri: https://github.com/fmauch/universal_robot.git
    version: calibration_devel
- git:
    local-name: Universal_Robots_ROS_Driver
    uri: https://github.com/UniversalRobots/Universal_Robots_ROS_Driver.git
    version: master
- git:
    local-name: Universal_Robots_Client_Library
    uri: https://github.com/UniversalRobots/Universal_Robots_Client_Library.git
    version: master
- git:
    local-name: moveit_task_constructor
    uri: https://github.com/ros-planning/moveit_task_constructor.git
    version: master
# Temporary, till noetic release is available
- git:
    local-name: industrial_core
    uri: https://github.com/simonschmeisser/industrial_core.git
    version: noetic-compat
- git:
    local-name: industrial_robot_status_controller
    uri: https://github.com/gavanderhoorn/industrial_robot_status_controller.git
    version: master
- git:
    local-name: moveit_visual_tools
    uri:  https://github.com/ros-planning/moveit_visual_tools.git
    version: noetic-devel
- git:
    local-name: geometric_shapes
    uri:  https://github.com/ros-planning/geometric_shapes.git
    version: noetic-devel
- git:
    local-name: moveit_resources
    uri:  https://github.com/ros-planning/moveit_resources.git
    version: master
- git:
    local-name: rviz_visual_tools
    uri:  https://github.com/PickNikRobotics/rviz_visual_tools.git
    version: master
```

Additionally to those packages I have my own package which contains all my project specific code & stuff. But I think that shouldn't effect MoveIt.

A colleague also tried building it on another machine with the same workspace and got the same error. I tried `catkin clean`, but without effect.

rhaschke, 2021-07-09:
Hm. I confirm the issue for a local build. @v4hn, did you try to build #2604 locally before merging? I don't yet see why CI succeeds, but a local build fails.

rhaschke, 2021-07-09:
I think the key difference of CI is that it uses an install space, while my local build doesn't! The header files are generated in `build`, which doesn't become part of the `devel` include folders!

gavanderhoorn, 2021-07-09:
Generated code might need a different path for `(target_)include_directories(..)`.

Perhaps a generator expression or a Catkin CMake extras file?

rhaschke, 2021-07-09:
We have a similar issue with `version.h` which is generated during build. The solution, indeed is to generate the file in devel space and configure all include dirs correctly. I added a corresponding commit to my fixup PR #2758.

v4hn, 2021-07-09:
> v4hn, did you try to build #2604 locally before merging? I don't yet see why CI succeeds, but a local build fails.

No, I relied on CI. But I also would not have noticed it as my main moveit workspace is an install workspace...
It sounds like a good idea to migrate one of our CI jobs to a devel workspace.

v4hn, 2021-07-09:
Should be fixed with Robert's commit in #2758 . Sorry for your time. :(

closed 2021-07-09 by v4hn