moveit/moveit #2763: move a CI job to build a devel workspace instead of an install workspace

v4hn, 2021-07-09:
Related to [this comment](https://github.com/ros-planning/moveit/issues/2760#issuecomment-877437306).

Some issues in cmake configurations for ROS appear only *either* in `devel` *or* in `install` workspaces.
So it would make sense to move a CI job for the latest branch (`master-ci-shadow-fixed` or `master-ci`) to use a `devel` workspace instead.
I very briefly looked into `industrial_ci` documentation, but could not find an obvious flag for it.

rhaschke, 2021-07-10:
I checked this as well. But there is no option to use devel space workspaces. 
@ipa-mdl: Do you have any suggestions how to test devel-space builds?

ipa-mdl, 2021-07-15:
> Do you have any suggestions how to test devel-space builds?

Right now this is not supported, but I see the need for it in your case.
The `install` space is hard-coded everywhere, but we could make it an internal configuration option, which can be overwritten in an `* _EMBED` hook. Does `BUILDER=catkin_tools` build the `devel` space anyway?

ipa-mdl, 2021-07-15:
> Does BUILDER=catkin_tools build the devel space anyway?

Yes, it does.

Would it be okay to just run the tests and the downstream build from the devel space, .i.e. upstream and target workspace would still use install?

ipa-mdl, 2021-07-15:
With https://github.com/ros-industrial/industrial_ci/pull/698 you can configure devel space builds (and proper chaining) in a custom builder (overlaying catkin_tools):

```yaml
BUILDER: .github/workflows/catkin_tools_devel.sh
```

.github/workflows/catkin_tools_devel.sh:
```bash
BUILDER=catkin_tools ici_source_builder

function ici_extend_space {
  echo "$1/devel"
}
function builder_run_build {
  ici_exec_in_workspace "$1" "$2" catkin build --summarize  --no-status "${@:3}"
}
```
This makes the MR in question fail the tests.

If you want to use the devel space only for certain parts, you can source this in an `*_EMBED` hook instead.

I am not sure yet, if I want to bundle this builder with `industrial_ci`, but it could be offered next to https://github.com/ros-industrial/industrial_ci/pull/631.

rhaschke, 2021-07-15:
You are right: `catkin_tools` creates both, a devel and an install space. However, ROS packages are found from the install space, not the devel space when building with `--install` option. Thus your suggestion from https://github.com/ros-planning/moveit/issues/2763#issuecomment-880604870 doesn't help. The issue #2760 was caused by a header not being found from `devel` space, while it would have been available from `install` space.

The solution offered in https://github.com/ros-industrial/industrial_ci/pull/698 and https://github.com/ros-planning/moveit/issues/2763#issuecomment-880708996 should work. I will test it immediately. Thanks!
I would appreciate bundling this solution with `industrial_ci` in a similar fashion as https://github.com/ros-industrial/industrial_ci/pull/631 :smile:

ipa-mdl, 2021-07-15:
@rhaschke: I created a draft in https://github.com/ros-industrial/industrial_ci/pull/700, please test it.

rhaschke, 2021-07-15:
Thanks! Test is triggered: https://github.com/ubi-agni/moveit/runs/3078295943?check_suite_focus=true
Manually overlaying catkin_tools already [revealed the previously hidden issue](https://github.com/ubi-agni/moveit/runs/3077922422?check_suite_focus=true).

ipa-mdl, 2021-07-15:
I forgot to change `ici_extend_space`. Should now work as expected.

rhaschke, 2021-07-15:
There are still some `install` folders used:
https://github.com/ubi-agni/moveit/runs/3079253201?check_suite_focus=true

PR #2772 (2021-07-21): CI improvements
See individual commit messages. Closes #2763.
closed 2021-07-21 by rhaschke