moveit/moveit #2764: PlanningSceneInterface::clearScene()

v4hn, 2021-07-09:
The best reliable way I know for user nodes to remove all objects from the planning scene is this:
```
   moveit::planning_interface::PlanningSceneInterface psi;
   moveit_msgs::PlanningScene rm;
   rm.is_diff = true;
   rm.robot_state.is_diff = true;
   rm.robot_state.attached_collision_objects.resize(1);
   rm.robot_state.attached_collision_objects[0].object.operation = moveit_msgs::CollisionObject::REMOVE;
   rm.world.collision_objects.resize(1);
   rm.world.collision_objects[0].operation = moveit_msgs::CollisionObject::REMOVE;
   psi.applyPlanningScene(rm);
```
I would propose to simplify this to
```
   moveit::planning_interface::PlanningSceneInterface psi;
   psi.clearScene();
```

Opinions? Additions? Alternatives?

felixvd, 2021-07-12:
I am all for this - in fact, it already [exists in Python](https://docs.ros.org/en/api/moveit_commander/html/planning__scene__interface_8py_source.html#l00163) and the function should be linked if a C++ implementation is added.

pvanlaar, 2021-07-15:
PlanningScene has a `removeAllCollisionObjects()` method. Doesn't that work in this case? The planning_interface does not work with MoveItCpp I believe, but then the PlanningScene can be accessed through the PlanninSceneMonitor. So `removeAllCollisionObjects()` works for both MoveGroupInterface and MoveItCpp.

v4hn, 2021-07-19:
> PlanningScene has a `removeAllCollisionObjects()` method. Doesn't that work in this case?

Using the MGI/PSI you usually interact with the PlanningScene of a `move_group` ROS node.
You don't have access to this method of the C++ PlanningScene class from outside the node and the PSM does not help with that.

For MoveItCpp I agree though.

rhaschke, 2023-10-20:
@aaryanmurgunde, you are very welcome to work on this. Concrete TODOs:
- [ ] Wrap the listed code snippet into a new function in `PlanningSceneInterface::clear()`.
- [ ] Optionally use that function also in python code as suggested [here](https://github.com/ros-planning/moveit/issues/2764#issuecomment-877979974)
- [ ] Test everything. Ideally, provide a new unit test for it.

aaryanmurgunde, 2023-10-21:
> @aaryanmurgunde, you are very welcome to work on this. Concrete TODOs:
> 
> * [x]  Wrap the listed code snippet into a new function in `PlanningSceneInterface::clear()`.
> * [ ]  Optionally use that function also in python code as suggested [here](https://github.com/ros-planning/moveit/issues/2764#issuecomment-877979974)
> * [x]  Test everything. Ideally, provide a new unit test for it.

@rhaschke I'm not sure about the wrapper for python if you could help me understand that then I can build for the same as well.

Thank You : )

rhaschke, 2023-10-21:
Just use the newly implemented function in python, i.e. call `self._psi.clear()`. This requires `clear` to be exposed by the python wrapper of `PlanningSceneInterface`.

aaryanmurgunde, 2023-10-23:
@rhaschke For the python wrapper file wrapper_python_planning_scene_interface.cpp on line [L112](https://github.com/ros-planning/moveit/blob/984a0ea2a00b4190cad9e08ea0d764c19dbf9975/moveit_ros/planning_interface/planning_scene_interface/src/wrap_python_planning_scene_interface.cpp#L112) adding 
```c++
  bool clearPython()
  {
    return clear();
  }
```
and on line [L126](https://github.com/ros-planning/moveit/blob/984a0ea2a00b4190cad9e08ea0d764c19dbf9975/moveit_ros/planning_interface/planning_scene_interface/src/wrap_python_planning_scene_interface.cpp#L126)
```c++
++ planning_scene_class.def("clear", &PlanningSceneInterfaceWrapper::clearPython);
```

and in the planning_scene_interface.py on line [L171](https://github.com/ros-planning/moveit/blob/984a0ea2a00b4190cad9e08ea0d764c19dbf9975/moveit_commander/src/moveit_commander/planning_scene_interface.py#L171)

```python
# replace 
-- self.remove_attached_object()
-- self.remove_world_object()

# with
++ self._psi.clear()
```

Any other changes that need to made ?

rhaschke, 2023-10-23:
I suffices to define the wrapper like this:
```diff
- planning_scene_class.def("clear", &PlanningSceneInterfaceWrapper::clearPython);
+ planning_scene_class.def("clear", &PlanningSceneInterfaceWrapper::clear);
```

aaryanmurgunde, 2023-10-23:
Done! pushed a commit for that as well.

Thank you so much for helping me out @rhaschke. I has been a big learning curve, I'm now much more prepared to work more on fixing issues and contributing to Open Source.

aaryanmurgunde, 2023-10-25:
Let's Close this issue then @rhaschke ?

rhaschke, 2023-10-25:
This issue will be closed via #2764.

closed 2023-10-25 by rhaschke