moveit/moveit #2766: avoid `boost`/`std::bind`

v4hn, 2021-07-10:
Similar in spirit to #2757, here is another nagging detail scattered throughout the code base:

Scott Meyer's "Effective Modern C++" Item 34 as well as [TOTW 108](https://abseil.io/tips/108) state:

> Avoid `boost`/`std::bind` [in favor of lambdas where applicable]

MoveIt uses `bind` in various places and readability suffers from it.
We should update the definitions to use lambdas with properly named arguments instead!

tylerjw, 2021-11-19:
All the lambdas!

rhaschke, 2021-11-19:
There is clang-tidy modernize-avoid-bind: https://github.com/ros-planning/moveit/commit/29d2de102e47c5ab88207a624f91b2b6c8752ed6 :smile: I will prepare a PR if clang-tidy is done...

v4hn, 2021-11-19:
> There is clang-tidy modernize-avoid-bind

:open_mouth:  Cool!

rhaschke, 2021-11-19:
Unfortunately, it doesn't translate placeholders into arguments :disappointed:  I'm trying clang-tidy-12 now...

PR #3106 (2022-04-02): bind() them all (not)
Fixes #2766. Replaces #2972 .

@Abishalini will hate me for this changeset for its ease of porting to moveit2... :cry: 
But it was overdue, I managed to do it over a few evenings and I believe it makes the code much more readable.

This PR should get some priority because patches are quite likely to generate conflicts with this due to its size.
I split all non-trivial changes into separate commits with one huge commit in the end addressing all "trivial" (up to identifier naming) cases. During this work I found and reworked a number of internal details, often related to po [... cut]
closed 2022-04-02 by v4hn