moveit/moveit #2767: Plan and execute to an already reached goal fails

Martin-Oehler, 2021-07-12:
### Description

I am trying to a plan to a goal state, that has already been reached. This should trivially succeed. Unfortunately, it does not. The problem seems to be, that the planner (geometric::RRTConnect) generates a trajectory with only one point, which is the start state, with "time_from_state: 0". The joint_trajectory_controller always discards the first trajectory point (see [ros_controllers#291](https://github.com/ros-controls/ros_controllers/issues/291)) and fails due to the now empty goal.

Can anyone replicate this issue?

### Your environment
* ROS Distro: Noetic
* OS Version: Ubuntu 20.04
* Binary Version: 1.1.5-1focal.20210601.160327

### Steps to reproduce
Plan and execute a trajectory to a goal state. Plan to the same goal and execute again.

### Expected behaviour
The second execution should trivially succeed.

### Actual behaviour
The joint trajectory controller fails with error "INVALID_GOAL".

### Console output

```
[ INFO] [1626081730.155338431, 136.241000000] [/move_group]: ParallelPlan::solve(): Solution found by one or more threads in 0.056407 seconds
[ INFO] [1626081730.155430063, 136.241000000] [/move_group]: SimpleSetup: Path simplification took 0.000005 seconds and changed from 2 to 2 states
[ INFO] [1626081733.409343666, 139.474000000] [/move_group]: Execution request received
[ WARN] [1626081733.426189726, 139.491000000] [/gazebo]: Dropping all 1 trajectory point(s), as they occur before the current time.
Last point is 0.000000s in the past.
[ WARN] [1626081733.428119595, 139.492000000] [/move_group]: Controller /telemax_control/manipulator_arm_traj_controller failed with error INVALID_GOAL: Dropping all 1 trajectory point(s), as they occur before the current time.
Last point is 0.000000s in the past.
[ WARN] [1626081733.428201170, 139.492000000] [/move_group]: Controller handle /telemax_control/manipulator_arm_traj_controller reports status FAILED
```

simonschmeisser, 2021-07-17:
We have code that checks this in our application but likely that should also be handled somehow withing MoveIt. Ideally before even starting a planner (OMPL in your case)

felixvd, 2021-07-19:
I have a similar issue, but I haven't had time to look into it in detail. I assumed that controller noise is the cause.

v4hn, 2021-07-20:
Are you using the MoveGroup action of `move_group`?
It actually [checks whether the goal is trivially true](https://github.com/ros-planning/moveit/blob/master/moveit_ros/move_group/src/default_capabilities/move_action_capability.cpp#L111-L120) and should not send any trajectory to execution in this case.

Martin-Oehler, 2021-07-23:
Thanks for your replies.
@v4hn I replicated this behavior using the rviz MotionPlanning plugin. 

I did some investigation: If you click on "Plan & Execute", this problem does not appear, likely due to the mentioned check:
```
[ INFO] [1627034873.820536557, 196.122000000] [/move_group]: Combined planning and execution request received for MoveGroup action. Forwarding to planning and execution pipeline.
[ INFO] [1627034873.820667509, 196.122000000] [/move_group]: Goal constraints are already satisfied. No need to plan or execute any motions
```
However, if you click on "Plan" and then on "Execute" separately, a motion plan is generated and sent to the controllers as initially reported.

Even if there is no check for trivial success, moveit should always sent a valid trajectory to the joint trajectory controllers, which appears to not always be the case.

v4hn, 2021-07-23:
> However, if you click on "Plan" and then on "Execute" separately, a motion plan is generated and sent to the controllers as initially reported.

Hmm, interesting case...
I would expect that `plan` returns a *one-waypoint* trajectory, which is apparently what it does.
The alternative here would be to return an *empty* trajectory and that would solve your problem, but as a user I would expect at least one valid joint state in the result.

The independent other question is what MoveIt's execution should do when it receives a one-waypoint trajectory.
The error you observe happens because ExecuteTrajectoryAction is mainly a thin wrapper around multiple FollowJointTrajectory action servers and just forwards the trajectory in this case.
Would you expect this wrapper to inspect the trajectory and not send it if the timestamp is unset and time_from_start for the only waypoint is 0?
We could do that, though I don't think it's necessarily the job of this wrapper.

I'm rather undecided right now. I don't like the behavior you describe (error message for regular use of the RViz GUI), but I also don't think any of the involved components does anything wrong.

Personally I blame ros_control for printing the error at all.
If you command the robot to go where it is, that's a trivial request, and not invalid input.
It's only invalid input if the described state *is not the current state*, but from what I have read this is not checked.

Martin-Oehler, 2021-07-23:
There are definitely multiple ways to solve this problem, but I am also conflicted, what's the most intuitive way.

> If you command the robot to go where it is, that's a trivial request, and not invalid input.

One could argue, that a trajectory to the current position is no trajectory at all.

I see two options:
1) Make the execution logic in MoveIt check the goal constraints similar to the plan-execute code path. Checking timestamp and time_from_start feels more like a hack.
2) Convince the people at ros_control, that a trivial trajectory is indeed a trajectory and the joint trajectory controller should check goal constraints for the start state.

simonschmeisser, 2021-07-23:
We don't use ros_control but still implemented a workaround so I guess there are other controllers with the same "issue"

v4hn, 2021-07-26:
> One could argue, that a trajectory to the current position is no trajectory at all.

It's definitely still a trajectory by data type. :) But if Simon knows other controllers with similar restrictions, it makes sense for MoveIt to add a safety net instead of blaming the drivers.

> 1) Make the execution logic in MoveIt check the goal constraints similar to the plan-execute code path.

This is not an option because the trajectory execution does not know the goal constraints anymore. It just receives a trajectory.

Instead of fixing it in either interface (`PlanningPipeline` or `TrajectoryExecutionManager`), we could also add a check for a trivial trajectory to RViz and disable execution in that case.  But then we leave the problem open for any user using `mgi.plan(...); mgi.execute(...)` and I would not want to force them to add logic to this trivial use case.

> Checking timestamp and time_from_start feels more like a hack.

It's the only general way to solve this from my perspective and I would encourage you to add a pull-request for it.
On top, we could actually add error messages if the current state of the robot is too far away from the requested state.

felixvd, 2021-07-26:
Requiring a minimum trajectory time is fine. A single-point trajectory with `time_from_start=0` makes no sense.

oskarsVismanis, 2023-03-30:
Hey, what's the status of this issue? I'm encountering it as well and the first thought was to make a workaround by checking if the coordinates differ or no, but I would like to be involved with fixing this in the _correct_ way!

rhaschke, 2023-04-01:
This was fixed recently in #3362. It's not yet released!

closed 2023-04-01 by rhaschke