moveit/moveit #2768: Is the Pilz planner's MoveGroupSequence thread-safe?

cambel, 2021-07-15:
This stack trace seems to imply that there is something wrong with the memory allocation in this section of the MoveGroupSequence planning. We are planning in multiple threads when this error occurred. Still looking for a way to reliably reproduce this.

### Your environment
Ubuntu 18.04, Melodic, source build

### Steps to reproduce
Not sure yet

### Backtrace or Console output

https://gist.github.com/cambel/d7fc12c65b6117ead266eacc68544066

felixvd, 2021-07-15:
@jschleicher @ct2034 If you have insights from the development of this, please feel free to share. I have been debugging this as well and don't see where it is going off the rails.

cambel, 2021-07-16:
More information about this.

The problem seems to happen only when using multithreading while planning with the MoveGroupSequenceService. If done with the MoveGroupSequenceAction, it does not happen (at least not so far). So, are ros services thread safe?

gavanderhoorn, 2021-07-16:
In general that's not really something you can ask, nor answer.

"ROS services" just provide an RPC mechanism.

"thread safety" of that would come down to whether the server service *handler* (ie: the callback) has been written with thread safety in mind.

If not, *and* a multi-threaded execution environment is used (ie: using an `AsyncSpinner` or `MultiThreadedSpinner`), bad things can happen, as service callbacks can now be called in parallel.

I would expect there to be no difference between actions and services, so I can't say why you observe something different there. Perhaps the `MoveGroupSequenceAction` does use locking or in some other way prevent concurrent access to a shared resource.

(if you're interested: the threading model of both `roscpp` and `rospy` have been discussed extensively on ROS Answers, an the ROS wiki also has sections about it)

felixvd, 2021-07-16:
Yes, that's all correct. Thanks for the always informative answers. To clarify my question to @ct2034 and @jschleicher , I'd like to know if the `MoveGroupSequence` action and service have been written with thread safety in mind.

gavanderhoorn, 2021-07-16:
@felixvd: do you see any resources which could potentially be shared if the callback(s) would be called concurrently for different service clients?

Are there any `lock`s or other serialisation barriers?

If not: it's probably not safe.

felixvd, 2021-07-16:
No, I don't see any obstacles to serialization, but the code is very nested and I also hadn't noticed that it ignores attached bodies, so I would need to do a pretty in-depth read to draw a conclusion. I figure that asking is more efficient here.