moveit/moveit #2773: Pilz planner always allow collision between attached bodies and robot/world

cambel, 2021-07-16:
### Description

When planning with Pilz planners, the attached bodies are not considered for collision checking.

### Your environment
Ubuntu 18.04, Melodic, source build

### Steps to reproduce
Attach a body to the robot, define a target pose where ONLY the attach body is in collision with some other object, plan with any Pilz planner.

### Expected behaviour
Plan must fail.

### Actual behaviour
Plan succeed.

### Backtrace or Console output

Here is a video comparing planning with OMPL vs Pilz and the corresponding [logs](https://gist.github.com/cambel/5a7a01ec3ccc6e4a399d04a095237e30) 

https://user-images.githubusercontent.com/3798796/125898070-4ce1f7b1-c9aa-44af-8cc1-fbfb41cf53a4.mp4

cambel, 2021-07-16:
This is on top of #2676, after applying the changes in PR #2754, the collision check between the robot and the world with respect to the AllowCollisionMatrix works okay. However, the attached bodies seem to be ignored.

How does the planning scene knows that it should check the attached bodies when requesting a collisionCheck? I am checking this issue but I don't really get why the planning pipeline is ignoring this collision checks. Here in the planning_pipeline, there is a recheck of the plan generated by the planner. And as you can see in the logs above (with debug activated), the plan seems to be valid: `Planned path was found to be valid when rechecked`
https://github.com/ros-planning/moveit/blob/e8b54db3ae61a1314e49656ad84795d4cfc2e523/moveit_ros/planning/planning_pipeline/src/planning_pipeline.cpp#L260-L336

cambel, 2021-07-16:
Thanks to @felixvd we were able to pin-point the problem. The Pilz planner were creating a `start_state` based only on the `robot_model` stored on instantiation of the planner, thus the attached bodies were not present. The solution is to use the current state from planning scene instead.
Since this issue is related to an open PR (#2754), so we leave the proposed solution [here](https://github.com/cambel/moveit/commit/b97e32085e7bf774772737c0ed3c06aba4730008) for now until the open PR is closed, or this changes can be merge there.

rhaschke, 2021-08-18:
Should be fixed via #2803.

cambel, 2021-08-18:
@rhaschke, no, this issue is not solved yet with #2803 but now I can PR the solution we mentioned in my previous comment.