moveit/moveit #2774: PlanningSceneInterface::setStartState(moveit_msgs::RobotState) crashes if there is no "current" robot state.

werner291, 2021-07-17:
### Description

Hello,

I'm using Moveit in a situation without a robot, simulated or otherwise; I'd like to plan paths between hypothetical start-and-end states. I only have the URDF/SRDF model, so the notion of the "current" robot position is undefined.

In the implementation of `PlanningSceneInterface::setStartState(moveit_msgs::RobotState)`, note the line `impl_->getCurrentState(rs);` (https://github.com/ros-planning/moveit/blob/master/moveit_ros/planning_interface/move_group_interface/src/move_group_interface.cpp#L1611)

This means that, even if the `moveit_msgs::RobotState` fully defines the current robot state, this method call fails with an errorlike this: `[ERROR] [1626532077.440248038]: Failed to fetch current robot state`.

### Your environment
* ROS Distro: Noetic
* OS Version: e.g. Ubuntu 20.04
* Binary build. installed through APT.

### Steps to reproduce

Simply call the method referred to using a moveit_msgs::RobotState, with every joint setting defined, but without a source of current robot joint settings.

The cause seems pretty obvious from the source code.

### Expected behaviour

The provided state is used as the start state in planning.

### Actual behaviour

The aforementioned error is produced, because the method assumes that there is a source of "current" robot states.

### Backtrace or Console output

`[ERROR] [1626532077.440248038]: Failed to fetch current robot state`

werner291, 2021-07-17:
Note that `moveit::core::RobotState` also provides a constructor that just takes a `const RobotModelConstPtr&`, perhaps that's a way out?

rhaschke, 2021-07-17:
Thanks for reporting this issue. Please try https://github.com/ros-planning/moveit/pull/2775.

PR #2775 (2021-07-20): MGI::setStartState: Only fetch current state when new state is diff
Fixes #2774.
closed 2021-07-20 by v4hn