moveit/moveit #2777: Picked collision objects lose their color after placed

corot, 2021-07-18:
### Description

Picked collision objects lose their color after placed. Also the type field gets emptied. I suppose somewhere in the code we transform the co from [CollisionObject](http://docs.ros.org/en/melodic/api/moveit_msgs/html/msg/CollisionObject.html) to [AttachedCollisionObject](http://docs.ros.org/en/melodic/api/moveit_msgs/html/msg/AttachedCollisionObject.html) w/o copying the color and the type, but I cannot find where.

https://user-images.githubusercontent.com/322610/126053927-124f630b-3f92-4147-b7fd-4ef8c5895c71.mp4

### Your environment
* ROS Distro: Melodic|
* OS Version: e.g. Ubuntu 18.04
* Source build with latest main

### Steps to reproduce
- Add collision objects with color to planning scene
- Pick and place anyone
- Upon picked, color changes to the default attached CO color (purple)
- Upon placed, color changes to the default CO color (green)

### Expected behaviour
Color remains after place

### Actual behaviour
Color reverts to default

corot, 2021-07-19:
For the color, we lose it on [this line](https://github.com/ros-planning/moveit/blob/master/moveit_core/planning_scene/src/planning_scene.cpp#L1810). Commenting it makes the color reappear after placing (though the attached object becomes White, not sure why). Kind of make sense... we remove the obj from the scene, as it becomes attached. But cannot see why we need to remove its color; if the object is not restored, it won't be used, but doesn't bother neither. @v4hn , u wrote these lines. What do u think?

https://user-images.githubusercontent.com/322610/126186196-fdc0c347-bfa6-4250-8ca1-e684a9c60132.mp4

v4hn, 2021-07-20:
> But cannot see why we need to remove its color; if the object is not restored, it won't be used, but doesn't bother neither.

It would block memory, which is bad design and can be a problem when adding/removing objects over longer time frames.
Removing an object is semantically different from attaching it though and the meta information should only be removed in the former case.
Could you provide a patch and/or test for this issue?

felixvd, 2021-07-20:
For the record, I've always thought that a separate `DETACH` keyword would make a lot more sense than using REMOVE for detaching an object from the robot.

v4hn, 2021-07-20:
> For the record, I've always thought that a separate `DETACH` keyword would make a lot more sense than using REMOVE for detaching an object from the robot.

I agree. But I don't think it's a good idea to change/add API because the bug should also be fixed in the current API and it's feasible to check whether objects become/are attached in the respective code. ;)
Feel free to tackle `DETACH` as your next project Felix :sunglasses:

corot, 2021-07-21:
> It would block memory, which is bad design and can be a problem when adding/removing objects over longer time frames. Removing an object is semantically different from attaching it though and the [...]

mmm....  I'm not familiar with the code, but at first glance doesn't look easy.
U mean calling [here](https://github.com/ros-planning/moveit/blob/d7720f4397133d6a09f9c5a33880058bc31fff93/moveit_core/planning_scene/src/planning_scene.cpp#L1560) something like `world_->detachObject(object.object.id)`, and add a DETACH bit [here](https://github.com/ros-planning/moveit/blob/a29a30caaecbd130d85056d959d4eb1c30d4088f/moveit_core/collision_detection/include/moveit/collision_detection/world.h#L207)?  But then,,,  if I got how the observers do their job, the manipulate the planning scene by adding CollisionObjects with ADD, MOVE, REMOVE... and so we need the DETACH action proposed by @felixvd. Or is there any other (non-hackish) way to do it?

felixvd, 2021-07-21:
I didn't look at this too closely until now and didn't remember that only the `planning_scene` is keeping track of the object colors. I was expecting it to be part of each world object and AttachedBody.

That said, you're on the right track, yes. I'm not sure about adding an ActionBit. We could consider it a MOVE action too, but I'm open to either.

corot, 2021-07-26:
I tried to implement yesterday what I mentioned, but didn't managed with my shallow knowledge of the code
However, with [this simple change](#2785) we keep the color and type of attached objects, so it's available on detach

  Sounds like a reasonable solution? At least the changes to the code are minimal