moveit/moveit #2778: Speed scaling factor on python

GitNzino, 2021-07-19:
# Problem and question

I would know if there is a way to change the speed scaling factor on using a some instruction in a Python code in the same way as it is possible in a C++ code.

Thank you for support in advice

GitNzino, 2021-07-19:
Yes of course, but before a pull request i want to know if the functionality that i search is already implemented or not

rhaschke, 2021-07-19:
See here:
https://github.com/ros-planning/moveit/blob/831c638bdbac674ba305a08ac5f75c7a61f5d15e/moveit_commander/src/moveit_commander/move_group.py#L571

https://github.com/ros-planning/moveit/blob/831c638bdbac674ba305a08ac5f75c7a61f5d15e/moveit_commander/src/moveit_commander/move_group.py#L581

closed 2021-07-19 by rhaschke

felixvd, 2021-07-20:
@GitNzino Would you have found this on your own if "speed" was in these functions' descriptions?

GitNzino, 2021-07-20:
I was reading the python example on the [official webpage](http://docs.ros.org/en/melodic/api/moveit_tutorials/html/doc/move_group_python_interface/move_group_python_interface_tutorial.html) but the only example given about the velocity is in C++.

The I googled about the speed scaling factor but there is not so much information about and for that reason I not spend much effort on searching in the function description.
I want to suggest you to put the example of speed scaling factor in the official webpage, it will be more easy for all.

Thank you for patience.

felixvd, 2021-07-20:
The fastest way to add the example is to add it yourself [here](https://github.com/ros-planning/moveit_tutorials). I will make sure to review a PR with the content you mentioned.

GitNzino, 2021-07-20:
Thank you so much!
 You are very efficient