moveit/moveit #2782: The planning scene monitor repeatedly add and remove shapes to exclude to the octomap monitor on robot state update

Hugal31, 2021-07-21:
### Description

On any `SceneUpdateType` except UPDATE_NONE, if there is an octomap_monitor, the `PlanningSceneMonitor::scenePublishingThread` calls `excludeAttachedBodiesFromOctree` and `excludeWorldObjectsFromOctree` ([here](https://github.com/ros-planning/moveit/blob/a27e5d1a1e9f18389d3cb5fe444d39814da6a043/moveit_ros/planning/planning_scene_monitor/src/planning_scene_monitor.cpp#L395)).

This two functions first calls `includeAttachedBodiesInOctree` and [`includeWorldObjectsInOcree`](https://github.com/ros-planning/moveit/blob/a27e5d1a1e9f18389d3cb5fe444d39814da6a043/moveit_ros/planning/planning_scene_monitor/src/planning_scene_monitor.cpp#L787) before re-excluding all the objects. This is rather inefficient, as it cause already excluded shapes to be forget and re-excluded.
While this operation is relatively fast for most shapes, for Mesh shapes it make `point_containment_filter::ShapeMask` do a re-computation of the convex-hull every time. This lead to a big CPU usage increase (from 2-3% to 18% in my case).

### Your environment
* ROS Distro: Noetic
* OS Version: Ubuntu 20.04
* Binary build 1.1.5 from the ROS Debian repository.

### Steps to reproduce
- Create a moveit pipeline with a sensor, for example a PointCloudOctomapUpdater. You do not need to provide the input point cloud to the octomap updater.
- Run it with a basic robot, make sure its state changes regularly (by publishing in /joint_states) and note the CPU usage.
- Add a mesh in the scene (make it complex so the effects are visible, mine have about 10,000 vertices).
- Note the CPU usage.

### Expected behaviour
The CPU usage should not be increase after the initial addition of the mesh.

### Actual behaviour
The CPU usage increase greatly when a mesh is included in the scene, until it is removed.

### Potential solutions

I am not familiar with the MoveIt! source code, but I can see two potential solutions:
- Do not call `exclude...FromOctree` if `new_scene_update_` is `UPDATE_STATE` (but rather if it is `UPDATE_GEOMETRY`).
- Or do not call include...InOctree and rather pick which bodies to exclude based on if they are already excluded.

simonschmeisser, 2021-07-29:
I haven't looked deeply at this but UPDATE_STATE means the robot has moved and maybe the attached object as well? But the world hasn't so it should be correct to just call excludeAttached but not excludeWorld in that case? That sounds like it may be relatively easy to implement?
Please try to open a PR and don't hesitate to ask for guidance where needed :-)

v4hn, 2021-09-03:
@Hugal31 Do you have a patch for this issue locally by now? I agree that it would be nice to simplify the case Simon describes or even cache the shapes when objects only move.