moveit/moveit #2784: Node or topic that shows robot input

GitNzino, 2021-07-22:
#Summary

I want to known if there is any node or topic that shows the input given to the robot in terms of joint position and joint velocity.

#Use case

I am using moveit planner to obtain some advanced feature in robot movement and collision avoiding, but i do not known what is the input that the planner give to the robot. I can also obtain the feedback that in the case of velocity is with some noise.

Having a topic with input velocities and position is better to recognize the type of input that the joint is following

simonschmeisser, 2021-07-26:
You can listen to the goal topic of the http://docs.ros.org/en/api/control_msgs/html/action/FollowJointTrajectory.html action

Please also note that questions like this should best be asked at answers.ros.org  (and please close this if your question has been answered)

GitNzino, 2021-07-27:
Thank you for the response, next time I will surely go to the ros site to ask the question.

I tried to see the topic that you suggested, but i obtain only a sequence of point that in my opinion should be the waypoints of the trajectory. However nothing of them is printable continuatively using rqt_plot or plotjuggler, how should i do?

[image]

Thank you for help, kind regards

felixvd, 2021-07-27:
This package should help: http://wiki.ros.org/rqt_joint_trajectory_plot

GitNzino, 2021-07-27:
Thank you, it works properly!

But sometimes, the planner crash (I'm using Pilz) and put as output this error:
[image]

Can You help me to understand what is not going on?
The robot is in pre-imposted position, but non works.
I tried to launch moveit and rviz with the robot in other position but it continues to not work

simonschmeisser, 2021-09-06:
Are you sure that your robots position is within the joint limits? Those can be set both in URDF and in joint_limits.yaml. The yaml file wins if in doubt.

GitNzino, 2021-09-07:
Yes, it was a similar issue! Thank you!

closed 2021-09-07 by GitNzino