moveit/moveit #2789: Pilz try to lauch from python code and requires an action server that is not present

GitNzino, 2021-07-28:
### Description

Trying to launch a script in python chosen from the example of [pilz page](https://github.com/PilzDE/pilz_industrial_motion/blob/melodic-devel/pilz_robot_programming/examples/demo_program.py) but the demo does not start because not find the server in the title.

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04 LTS
* Source or Binary build? source (through apt)
* If binary, which release version?
* If source, which branch? lastest

### Steps to reproduce

- [x] Start the nodes with the robot driver and moveit with roslaunch;
- [ ]  Start the node that give the input pilz in python.

### Expected behaviour
Move the robot as programmed in the demo

### Actual behaviour
Not reach the action server `sequence_move_group`

### Backtrace or Console output

Output of the command:
`[INFO] [1627478742.506857]: Waiting for connection to action server sequence_move_group...`

Output after terminating the process:
```
^C
[INFO] [1627480409.791315]: Waiting for connection to service /prbt/get_speed_override...
Traceback (most recent call last):
  File "/home/lorenzo/cw_kinova/src/kortex_planner/scripts/pliz_motion_planner_2.py", line 22, in <module>
    r = Robot(__REQUIRED_API_VERSION__)  # instance of the robot
  File "/opt/ros/melodic/lib/python2.7/dist-packages/pilz_robot_programming/robot.py", line 137, in __init__
    self._establish_connections()
  File "/opt/ros/melodic/lib/python2.7/dist-packages/pilz_robot_programming/robot.py", line 542, in _establish_connections
    rospy.wait_for_service(self._GET_SPEED_OVERRIDE_SRV, self._SERVICE_WAIT_TIMEOUT_S)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 146, in wait_for_service
    raise ROSInterruptException("rospy shutdown")
rospy.exceptions.ROSInterruptException: rospy shutdown
```

Thank you for help in advance, have a nice days!
Lorenzo Baglieri

cambel, 2021-07-30:
We use Pilz `sequence_move_group` service without the Pilz python wrappers. The last time I try, it seemed that those wrappers are not fully compatible with MoveIt, so I gave up trying to use it.

In this particular case, your problem is not that the `sequence_move_group` is not reached, in fact, it does connect, the problem is that there are some additional services that are not available, as you can see in this part of the [code](https://github.com/PilzDE/pilz_industrial_motion/blob/2f016c67658c8375dde15f9ba3adc7f561f7b76f/pilz_robot_programming/src/pilz_robot_programming/robot.py#L527-L545). 

I don't know if these services are available in MoveIt, but maybe not.
``` python
    _BRAKE_TEST_EXECUTE_SRV = "/prbt/execute_braketest"
    _GET_SPEED_OVERRIDE_SRV = "/prbt/get_speed_override"
    _BRAKE_TEST_REQUIRED_SRV = "/prbt/brake_test_required"
```

For reference, this is the approach we use:
``` python
group = moveit_commander.MoveGroupCommander(group_name)
sequence_move_group_srv = actionlib.SimpleActionClient("/sequence_move_group", moveit_msgs.msg.MoveGroupSequenceAction)

motion_plan_requests = []
# just in case, set the first point as the current pose
group.set_pose_target(group.get_current_pose(end_effector_link))
# Build request items
msi = moveit_msgs.msg.MotionSequenceItem()
msi.req = group.construct_motion_plan_request()
msi.blend_radius = 0.0

motion_plan_requests.append(msi)

waypoints = [(pose1, 0.01), (pose2, 0.01), (pose3, 0.01)] # Pairs of pose, blend radius (just an example)
for wp, blend_radius in waypoints:
    group.clear_pose_targets() 
    group.set_pose_target(wp)
    msi = moveit_msgs.msg.MotionSequenceItem()
    msi.req = group.construct_motion_plan_request()
    msi.req.start_state = moveit_msgs.msg.RobotState() # only the first point can have a non-empty start state
    msi.blend_radius = blend_radius
    motion_plan_requests.append(msi)

# Force last point to be 0.0 to avoid raising an error in the planner
motion_plan_requests[-1].blend_radius = 0.0

# Make MotionSequence Request
goal = moveit_msgs.msg.MoveGroupSequenceGoal()
goal.request = moveit_msgs.msg.MotionSequenceRequest()
goal.request.items = motion_plan_requests

sequence_move_group_srv.send_goal_and_wait(goal)
response = sequence_move_group_srv.get_result()
    
group.clear_pose_targets()

if response.response.error_code.val == 1: 
    # The response is an array but I think there is always only one plan returned
    plan = response.response.planned_trajectories[0]
    group.execute(plan)
```

GitNzino, 2021-07-31:
Thank you for this answer!

This code reminded me another code that worked for ompl planner with my robot, the code is the following:

```
class MoveGroupPythonIntefaceTutorial(object):

	def __init__(self):

		# Initialize the node
		super(MoveGroupPythonIntefaceTutorial, self).__init__()
		moveit_commander.roscpp_initialize(sys.argv)
		rospy.init_node('constrain_path', anonymous=True)

		try:
			self.is_gripper_present = rospy.get_param(rospy.get_namespace() + "is_gripper_present", False)
			if self.is_gripper_present:
				gripper_joint_names = rospy.get_param(rospy.get_namespace() + "gripper_joint_names", [])
				self.gripper_joint_name = gripper_joint_names[0]
			else:
				gripper_joint_name = ""
		  	self.degrees_of_freedom = rospy.get_param(rospy.get_namespace() + "degrees_of_freedom", 6)

			# Create the MoveItInterface necessary objects
			arm_group_name = "arm"
			self.robot = moveit_commander.RobotCommander("robot_description")
			self.scene = moveit_commander.PlanningSceneInterface(ns=rospy.get_namespace())
			self.arm_group = moveit_commander.MoveGroupCommander(arm_group_name, ns=rospy.get_namespace())
			self.arm_group.set_max_velocity_scaling_factor(1)
			self.arm_group.set_max_acceleration_scaling_factor(1)
			self.arm_group.set_num_planning_attempts(20)
			self.display_trajectory_publisher = rospy.Publisher(rospy.get_namespace() + 'move_group/display_planned_path',
								                    moveit_msgs.msg.DisplayTrajectory,
								                    queue_size=50)

			if self.is_gripper_present:
				gripper_group_name = "gripper"
				self.gripper_group = moveit_commander.MoveGroupCommander(gripper_group_name, ns=rospy.get_namespace())

			rospy.loginfo("Initializing node in namespace " + rospy.get_namespace())
		except Exception as e:
			print (e)
			self.is_init_success = False
		else:
			self.is_init_success = True
```
Do you think that this type of initialization could work also for PILZ planner in addition to the code that you already sent to me?

cambel, 2021-08-01:
Sure, we actually initialize the move_group and services as class' objects like in your example code and it works fine regardless of the planner, we use OMPL for some motions and Pilz planners for others. My example was just a simplify straightforward approach.

GitNzino, 2021-08-04:
Thank you!
I'm trying part of what you say and the initialization works, but now I'm coming back to disturb you because I'm trying to understand how to do a movement link Ptp and Lin with this type of initialization. [This is a simple example of how ptp and lin movement are used by pilz](https://github.com/PilzDE/pilz_tutorials/blob/master/pilz_robots_tutorials/pilz_tutorial_3/pilz_tutorial/scripts/myFirstApplication.py) and [this is a more complex and long example](https://github.com/PilzDE/pilz_industrial_motion/blob/melodic-devel/pilz_robot_programming/examples/demo_program.py).

I tried to write a simple code that 

- [x] print on terminal the position of the robot;
- [ ] move the robot to reach a pose with the Ptp command.

### The code used 

```
#!/usr/bin/env python

import sys
import time
import copy
import rospy
import moveit_commander
import moveit_msgs.msg
import geometry_msgs.msg
import math
from math import pi
from std_srvs.srv import Empty
from std_msgs.msg import String
from moveit_commander.conversions import pose_to_list
from geometry_msgs.msg import Pose, Point
from pilz_robot_programming import *

class MoveGroupPythonIntefaceTutorial(object):

	def __init__(self):

		# Initialize the node
		super(MoveGroupPythonIntefaceTutorial, self).__init__()
		moveit_commander.roscpp_initialize(sys.argv)
		rospy.init_node('pilz_motion_planner', anonymous=True)

		try:
			self.is_gripper_present = rospy.get_param(rospy.get_namespace() + "is_gripper_present", False)
			if self.is_gripper_present:
				gripper_joint_names = rospy.get_param(rospy.get_namespace() + "gripper_joint_names", [])
				self.gripper_joint_name = gripper_joint_names[0]
			else:
				gripper_joint_name = ""
		  	self.degrees_of_freedom = rospy.get_param(rospy.get_namespace() + "degrees_of_freedom", 6)

			# Create the MoveItInterface necessary objects
			arm_group_name = "arm"
			self.robot = moveit_commander.RobotCommander("robot_description")
			self.scene = moveit_commander.PlanningSceneInterface(ns=rospy.get_namespace())
			self.arm_group = moveit_commander.MoveGroupCommander(arm_group_name, ns=rospy.get_namespace())
			self.arm_group.set_max_velocity_scaling_factor(1)
			self.arm_group.set_max_acceleration_scaling_factor(1)
			self.arm_group.set_num_planning_attempts(20)
			self.display_trajectory_publisher = rospy.Publisher(rospy.get_namespace() + 'move_group/display_planned_path',
								                    moveit_msgs.msg.DisplayTrajectory,
								                    queue_size=50)

			if self.is_gripper_present:
				gripper_group_name = "gripper"
				self.gripper_group = moveit_commander.MoveGroupCommander(gripper_group_name, ns=rospy.get_namespace())

			rospy.loginfo("Initializing node in namespace " + rospy.get_namespace())
		except Exception as e:
			print (e)
			self.is_init_success = False
		else:
			self.is_init_success = True
			
	def get_cartesian_pose(self):
[... 48 lines cut ...]
  z: 0.149721693375
orientation: 
  x: -0.646212123986
  y: 0.736560696068
  z: -0.132190557799
  w: 0.149712685666
Traceback (most recent call last):
  File "/home/lorenzo/cw_kinova/src/kortex_planner/scripts/pliz_motion_planner_2.py", line 93, in <module>
    main()
  File "/home/lorenzo/cw_kinova/src/kortex_planner/scripts/pliz_motion_planner_2.py", line 87, in main
    r.arm_group.set_pose_target(Ptp(goal=cartesian_goal, vel_scale=0.9))
  File "/opt/ros/melodic/lib/python2.7/dist-packages/moveit_commander/move_group.py", line 350, in set_pose_target
    ok = self._g.set_pose_target(pose, end_effector_link)
Boost.Python.ArgumentError: Python argument types in
    MoveGroupInterface.set_pose_target(MoveGroupInterface, Ptp, str)
did not match C++ signature:
    set_pose_target(moveit::planning_interface::MoveGroupInterfaceWrapper {lvalue}, boost::python::list {lvalue}, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)
[my_gen3_lite/pliz_motion_planner-1] process has died [pid 20077, exit code 1, cmd bash -c sleep 0; $0 $@ /home/lorenzo/cw_kinova/src/kortex_planner/scripts/pliz_motion_planner_2.py __name:=pliz_motion_planner __log:=/home/lorenzo/.ros/log/e1a0972e-f51b-11eb-9776-024268fc88a0/my_gen3_lite-pliz_motion_planner-1.log].
log file: /home/lorenzo/.ros/log/e1a0972e-f51b-11eb-9776-024268fc88a0/my_gen3_lite-pliz_motion_planner-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```

GitNzino, 2021-08-05:
Today I tried to do a sequence of movement, as suggested from you in this message:

> ```python
> group = moveit_commander.MoveGroupCommander(group_name)
> sequence_move_group_srv = actionlib.SimpleActionClient("/sequence_move_group", moveit_msgs.msg.MoveGroupSequenceAction) [...]

So i produced this code:

```python
class MoveGroupPythonIntefaceTutorial(object):

	def __init__(self):

		# Initialize the node
		super(MoveGroupPythonIntefaceTutorial, self).__init__()
		moveit_commander.roscpp_initialize(sys.argv)
		rospy.init_node('pilz_motion_planner', anonymous=True)

		try:
			self.is_gripper_present = rospy.get_param(rospy.get_namespace() + "is_gripper_present", False)
			if self.is_gripper_present:
				gripper_joint_names = rospy.get_param(rospy.get_namespace() + "gripper_joint_names", [])
				self.gripper_joint_name = gripper_joint_names[0]
			else:
				gripper_joint_name = ""
			self.degrees_of_freedom = rospy.get_param(rospy.get_namespace() + "degrees_of_freedom", 6)

			# Create the MoveItInterface necessary objects
			arm_group_name = "arm"
			self.robot = moveit_commander.RobotCommander("robot_description")
			self.scene = moveit_commander.PlanningSceneInterface(ns=rospy.get_namespace())
			self.arm_group = moveit_commander.MoveGroupCommander(arm_group_name, ns=rospy.get_namespace())
			self.arm_group.set_max_velocity_scaling_factor(__ROBOT_VELOCITY__)
			self.arm_group.set_max_acceleration_scaling_factor(0.4)
			self.arm_group.set_num_planning_attempts(20)
			self.display_trajectory_publisher = rospy.Publisher(rospy.get_namespace() + 'move_group/display_planned_path',
								                    moveit_msgs.msg.DisplayTrajectory,
								                    queue_size=50)
			self.msi.req = self.arm_goup.construct_motion_plan_request()
				

			if self.is_gripper_present:
				gripper_group_name = "gripper"
				self.gripper_group = moveit_commander.MoveGroupCommander(gripper_group_name, ns=rospy.get_namespace())

			rospy.loginfo("Initializing node in namespace " + rospy.get_namespace())
		except Exception as e:
			print (e)
			self.is_init_success = False
		else:
			self.is_init_success = True

def main():
	r = MoveGroupPythonIntefaceTutorial()
	
	sequence_move_group_srv = actionlib.SimpleActionClient("/sequence_move_group", moveit_msgs.msg.MoveGroupSequenceAction) 
	pose1 = Pose(position=Point(0.3, -0.25, 0.45), orientation=from_euler(90*pi/180, 0, 90*pi/180))
	pose2 = Pose(position=Point(0.3, -0.45, 0.45), orientation=from_euler(90*pi/180, 0, 90*pi/180))
	pose3 = Pose(position=Point(0.3, -0.45, 0.25), orientation=from_euler(90*pi/180, 0, 90*pi/180))
	pose4 = Pose(position=Point(0.3, -0.25, 0.25), orientation=from_euler(90*pi/180, 0, 90*pi/180))
	pose5 = Pose(position=Point(0.3, -0.25, 0.45), orientation=from_euler(90*pi/180, 0, 90*pi/180))
	
	
	motion_plan_requests = []
	# just in case, set the first point as the current pose
	r.arm_group.set_pose_target(r.arm_group.get_current_pose())
	
	# Build request items
	msi = moveit_msgs.msg.MotionSequenceItem()
	msi.req = r.arm_group.construct_motion_plan_request()
[... 33 lines cut ...]
```

but I obtained this unexpected output in the terminal:

```
[ INFO] [1628181395.565517245, 2.275000000]: Loading robot model 'gen3_lite_gen3_lite_2f'...
[ INFO] [1628181395.566510856, 2.276000000]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ WARN] [1628181395.820428314, 2.447000000]: Could not identify parent group for end-effector 'end_effector'
[ INFO] [1628181396.928687572, 3.258000000]: Ready to take commands for planning group arm.
'MoveGroupPythonIntefaceTutorial' object has no attribute 'arm_goup'
Traceback (most recent call last):
  File "/home/lorenzo/cw_kinova/src/kortex_planner/scripts/pliz_motion_planner_2.py", line 191, in <module>
    main()
  File "/home/lorenzo/cw_kinova/src/kortex_planner/scripts/pliz_motion_planner_2.py", line 157, in main
    msi.req = r.arm_group.construct_motion_plan_request()
AttributeError: 'MoveGroupCommander' object has no attribute 'construct_motion_plan_request'
[my_gen3_lite/pliz_motion_planner-1] process has died [pid 18151, exit code 1, cmd bash -c sleep 0; $0 $@ /home/lorenzo/cw_kinova/src/kortex_planner/scripts/pliz_motion_planner_2.py __name:=pliz_motion_planner __log:=/home/lorenzo/.ros/log/43586a2e-f60b-11eb-af6e-0242f5280b6d/my_gen3_lite-pliz_motion_planner-1.log].
log file: /home/lorenzo/.ros/log/43586a2e-f60b-11eb-af6e-0242f5280b6d/my_gen3_lite-pliz_motion_planner-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

This error: `AttributeError: 'MoveGroupCommander' object has no attribute 'construct_motion_plan_request'` sounds strange to me because in the [definition](https://docs.ros.org/en/noetic/api/moveit_commander/html/move__group_8py_source.html) of `MoveGroupCommander` there is the `construct_motion_plan_request`

cambel, 2021-08-05:
You said:

> Source or Binary build? source (through apt)

So I assumed you meant Binary, which version do you have? I think the `construct_motion_plan_request` on Python is a very recent addition so you may need to update your binaries `apt update && apt upgrade`. I don't actually know if the binaries have already been updated though. I have not tested it, since we are using MoveIt from the source code. However, in case you are actually using MoveIt from the source code too, you may need to fetch the latest version of the code and compile it.

GitNzino, 2021-08-06:
Yes I have installed all the available package of PILZ and some package of moveit through `sudo apt install` and they are all update and upgrade. On the other hand I not installed all the moveit package and this is the list of not installed moveit package:

```
The following additional packages will be installed:
  ros-melodic-pr2-controllers-msgs ros-melodic-pr2-description
  ros-melodic-pr2-mechanism-msgs ros-melodic-pr2-moveit-config
  ros-melodic-pr2-moveit-plugins
The following NEW packages will be installed:
  ros-melodic-moveit-chomp-optimizer-adapter
  ros-melodic-moveit-chomp-optimizer-adapter-dbgsym
  ros-melodic-moveit-controller-manager-example
  ros-melodic-moveit-controller-manager-example-dbgsym
  ros-melodic-moveit-core-dbgsym ros-melodic-moveit-experimental
  ros-melodic-moveit-fake-controller-manager-dbgsym
  ros-melodic-moveit-kinematics-dbgsym
  ros-melodic-moveit-opw-kinematics-plugin
  ros-melodic-moveit-opw-kinematics-plugin-dbgsym
  ros-melodic-moveit-planners-chomp-dbgsym
  ros-melodic-moveit-planners-ompl-dbgsym ros-melodic-moveit-pr2
  ros-melodic-moveit-python ros-melodic-moveit-resources
  ros-melodic-moveit-resources-fanuc-description
  ros-melodic-moveit-resources-fanuc-moveit-config
  ros-melodic-moveit-resources-pr2-description
  ros-melodic-moveit-resources-prbt-ikfast-manipulator-plugin
  ros-melodic-moveit-resources-prbt-ikfast-manipulator-plugin-dbgsym
  ros-melodic-moveit-resources-prbt-moveit-config
  ros-melodic-moveit-resources-prbt-pg70-support
  ros-melodic-moveit-resources-prbt-support
  ros-melodic-moveit-ros-benchmarks-dbgsym
  ros-melodic-moveit-ros-control-interface-dbgsym
  ros-melodic-moveit-ros-manipulation-dbgsym
  ros-melodic-moveit-ros-move-group-dbgsym
  ros-melodic-moveit-ros-occupancy-map-monitor-dbgsym
  ros-melodic-moveit-ros-perception-dbgsym
  ros-melodic-moveit-ros-planning-dbgsym
  ros-melodic-moveit-ros-planning-interface-dbgsym
  ros-melodic-moveit-ros-robot-interaction-dbgsym
  ros-melodic-moveit-ros-visualization-dbgsym
  ros-melodic-moveit-ros-warehouse-dbgsym ros-melodic-moveit-runtime
  ros-melodic-moveit-servo ros-melodic-moveit-servo-dbgsym
  ros-melodic-moveit-setup-assistant-dbgsym
  ros-melodic-moveit-visual-tools-dbgsym
  ros-melodic-pr2-controllers-msgs ros-melodic-pr2-description
  ros-melodic-pr2-mechanism-msgs ros-melodic-pr2-moveit-config
  ros-melodic-pr2-moveit-plugins

```
There are some package that I have to install to ensure the functionality of the code?

v4hn, 2021-08-06:
I will not look into all the details here, but the first post rather invalidates the issue:

> Trying to launch a script in python chosen from the example of pilz page but the demo does not start because not find the server in the title.

We don't support pilz's upstream scripts and there is nothing we can fix in MoveIt to address this issue.
If you have specific questions on ROS/MoveIt use, please create questions at [answers.ros.org](https://answers.ros.org) instead.

closed 2021-08-06 by v4hn