moveit/moveit #2796: setup_assistant unable to load large URDF files to parameter server

0Nel, 2021-08-03:
### Description

Seeing this line in the code made me think my error is ignorable..
`  // ROS_WARN("Ignore the following error message 'Failed to contact master'. This is a known issue.");` 
[start_screen_widget.cpp - L586](https://github.com/ros-planning/moveit/blob/melodic-devel/moveit_setup_assistant/src/widget/start_screen_widget.cpp#L586)

Sadly this is not the case for me as the setup_assistant fails to load my URDF, stalls and produces the following error message:

```ROS
[ INFO] [1628001031.090026515]: Setting Param Server with Robot Description
[ERROR] [1628001031.090521955]: [setParam] Failed to contact master at [localhost:11311].  Retrying...
```

Interestingly this does not happen all the time and is dependent on the URDF/XACRO file. To start the setup assistant I ran `roslaunch moveit_setup_assistant setup_assistant.launch`. Therefore the fix is not: "start a roscore in advance" ;)

Since my xacro spreads over quite some files it was hard for me to pin point where the problem is and when it occurs. I came to the conclusion that the actual length of the URDF is causing the problem. The critical URDF file lenght or number of lines seems to vary depending on the URDF/XACRO. I suspect that the number of characters is actually more important than the number of lines, which would suggest that maybe the data structure in the code is not sufficient to store the entire file?! For my test case 1000 lines appeared to be critical. What bugs me on this issue is that uploading a textfile from the console using `rosparam set` has no issue whatsoever with files of this size. Which might be another indication that the data structure in the code is not suited. However I do not see how a std::string should have an issue as long as there is still memory available on the system. Despite these oddities I was able to reproduce this issue quite easily and I dearly hope it translates to your setup!

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04, running in docker on a Ubuntu 18.04
* Source or Binary build? both!
* If binary, which release version? Version: 1.0.8-1bionic.20210601.180338
* If source, which branch? Branch: melodic-devel

### Steps to reproduce
Luckily this issue should be easy to reproduce. At least it is on my setup.

First of make a short urdf called test.urdf with the following code. It is basically just a box and loads without any issue in the setup_assistant:

```XML
<?xml version="1.0"?>
  <robot name="test">
    <link name="base">
      <inertial>
        <mass value="10"/>
        <origin rpy="0 0 0" xyz="0 0 0"/>
        <inertia ixx="1.0" ixy="1.0" ixz="1.0" iyy="1.0" iyz="1.0" izz="1.0"/>
      </inertial>
      <visual name="visual_base">
        <origin xyz="0 0 0" rpy="0 0 0"/>
        <geometry>
          <box size="1 1 1"/>
        </geometry>
      </visual>
      <collision name="collision_base">
        <origin xyz="0 0 0" rpy="0 0 0"/>
        <geometry>
          <box size="1 1 1"/>
        </geometry>
      </collision>
    </link>
</robot>
```
Afterwards use bash to append 1000 dummy lines as follows:
`for i in $(seq 1 1000); do echo '' >> test.urdf ; done`

Now the setup_assistant is not able to load the file to the parameter server anymore and stalls on the setParam error message, which I pasted in the introduction.

### Expected behaviour
Load the URDF into the setup_assistant as usual.

### Actual behaviour
The setup assistant is stopping to load the URDF at 20-30 \%, prints the following error and stalls afterwards.

```ROS
[ INFO] [1628001031.090026515]: Setting Param Server with Robot Description
[ERROR] [1628001031.090521955]: [setParam] Failed to contact master at [localhost:11311].  Retrying...
```

### Backtrace or Console output

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

MohmadAyman, 2021-09-23:
Couldn't reproduce the issue using noetic latest.

rhaschke, 2021-09-24:
I couldn't reproduce the issue either. @0Nel can you still reproduce the issue, e.g. after a reboot?
Which version of ROS+MoveIt are you using?

0Nel, 2021-09-24:
Hi, thanks for looking into it. I just tried and I cannot reproduce it anymore. Back then a colleague and I we were both able to reproduce it, but somehow the issue resolved itself after some time. Not sure why, but I suppose the issue can be closed if you are both unable to reproduce it on your side!

I was using ros melodic and was able to reproduce it both with moveit apt packages as well as moveit source build of the melodic-devel branch. Seems odd, but if it works fine again now, I am happy to close the issue.

rhaschke, 2021-09-24:
Hm. Maybe it was a quirk due to some broken state of your base ROS, which was meanwhile fixed? Closing here.

closed 2021-09-24 by rhaschke