moveit/moveit #2797: `MoveGroupInterface.execute()` called from rviz plugin causes rviz to freeze

aa-tom, 2021-08-05:
### Description

I've created a custom rviz plugin which interacts with MoveIt to plan and execute a trajectory. When I click a button to plan the trajectory, the logs indicate a success and the translucent visual of the robot moving through the scene verifies this. When I click the button to execute this path, the logs also indicate a success, however, the rviz window freezes until (I assume) the execution is finished, at which point the window unfreezes and the robot is at the goal pose. The path is planned using the pilz PTP planner.

### Your environment
* ROS Distro: Noetic
* OS Version: e.g. Ubuntu 20.04
* Source or Binary build? Binary
* If binary, which release version? 1.1.5-1focal.20210719.215118

### Steps to reproduce

https://user-images.githubusercontent.com/86778266/128354271-e093e578-d0d4-42ce-839a-f1114b81d150.mp4

### Expected behaviour
I expect rviz to remain active whilst the path is being executed so that the execution can be visualised, similar to how the MoveIt rviz panel works.

### Actual behaviour
rviz freezes until MoveIt finishes executing, preventing the execution from being visualised

### Backtrace or Console output

```
You can start planning now!

[ INFO] [1628167670.762047218] [/fake/rviz]: Setting up rviz_mesh_paint_tool
[ INFO] [1628167670.777344719] [/fake/rviz]: Setting up toolpath_planner_rviz_panel
[ INFO] [1628167670.825587740] [/fake/rviz]: Loading robot model 'yaskawa'...
[ INFO] [1628167670.825640136] [/fake/rviz]: No root/virtual joint specified in SRDF. Assuming fixed joint
[DEBUG] [1628167670.869326298] [/fake/rviz]: Loaded information about the following groups: 'positioner gp8_only gp8_toolchange '
[ INFO] [1628167670.879531228] [/fake/rviz]: Using empty link_prefix.
[ INFO] [1628167670.883273695] [/fake/rviz]: Using empty link_prefix.
[DEBUG] [1628167670.883654396] [/fake/rviz]: Loaded kinematic model in 0.0608571 seconds
[DEBUG] [1628167670.948322592] [/fake/rviz]: Trying to publish message of type [dynamic_reconfigure/ConfigDescription/757ce9d44ba8ddd801bb30bc456f946f] on a publisher with type [dynamic_reconfigure/ConfigDescription/757ce9d44ba8ddd801bb30bc456f946f]
[DEBUG] [1628167670.951799033] [/fake/rviz]: Trying to publish message of type [dynamic_reconfigure/Config/958f16a05573709014982821e6822580] on a publisher with type [dynamic_reconfigure/Config/958f16a05573709014982821e6822580]
[ INFO] [1628167670.983547856] [/fake/rviz]: Starting planning scene monitor
[ INFO] [1628167670.987347271] [/fake/rviz]: Listening to '/fake/move_group/monitored_planning_scene'
[ INFO] [1628167671.292516285] [/fake/rviz]: Constructing new MoveGroup connection for group 'gp8_only' in namespace ''
[DEBUG] [1628167671.303143479] [/fake/rviz]: Waiting for move_group action server (move_group)...
[DEBUG] [1628167671.549432965] [/fake/rviz]: Connected to 'move_group'
[DEBUG] [1628167671.553808226] [/fake/rviz]: Waiting for move_group action server (pickup)...
[DEBUG] [1628167671.808828128] [/fake/rviz]: Connected to 'pickup'
[DEBUG] [1628167671.816081743] [/fake/rviz]: Waiting for move_group action server (place)...
[DEBUG] [1628167672.038291007] [/fake/rviz]: Connected to 'place'
[DEBUG] [1628167672.045103651] [/fake/rviz]: Waiting for move_group action server (execute_trajectory)...
[DEBUG] [1628167672.311433826] [/fake/rviz]: Connected to 'execute_trajectory'
[ INFO] [1628167672.311563507] [/fake/rviz]: Ready to take commands for planning group gp8_only.
[ INFO] [1628167672.311635494] [/fake/rviz]: Looking around: no
[ INFO] [1628167672.311682500] [/fake/rviz]: Replanning: no
[ INFO] [1628167750.115562842] [/fake/toolpath_mesh_server]: Loading part from /home/rob/checkout/halfsphere.stl
[ INFO] [1628167750.123535730] [/fake/toolpath_mesh_server]: 8373 vertices found
[ INFO] [1628167750.123559743] [/fake/toolpath_mesh_server]: 16562 faces found
[ INFO] [1628167750.125009923] [/fake/toolpath_mesh_server]: Built cell locator
[ INFO] [1628167750.133836045] [/fake/toolpath_mesh_server]: Computed 16562 normals
[ INFO] [1628167780.148981099] [/fake/rviz]: Generating path over surface selection of 444 faces using parameters:
        Tooltip radius: 0.010000
        Path resolution: 0.010000
        Raster rot offset: 0.000000
        Min hole size: 0.010000
        Min segment size: 0.010000
        Search radius: 0.020000
[ INFO] [1628167780.151987481] [/fake/toolpath_planner]: Handling toolpath plan request
[ INFO] [1628167780.153044455] [/fake/toolpath_planner]: Loading mesh
[ INFO] [1628167780.153061701] [/fake/toolpath_planner]: Loading mesh from service request
[ INFO] [1628167780.153095303] [/fake/toolpath_planner]: Loading parameters
[ INFO] [1628167780.153143422] [/fake/toolpath_planner]: Generating surface paths
[ INFO] [1628167780.153654578] [/fake/tool_path_planner_server]: Starting path generation
[ INFO] [1628167780.154368433] [/fake/tool_path_planner_server]: Planning path
Warning: tool_path_planner::PlaneSlicerRasterGenerator generating normal data
         at line 400 in /home/rob/checkout/external/src/noether/tool_path_planner/src/plane_slicer_raster_generator.cpp
[... 155 lines cut ...]
[DEBUG] [1628167839.010844203] [/fake/rviz]: Listening to joint states on topic '/fake/joint_states'
[DEBUG] [1628167839.224425754] [/fake/move_group]: The action server has received a new goal request
[DEBUG] [1628167839.224569516] [/fake/move_group]: A new goal has been recieved by the single goal action server
[DEBUG] [1628167839.224868492] [/fake/move_group]: Accepting a new goal
[DEBUG] [1628167839.224995666] [/fake/move_group]: Accepting goal, id: /fake/rviz-1-1628167839.223342994, stamp: 1628167839.22
[ INFO] [1628167839.225092420] [/fake/move_group]: Execution request received
[DEBUG] [1628167839.225425992] [/fake/move_group]: Publishing feedback for goal, id: /fake/rviz-1-1628167839.223342994, stamp: 1628167839.22
[DEBUG] [1628167839.225479059] [/fake/move_group]: Publishing feedback for goal with id: /fake/rviz-1-1628167839.223342994 and stamp: 1628167839.22
[DEBUG] [1628167839.225530654] [/fake/move_group]: Trying to publish message of type [moveit_msgs/ExecuteTrajectoryActionFeedback/12232ef97486c7962f264c105aae2958] on a publisher with type [moveit_msgs/ExecuteTrajectoryActionFeedback/12232ef97486c7962f264c105aae2958]
[DEBUG] [1628167839.225611401] [/fake/move_group]: Validating trajectory with allowed_start_tolerance 0.03
[DEBUG] [1628167839.322722962] [/fake/move_group]: Starting trajectory execution ...
[DEBUG] [1628167839.322842370] [/fake/move_group]: Controller fake_manipulator_controller is already active
[ INFO] [1628167839.323745125] [/fake/move_group]: Fake execution of trajectory
[ INFO] [1628167862.222747307] [/fake/move_group]: Completed trajectory execution with status SUCCEEDED ...
[ INFO] [1628167862.223000422] [/fake/move_group]: Execution completed: SUCCEEDED
[DEBUG] [1628167862.223120057] [/fake/move_group]: Setting the current goal as succeeded
[DEBUG] [1628167862.223186583] [/fake/move_group]: Setting status to succeeded on goal, id: /fake/rviz-1-1628167839.223342994, stamp: 1628167839.22
[DEBUG] [1628167862.223241498] [/fake/move_group]: Publishing result for goal with id: /fake/rviz-1-1628167839.223342994 and stamp: 1628167839.22
[DEBUG] [1628167862.223282814] [/fake/move_group]: Trying to publish message of type [moveit_msgs/ExecuteTrajectoryActionResult/3aa24b8b6dd690aa38a02c1bea6f419f] on a publisher with type [moveit_msgs/ExecuteTrajectoryActionResult/3aa24b8b6dd690aa38a02c1bea6f419f]
[DEBUG] [1628167862.223346796] [/fake/move_group]: Publishing feedback for goal, id: /fake/rviz-1-1628167839.223342994, stamp: 1628167839.22
[DEBUG] [1628167862.223394524] [/fake/move_group]: Publishing feedback for goal with id: /fake/rviz-1-1628167839.223342994 and stamp: 1628167839.22

```

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

aa-tom, 2021-08-05:
Below is the code used in the plugin to execute the trajectory. It's a simple qt callback executed when the button is clicked

```
    void ToolpathPlanningPanel::executeApproachTrajectory()
    {
        std::string move_group = trajectory_page_->trajectory_move_group_textbox_->text().toUtf8().constData();
        moveit::planning_interface::MoveGroupInterface tool_move_group(move_group);
        moveit_msgs::RobotState start_state;
        ros::AsyncSpinner spinner(1);
        spinner.start();
        moveit::core::robotStateToRobotStateMsg(*(tool_move_group.getCurrentState()), start_state);
        moveit::planning_interface::MoveGroupInterface::Plan plan{start_state, approach_trajectory_, 0.0};
        tool_move_group.execute(plan);
    }
```

JafarAbdi, 2021-08-05:
@aa-tom I think it's because you're running the blocking execute function, can you try calling `tool_move_group.asyncExecute(plan)` .?

aa-tom, 2021-08-05:
That did the trick. Thanks a bunch!

closed 2021-08-05 by aa-tom