moveit/moveit #2798: launch file using MoveIt with pilz industrial motion planner not launching

frankymacster, 2021-08-06:
### Description

launch file using MoveIt with pilz industrial motion planner not launching

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 20.04
* Binary build
```
Get:31 http://packages.ros.org/ros/ubuntu bionic/main amd64 ros-melodic-pilz-trajectory-generation amd64 0.4.14-1bionic.20210722.190016 [394 kB]
Get:32 http://packages.ros.org/ros/ubuntu bionic/main amd64 ros-melodic-pilz-industrial-motion amd64 0.4.14-1bionic.20210722.190821 [2312 B]
```
* used to work with Binary build `ros-melodic-pilz-industrial-motion_0.4.12-1bionic.20210601.182337_amd64`

### Steps to reproduce
start a launch file using ur5e with [MoveIt configs](https://github.com/ros-industrial/universal_robot/tree/kinetic-devel/ur5_e_moveit_config) using `pilz_command_planner` as the planner

```
      <node name="joint_state_publisher" pkg="joint_state_publisher" type="joint_state_publisher">
        <param name="/use_gui" value="false"/>
        <rosparam param="/source_list">[/move_group/fake_controller_joint_states]</rosparam>
      </node>

      <node name="robot_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher" respawn="true" output="screen"/>

      <include file="$(eval find('ur5_e_moveit_config') + '/launch/moveit_rviz.launch')"  if="$(eval rviz==true)">
        <arg name="config" value="true"/>
        <arg name="debug" value="$(arg debug)"/>
      </include>
    </group>

    <include file="$(eval find('ur5_e_moveit_config') + '/launch/move_group.launch')">
      <arg name="allow_trajectory_execution" value="true"/>
      <arg name="pipeline" value="pilz_command_planner"/>
      <arg name="fake_execution" value="true"/>
      <arg name="info" value="true"/>
      <arg name="debug" value="$(arg debug)"/>
    </include>
```

### Expected behaviour
launch file using MoveIt with pilz industrial motion planner should launch

### Actual behaviour
launch file using MoveIt with pilz industrial motion planner not launching and gives the Console output below

### Backtrace or Console output
```
[ERROR] [1628185357.193634727]: Client [/motion_player] wants topic /sequence_move_group/result to have datatype/md5sum [moveit_msgs/MoveGroupSequenceActionResult/e995e5b2d6c322a0395df341caa51d8e], but our version has [pilz_msgs/MoveGroupSequenceActionResult/bcd43e7481eddf90e6befe0e8788d0a3]. Dropping connection.
```

felixvd, 2021-08-07:
1) Did you do a clean rebuild of your workspace, and does it include `pilz_msgs`? Try rebuilding that from source as well

2) The planner should be called `pilz_industrial_motion_planner`, not `pilz_command_planner`. Are you sure your versions are matching?

frankymacster, 2021-08-07:
Thank you @felixvd 

> 1. Did you do a clean rebuild of your workspace, and does it include `pilz_msgs`?

yes, when I check `rospack list` I see `pilz_msgs`.
I'm still getting
```
[ERROR] [1628185357.193634727]: Client [/motion_player] wants topic /sequence_move_group/result to have datatype/md5sum [moveit_msgs/MoveGroupSequenceActionResult/e995e5b2d6c322a0395df341caa51d8e], but our version has [pilz_msgs/MoveGroupSequenceActionResult/bcd43e7481eddf90e6befe0e8788d0a3]. Dropping connection.
```

> Try rebuilding that from source as well

I tried rebuilding from source using the branch `melodic-devel`.
I'm still getting
```
[ERROR] [1628185357.193634727]: Client [/motion_player] wants topic /sequence_move_group/result to have datatype/md5sum [moveit_msgs/MoveGroupSequenceActionResult/e995e5b2d6c322a0395df341caa51d8e], but our version has [pilz_msgs/MoveGroupSequenceActionResult/bcd43e7481eddf90e6befe0e8788d0a3]. Dropping connection.
```

> 2. The planner should be called `pilz_industrial_motion_planner`, not `pilz_command_planner`.

I switched the planner name to `pilz_industrial_motion_planner`. 
I'm still getting
```
[ERROR] [1628185357.193634727]: Client [/motion_player] wants topic /sequence_move_group/result to have datatype/md5sum [moveit_msgs/MoveGroupSequenceActionResult/e995e5b2d6c322a0395df341caa51d8e], but our version has [pilz_msgs/MoveGroupSequenceActionResult/bcd43e7481eddf90e6befe0e8788d0a3]. Dropping connection.
```

> Are you sure your versions are matching?

I'm not sure... how would I make sure of that? is there somewhere where I can find this info?

felixvd, 2021-08-08:
Are both moveit and pilz_msgs in your workspace (or whatever package `motion_player` is from), and did you do `catkin clean` or delete your build directory before rebuilding?

> I'm not sure... how would I make sure of that? is there somewhere where I can find this info?

I don't know where `pilz_command_planner` would come from, so I assume that might be old.

frankymacster, 2021-08-08:
@felixvd 

> Are both moveit and pilz_msgs in your workspace (or whatever package motion_player is from)

I'm not building them from source they were installed

> did you do `catkin clean` or delete your build directory before rebuilding?

I did `catkin clean` before rebuilding

> I don't know where pilz_command_planner would come from, so I assume that might be old.

I think `pilz_command_planner` might be old yes since:
[source](http://wiki.ros.org/pilz_robots/Tutorials/MoveRobotWithPilzCommand_planner)
> Note: Please be aware that the Pilz command planner has moved to MoveIt and http://wiki.ros.org/pilz_robots and http://wiki.ros.org/pilz_industrial_motion are unmaintained

frankymacster, 2021-08-09:
I have a hunch that [this change](https://github.com/PilzDE/pilz_industrial_motion/commit/666c7d2bf0c60454439e7c8ce1e01fd11896988b#diff-1d92257ac4b9c507dcf2d98e2e87ca9d7417d78799d26beab2bf7ffb1c51c6f9R33) is what makes this error come up:
```
[ERROR] [1628185357.193634727]: Client [/motion_player] wants topic /sequence_move_group/result to have datatype/md5sum [moveit_msgs/MoveGroupSequenceActionResult/e995e5b2d6c322a0395df341caa51d8e], but our version has [pilz_msgs/MoveGroupSequenceActionResult/bcd43e7481eddf90e6befe0e8788d0a3]. Dropping connection.
```
I don't know which `MoveGroupSequenceActionResult` it should be using now.. `moveit_msgs` or `pilz_msgs` and if `pilz_msgs` is no longer used why `move_group` still uses it in the `/sequence_move_group/result` message.

martiniil, 2021-08-25:
You are using an old version of our planner. Please remove `pilz_industrial_motion` and `pilz_trajectory_generation` and use `pipeline:=pilz_industrial_motion_planner` which should be included via moveit.