moveit/moveit #2800: FloatingJointModel::getVariableRandomPositionsNearBy ignores seed for translation when bounds are infinite

werner291, 2021-08-09:
Hello,

see this method: https://github.com/ros-planning/moveit/blob/779b7c8b019f70898d4de3189f9261c9697d9b9f/moveit_core/robot_model/src/floating_joint_model.cpp#L288

Note how the `seed` is ignored for the translation values when the bounds are infinite, while sampling around a center with a defined distance is a perfectly well-defined case. Hence, the method always returns a joint value with translation of `(0,0,0)` if the bounds are infinite, and the sample may be far from the provided seed point.

I belive this issue was caused by copying from the `FloatingJointModel::getVariableRandomPositions` method, where sampling really is undefined for infinite ranges, so returning `0` is a good fallback.

werner291, 2021-08-09:
Honestly, I think you can simply skip the checks and just sample. There's no special case to handle here.

rhaschke, 2021-08-09:
I agree. Could you please provide a PR? I also suggest reducing code bloat by putting the code for the three position components into a for-loop. The compiler will handle loop expansion based on its optimization settings.

werner291, 2021-08-10:
What's the definition of `angular_distance_weight_`?

I see the angles being scaled by it, but I don't see the translation distance scaled by `(1-angular_distance_weight_)`, is that correct?

I'll assume the current implementation is correct for now while writing the PR.

werner291, 2021-08-10:
Given this snippet:

```
double FloatingJointModel::distance(const double* values1, const double* values2) const
{
  return distanceTranslation(values1, values2) + angular_distance_weight_ * distanceRotation(values1, values2);
}
```

You'd technically be getting a distance that's larger than the sampling radius, but I'm wondering if the user might actually be expecting that.

werner291, 2021-08-10:
Actually, what is meant by `other_bounds`? Why "other", and why isn't that naming consistent in the header/implementation. And why isn't that method static?

I'm getting the impression that the joint itself must have some kind of bounds that aren't being used here.

werner291, 2021-08-10:
What is going on? `variable_bounds_` isn't being used in a large part of the implementation, which leads me to believe that bounds-checking is wrong in most of the implementation. It only uses the "other" bounds, never the joint model's own.

What's the intended usage? Take the intersection of the "other" bounds and the joint model's "own" bounds?

rhaschke, 2021-08-11:
The joint's `variable_bounds_` is passed into `other_bounds` into the generic API functions of the base class, e.g. here:
https://github.com/ros-planning/moveit/blob/2ab14dee3db52f4a838a9595f22e1c5ac311722f/moveit_core/robot_model/include/moveit/robot_model/joint_model.h#L259-L263

That is, if no bounds are specified, the joint's bounds are used.

rhaschke, 2021-08-11:
> Actually, what is meant by `other_bounds`? Why "other", and why isn't that naming consistent in the header/implementation. And why isn't that method static?

Actually, consistency of argument names in header/implementation should be enforced by clang-tidy.
Which specific method you are referring to? Why should it be static?

werner291, 2021-08-12:
Oh, that's interesting... Didn't realize there was one that fed variables into the other.

`getVariableRandomPositionsNearBy(random_numbers::RandomNumberGenerator& rng, double* values,
                                                const Bounds& other_bounds, const double* seed,
                                                const double distance)`

I was arguing that that one should be static since it doesn't use any part of the value it is being called on. But... I kinda just realized it's using the `JointModel`'s vtable to determine which implementation to call, whoops.

Nah, re-reading the header it doesn't seem too ambiguous.

What about the non-spherical position distribution and the weights? Do we want the property that `JointModel::distance(seed,generated) < generated_distance_parameter`? Because we aren't getting it with the current implementation; we'd have to properly generate a point uniformly in a sphere. I believe OMPL has an implementation we can borrow.

rhaschke, 2021-08-15:
I cannot comment on the design decision taken when implementing this. I guess it should be computationally very efficient and accuracy doesn't matter too much here. Thus, the original authors might have chosen the more simple sampling strategy.

werner291, 2021-08-16:
Fair enough, that seems reasonable.

v4hn, 2021-08-28:
@werner291 I guess you are waiting on more feedback about this question before creating the PR?

> What about the non-spherical position distribution and the weights? Do we want the property that `JointModel::distance(seed,generated) < generated_distance_parameter`? Because we aren't getting it [...]

I propose to leave the box-sampling (especially the independence between position and orientation sampling) as is. It's well-defined semantics to have the distance parameter in the method apply to each dimension and the orientation angle separately. If you add more convoluted strategies, the mathematical properties could be nicer, but interpretability might suffer. How much sampling distance do I want to allow to keep the orientation error below a threshold, but sample the pose in roughly this area? With the current implementation it's easy to specify that when you can set `angular_distance_weight_` as well.

If you want to add spherical sampling over box sampling just for the translation dimensions, I would be interested in use cases that demonstrate the performance advantage. In OMPL, sampling nearby poses is mostly relevant for stochastic projection of invalid samples and an accurate L2 distance is probably not at all relevant for that.

werner291, 2021-09-10:
Aha... I'm honestly not sure what I was waiting on. I think other work took over my time for a bit, my apologies.

That seems fair, though, to leave it at box sampling. The user can always throw in some kind of spherical sampler themselves if they want.

Only line I have a question about is this: `if (angular_sample_radius >= .25 * boost::math::constants::pi<double>())`

If I understand that right, the logic here is that, if the sampling angle range is bigger than a certain angle, we simply fall back to sampling over the full range of possible rotations? In that case, shoudlnn't that be `.5 * pi`, and not `.25`?

Other than that, please review for merging?