moveit/moveit #2801: JointConstraintSampler "tolerance" value widens the sampling range, leading to not-quite-unit quaternions

werner291, 2021-08-09:
I have a robot that includes a floating joint, and I'd like to plan from one exact configuration to another exact configuration.

However, the goal samplers appear to produce quaternions that are not quite unit length; they'll instead produce values like `6.97303e-17 -3.27952e-17 -7.03514e-17 1`. I've seen larger values too, such as in the `e-16` range.

This breaks some code further down the line, see this related issue: https://github.com/KavrakiLab/robowflex/issues/239

I believe that this has to do with these lines: https://github.com/ros-planning/moveit/blob/master/moveit_core/constraint_samplers/src/default_constraint_samplers.cpp#L89

How this feeds into the issue: When sampling joint constraints, MoveIt effectively treats every degree of freedom as a separate joint, and adds in a tiny extra "tolerance" value, on the order of `e-16`. This tolerance is actually added to the bounds when sampling.

Hence, if a joint with a range of `[0,0]` (such as a component of a unit quaternion that must be 0), this range effectively become `[-tolerance,tolerance]`.

For example, a JointConstraintSampler that's supposed to produce exactly the quaternion `(0,0,0,1)` will instead produce something like `(6.97303e-17, -3.27952e-17, -7.03514e-17, 1)`, which is does not have unit norm, and is therefore an invalid rotation.

I understand wanting the joint tolerance when looking for goal satisfaction due to rounding errors, but why is it used in the sampling range?

werner291, 2021-08-09:
For reference, those tolerance values come from here https://github.com/ros-planning/moveit/blob/master/moveit_core/kinematic_constraints/include/moveit/kinematic_constraints/utils.h#L100 in my particular case.

v4hn, 2021-08-11:
Hi @werner291 . Thanks for looking into and reporting all the issues you encounter lately. Contributions are more than appreciated :medal_sports: 

> I believe that this has to do with these lines: https://github.com/ros-planning/moveit/blob/master/moveit_core/constraint_samplers/src/default_constraint_samplers.cpp#L89

At first glance I'm not convinced this is the culprit. The tolerances mentioned there [are provided in the planning requests](https://github.com/ros-planning/moveit_msgs/blob/master/msg/JointConstraint.msg#L6-L7) and defined by the user.

Also I'm somewhat surprised because sampling the joints should happen through [JointModel interfaces](https://github.com/ros-planning/moveit/blob/master/moveit_core/robot_model/include/moveit/robot_model/floating_joint_model.h#L52) and the floating/planar joints [explicitly sample valid rotations](https://github.com/ros-planning/moveit/blob/master/moveit_core/robot_model/src/floating_joint_model.cpp#L281).

Instead I would expect that something else was missed in the `Eigen::Affine3d` -> `Eigen::Isometry3d` migration and the code implicitly normalized the offenders before that.

werner291, 2021-08-11:
> At first glance I'm not convinced this is the culprit. The tolerances mentioned there are provided in the planning requests and defined by the user.

Yes, they indeed are. However, I think there is some confusion as to the exact meaning of those tolerances: are they a range of accepted values when validating the constraint, or do they need to be explicitly sampled over that range?

For instance, you could say that the constraint is satisfied, even if you picked up a few rounding errors along the way; the tolerances make complete and total sense that way. However, when sampling, the meaning of "tolerances" are less clear, as you're effectively producing values that don't actually fully satisfy the constraint, and are just a few rounding errors away from violating it.

I think at least part of the problem is that the quaternion is effectively treated as four separate degrees of freedom; the fact that there are only effectively 3 DoF is not taken into account.

> Also I'm somewhat surprised because sampling the joints should happen through JointModel interfaces and the floating/planar joints explicitly sample valid rotations.

Indeed... I'm familiar with that code. That would be a lot better, if that were used. However, I doubt this code is used at all, otherwise #2800 would have been fixed years ago.

I'll happily fix this, since it's at the very core of what I'm working on right now, but could you please elaborate what the purpose of widening the sampling region with those tolerances is exactly?

werner291, 2021-08-11:
> Instead I would expect that something else was missed in the Eigen::Affine3d -> Eigen::Isometry3d migration and the code implicitly normalized the offenders before that.

The default OMPL planning pipeline actually does normalize them. However, it did lead to https://github.com/KavrakiLab/robowflex/issues/239, so users are indeed expecting these to be valid unit quaternions (which is quite reasonable, I'd say).

werner291, 2021-08-11:
From the perspective of me, as a user: If I give a constraint of, say, `x = 100.0` with tolerance `0.1`, I'd expect a sampler to always give me exactly `100.0`, but I'd expect a "Is a value of 100.05 valid?" to be true.

v4hn, 2021-08-12:
> Indeed... I'm familiar with that code. That would be a lot better, if that were used. However, I doubt this code is used at all, otherwise #2800 would have been fixed years ago.
>
> I'll happily fix this [...]

Please have a look at it. floating joints were always treated rather badly in Moveit and I'm not surprised there are issues there.

> From the perspective of me, as a user: If I give a constraint of, say, x = 100.0 with tolerance 0.1, I'd expect a sampler to always give me exactly 100.0, but I'd expect a "Is a value of 100.05 valid?" to be true.

 that's not the regular interpretation in Moveit where constraints always specify *regions* and not just target states with soft bounds. if you want support for your interpretation we would need to change the interfaces somewhere to allow for it.

werner291, 2021-08-12:
> that's not the regular interpretation in Moveit where constraints always specify regions and not just target states with soft bounds.

No that's fine, if that's the interpretation; maybe an additional comment would be nice, though; I'll add it. Then Robowflex is clearly using it wrong, I'll go inform the maintainer.

Yes, floating joints are finicky, I've definitely noticed that... Hopefully less so after I'm done with them, since I'm doing a project with them right now. Let's leave this issue open for now.

werner291, 2021-08-12:
How do we interpret the concept of an "upper" and a "lower" bound on an orientation constraint?

werner291, 2021-08-12:
```
    const moveit::core::VariableBounds& joint_bounds = jm->getVariableBounds(joint_constraint.getJointVariableName());
    JointInfo ji;
    std::map<std::string, JointInfo>::iterator it = bound_data.find(joint_constraint.getJointVariableName());
    if (it != bound_data.end())
      ji = it->second;
    else
      ji.index_ = jmg_->getVariableGroupIndex(joint_constraint.getJointVariableName());
    ji.potentiallyAdjustMinMaxBounds(
        std::max(joint_bounds.min_position_,
                 joint_constraint.getDesiredJointPosition() - joint_constraint.getJointToleranceBelow()),
        std::min(joint_bounds.max_position_,
                 joint_constraint.getDesiredJointPosition() + joint_constraint.getJointToleranceAbove()));
```

So... This is effectively taking the intersection of a number of constraint regions; this completely depends on the idea that degrees of freedom are completely independent, which they are most certainly not in our case.

I'll have to figure out how to sample the intersection of:
- Orientation constraints defined by Euler angle tolerances
- The same, but with angle-axis tolerances. Those might be equivalent...

werner291, 2021-08-12:
(Wrong button, oops)

werner291, 2021-08-12:
Ok, nothing makes sense... I think my interpretation of what's going on is very wrong. The JointConstraintSampler should not even be invoked for orientation constraints...

werner291, 2021-08-12:
Ok, I think the problem might be here, in this method that appears to assume every constraint is a 1-DoF joint constraint:
https://github.com/ros-planning/moveit/blob/master/moveit_core/kinematic_constraints/src/utils.cpp#L130

werner291, 2021-08-12:
Ok, while it's definitely the case that the current implementation returns non-unit quaternions, apparently adding those tolerance regions to the constraint sampler is intentional for 1-DoF joints.

@zkingston Perhaps you should consider whether that's what you expected.