moveit/moveit #2804: moveit_ros_planning fails to build with DCMAKE_BUILD_TYPE=Debug

werner291, 2021-08-10:
### Description

When building a ROS workspace with `catkin build --cmake-args -DCMAKE_BUILD_TYPE=Debug`, the build fails with a linker error on `moveit_ros_planning`.

I was hoping to debug MoveIt more easily, since I'll probably be making some modifications to it for a project I'm working on.

### Your environment
* ROS Distro: Noetic
* OS Version: Ubuntu 20.04 LTS

Fresh catkin workspace, with `moveit`, `moveit_visual_tools` and `moveit_resources` cloned, all from default branch (master, probably).

### Steps to reproduce

* Create a directory containing an src subdir
* Clone the three aforementioned repositories into it.
* Run `catkin build --cmake-args -DCMAKE_BUILD_TYPE=Debug`

### Expected behaviour

Buildl compiletes successfully, resulting in MoveIt with debug symbols.

### Actual behaviour

`moveit_ros_planning` compiles, but fails linking with the following error:

```
Errors     << moveit_ros_planning:make /home/werner/catkin_test/logs/moveit_ros_planning/build.make.000.log                                                                                                                                                                   
/usr/bin/ld: CMakeFiles/moveit_compare_collision_checking_speed_fcl_bullet.dir/src/compare_collision_speed_checking_fcl_bullet.cpp.o: undefined reference to symbol 'ccd_vec3_origin'
/usr/bin/ld: /lib/x86_64-linux-gnu/libccd.so.2: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [planning_components_tools/CMakeFiles/moveit_compare_collision_checking_speed_fcl_bullet.dir/build.make:247: /home/werner/catkin_test/devel/.private/moveit_ros_planning/lib/moveit_ros_planning/moveit_compare_collision_checking_speed_fcl_bullet] Error 1
make[1]: *** [CMakeFiles/Makefile2:3192: planning_components_tools/CMakeFiles/moveit_compare_collision_checking_speed_fcl_bullet.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:141: all] Error 2
cd /home/werner/catkin_test/build/moveit_ros_planning; catkin build --get-env moveit_ros_planning | catkin env -si  /usr/bin/make --jobserver-auth=3,4; cd -
```

### Backtrace or Console output

```
-------------------------------------------------------------
Profile:                     default
Extending:             [env] /opt/ros/noetic
Workspace:                   /home/werner/catkin_test
-------------------------------------------------------------
Build Space:        [exists] /home/werner/catkin_test/build
Devel Space:        [exists] /home/werner/catkin_test/devel
Install Space:      [unused] /home/werner/catkin_test/install
Log Space:         [missing] /home/werner/catkin_test/logs
Source Space:       [exists] /home/werner/catkin_test/src
DESTDIR:            [unused] None
-------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
-------------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Debug
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-------------------------------------------------------------
Workspace configuration appears valid.

NOTE: Forcing CMake to run for each package.
-------------------------------------------------------------
[build] Found '40' packages in 0.0 seconds.                                                                                                                                                                                                                                   
[... 55 lines cut ...]
Abandoned <<< moveit_kinematics                                              [ Unrelated job failed ]                                                                                                                                                                         
Abandoned <<< moveit_planners_ompl                                           [ Unrelated job failed ]                                                                                                                                                                         
Abandoned <<< moveit_ros_move_group                                          [ Unrelated job failed ]                                                                                                                                                                         
Abandoned <<< moveit_ros_manipulation                                        [ Unrelated job failed ]                                                                                                                                                                         
Abandoned <<< moveit_ros_perception                                          [ Unrelated job failed ]                                                                                                                                                                         
Abandoned <<< moveit_ros_robot_interaction                                   [ Unrelated job failed ]                                                                                                                                                                         
Abandoned <<< moveit_ros_warehouse                                           [ Unrelated job failed ]                                                                                                                                                                         
Abandoned <<< moveit_ros_benchmarks                                          [ Unrelated job failed ]                                                                                                                                                                         
Abandoned <<< moveit_ros_planning_interface                                  [ Unrelated job failed ]                                                                                                                                                                         
Abandoned <<< moveit_planners_chomp                                          [ Unrelated job failed ]                                                                                                                                                                         
Abandoned <<< moveit_ros_visualization                                       [ Unrelated job failed ]                                                                                                                                                                         
Abandoned <<< moveit_runtime                                                 [ Unrelated job failed ]                                                                                                                                                                         
Abandoned <<< moveit_servo                                                   [ Unrelated job failed ]                                                                                                                                                                         
Abandoned <<< moveit_setup_assistant                                         [ Unrelated job failed ]                                                                                                                                                                         
Abandoned <<< moveit_visual_tools                                            [ Unrelated job failed ]                                                                                                                                                                         
Abandoned <<< pilz_industrial_motion_planner_testutils                       [ Unrelated job failed ]                                                                                                                                                                         
Abandoned <<< pilz_industrial_motion_planner                                 [ Unrelated job failed ]                                                                                                                                                                         
[... cut ...]
[build]   Failed:    1 packages failed.                                                                                                                                                                                                                                       
[build] Runtime: 5 minutes and 2.9 seconds total.                                                                                                                                                                                                                             
[build] Note: Workspace packages have changed, please re-source setup files to use them.
```

werner291, 2021-08-10:
Looks like this: https://github.com/ros-planning/moveit/blob/master/moveit_ros/planning/planning_components_tools/src/compare_collision_speed_checking_fcl_bullet.cpp depends on something called `libccd`, which looks like some optional component of `Bullet` that isn't linked by default.

v4hn, 2021-08-11:
Interesting. I have never had that issue and I develop in DEBUG builds, but I don't run on stock ubuntu. I can reproduce this in a docker container on Ubuntu 20.04.
It seems the libraries detected [here](https://github.com/ros-planning/moveit/blob/master/moveit_core/CMakeLists.txt#L47) do not include libccd, which is a build-export dependency of `moveit_core/collision_detection_fcl`.

I can't look into this any further now, but I'm wondering why the moveit_core package compiles successfully then.

mintar, 2021-08-17:
I've dug a bit deeper into this issue.

I can confirm that this only happens when using `Debug` or `RelWithDebugInfo` builds, not `Release`. I've used `git bisect` to track down the commit that triggered this bug: 93c3f63 by @wolfv.

Specifically, that commit introduced the following lines:

https://github.com/ros-planning/moveit/blob/93c3f637b088000e5508f8efc6ce3b01e528e383/moveit_core/CMakeLists.txt#L45-L49

So after 93c3f63, we first try to use `find_package` to find `fcl` and only fall back to PkgConfig if that fails. The old behavior was to always use PkgConfig.

However, there is a bug in `fcl`. The `find_package` implementation is missing the dependencies:

```text
find_package: FCL_LIBRARIES=fcl
PkgConfig:    FCL_LIBRARIES=/opt/ros/noetic/lib/x86_64-linux-gnu/libfcl.so;/usr/lib/x86_64-linux-gnu/libccd.so;/usr/lib/x86_64-linux-gnu/libm.so;/opt/ros/noetic/lib/liboctomap.so;/opt/ros/noetic/lib/liboctomath.so
```

I'll open a pull request over at the fcl repo and report back here.

PR #2819 (2021-08-17): [moveit_core] Fix export of FCL dependency
Regression of 93c3f63
Closes: #2804
commit 4d6f805b (2021-08-17): [moveit_core] Fix export of FCL dependency (#2819)

Regression of 93c3f63
Closes: #2804
closed 2021-08-17 by v4hn