moveit/moveit #2805: Robot display frame rate is too slow

felixvd, 2021-08-10:
I've been wondering for a long time why the frame rate in Rviz is so bad. My "real" robot's visualization updates only at about 4 frames per second, which is very choppy and would look a lot better if it was smooth. When my robots move, the TF frames do update quickly (so the `joint_state_publisher` is not the issue), but both the PlanningScene display and MotionPlanning plugin lag behind. What is the bottleneck?

There are multiple threads on ROS answers about this:

- [This thread](https://answers.ros.org/question/333229/how-to-improve-the-robot-frame-rate-when-using-motion-planning-in-rviz/)  mentions the SceneDisplayTime parameter, but that does not seem to change anything.
- In [this one](https://answers.ros.org/question/307723/how-to-make-robot-movement-in-rviz-with-moveit-smoother/) I suggested to increase the `joint_state_publisher` publishing rate (that doesn't solve it), and @v4hn suggested using different controllers. But the issue also occurs with real robots, so MoveIt's fake controllers are not the main cause.

I looked at the code in`planning_scene_display.cpp` and noticed a few things:

1) `robot_state_needs_render_` is never set to true anywhere. @simonschmeisser It looks like you touched this code last year. Do you have any ideas?
2) Adding debug prints [here](https://github.com/ros-planning/moveit/blob/master/moveit_ros/visualization/planning_scene_rviz_plugin/src/planning_scene_display.cpp#L580) and [here](https://github.com/ros-planning/moveit/blob/master/moveit_ros/planning/planning_scene_monitor/src/planning_scene_monitor.cpp#L465) (like `ROS_INFO_STREAM("scene update: " << update_type);`), it looks like the [callback](https://github.com/ros-planning/moveit/blob/master/moveit_ros/visualization/planning_scene_rviz_plugin/src/planning_scene_display.cpp#L540) registered to the display's `planning_scene_monitor_` does not trigger correctly when there is an `UPDATE_STATE` event. The callback in the display only fires when the update_type is `UPDATE_SCENE`, but not `UPDATE_STATE`. If I register the callback in L540 twice, the callback inside the display is reached twice when the update type is `UPDATE_SCENE`, and still not at all for `UPDATE_STATE`.  
3) I tried printing the joint values of [this planning scene](https://github.com/ros-planning/moveit/blob/master/moveit_ros/visualization/planning_scene_rviz_plugin/src/planning_scene_display.cpp#L345). It seems that the `planning_scene_monitor_` in the display does not update its robot state when it receives a joint state. The joints are only updated when the `UPDATE_SCENE` event occurs. 

My open questions about this:

1) What is causing the `UPDATE_SCENE` event at about 4 Hz?
2) Why is the PSM not updating the robot state?
3) Why is [this](https://github.com/ros-planning/moveit/blob/master/moveit_ros/planning/planning_scene_monitor/src/planning_scene_monitor.cpp#L465) `update_callback` call not reaching [this](https://github.com/ros-planning/moveit/blob/master/moveit_ros/visualization/planning_scene_rviz_plugin/src/planning_scene_display.cpp#L580) `sceneMonitorReceivedUpdate` for `UPDATE_STATE` updates?

I'm still not extremely familiar with the `PlanningSceneMonitor` and Rviz plugin, so I'd appreciate ideas about what might be happening. Pinging the usual suspects @rhaschke @v4hn @simonschmeisser

felixvd, 2021-08-10:
I think 2 and 3 are fixed by adding `planning_scene_monitor_->startStateMonitor();` [here](https://github.com/ros-planning/moveit/blob/master/moveit_ros/visualization/planning_scene_rviz_plugin/src/planning_scene_display.cpp#L539).

Before:

https://user-images.githubusercontent.com/4535737/128925093-5b1bfb32-7c4a-473f-a2c0-88f11a569f3f.mp4

After:

https://user-images.githubusercontent.com/4535737/128925102-277639c1-a86a-4d8c-ba82-92f7f74ff98f.mp4

This obviously still has potential for optimization, but the difference should be clear.

simonschmeisser, 2021-08-10:
You should be able to increase the frame rate via dynamic reconfigure actually. There is (afaik) only one of the many rate limits still limiting in the default configuration. See #2791 for a current discussion about increasing the default and #2049 (the alternative title: "MoveIt like it's 2020!" unfortunately never worked out ...)

So the general idea of PlanningSceneDisplay (and derived classes) is to remotely show a consistent state of another PlanningScene(Monitor). There are some input rate limits (eg joint position) and a output limit. You can change the output limit via dynamic reconfigure/rosparam. This then gets packed in a PlanningSceneMsg (can be a diff) and transferred to the PlanningSceneDisplay. The Display has further rate limits but they should effectively be disabled (as you observed). The Display has another PlanningSceneMonitor that is used for collision checks in the MotionPlanningDisplay but not much otherwise. (To reduce memory consumption I set the collision checker of those Displays to AllValid - the dummy).

Starting the StateMonitor in the Display might mean that you get some visual glitches if e.g. some obstacle is not yet removed but the robot already moves through it.

I think it will also send a full update instead of a diff from time to time and that might cause the `UPDATE_SCENE` but maybe that is also just because the "remote" PlanningSceneMonitor thing is not implemented that well and turns a diff with only a state update into a scene update. 4hz is the default value btw

felixvd, 2021-08-11:
Thanks, I remember that PR.

> You should be able to increase the frame rate via dynamic reconfigure actually.

Using which parameter? Either of the ones mentioned in the OP did not work on their own.

(Quoting from Discord:)

> robot_state_needs_render is indeed superfluous. Note that render here means "reupload to Ogre scene" and when just some transformation changed we should not do that

I think `robot_state_needs_render_` actually blocks the scene from updating with the SceneDisplayTime parameter [here](https://github.com/ros-planning/moveit/blob/master/moveit_ros/visualization/planning_scene_rviz_plugin/src/planning_scene_display.cpp#L651).

> Starting the StateMonitor in the Display might mean that you get some visual glitches if e.g. some obstacle is not yet removed but the robot already moves through it.

If I don't start it, the robot state passed to [updateRobotPosition](https://github.com/ros-planning/moveit/blob/master/moveit_ros/visualization/planning_scene_rviz_plugin/src/planning_scene_display.cpp#L355) is not changed, and the joint values don't update.

simonschmeisser, 2021-08-11:
[image]

changing the `publish_planning_scene_hz` from 4 to 30 makes it all go butter smooth here

I'll try to recapitulate why and how that works exactly in PlanningSceneDisplay

felixvd, 2021-08-11:
The PlanningScene display itself works fine at 50 Hz in my dual-arm system (and it looks great), but the MotionPlanning plugin struggles even at 10 Hz. There must be a bottleneck somewhere, possibly because something locks the PlanningScene.

simonschmeisser, 2021-08-13:
Maybe check if enabling/disabling the start goal marker makes a difference? Anything else that could perform collision checks regularily?