moveit/moveit #2811: ompl_interface::ConstrainedGoalSampler::sampleUsingConstraintSampler doesn't appear to sample?

werner291, 2021-08-13:
Hello,

I was working on trying to implement a custom `constraint_samplers::ConstraintSampler`, but I noticed that only the `project` method gets called, never the `sample` method.

The state I get as input always appears to be the same (I think it's the same as the starting state?) when I print it out.

When reading through https://github.com/ros-planning/moveit/blob/master/moveit_planners/ompl/ompl_interface/src/detail/constrained_goal_sampler.cpp#L90, I also don't really see any source of randomness in there... it appears to just be projecting the same state over and over.

Is this the intended behavior?

werner291, 2021-08-13:
Looks like `sample` can be swapped into the place where `project` is used with no further changes to the code; it even takes the same arguments. Maybe that was what was supposed to happen there?

werner291, 2021-08-13:
Made the swap my own fork of the code... so far seems to work fine?

v4hn, 2021-08-28:
> Maybe that was what was supposed to happen there?

`project` was apparently just treated as another interface for sampling here where the first parameter serves as input and output.
From what I can see, the MoveIt code base only uses `project()` at the moment and if you change it to `sample`, then `project` will be unused. So they are basically redundant interfaces in case some plugin *and* user code wants to differentiate. Clearly no nice design.

I'm open to fix the semantics here. If you want to change it to
```
constraint_sampler_->sample(work_state_, work_state_, planning_context_->getMaximumStateSamplingAttempts()))
```
please file a pull-request.

werner291, 2021-09-10:
Well, the semantics seem quite different, don't they?

The way the documentation is written, "sample" generates a completely new robot state, writing it into the passed-in data-structure, whereas "project" actually takes the passed-in structure, and applies an operation to it that makes it compliant with the constraints.

Either way, see the new pull request: https://github.com/ros-planning/moveit/pull/2872

werner291, 2021-09-13:
Oooh, that looks fancy! I'll check it out; I've never really been a huge fan of the "carthesian planning + IK" approach, since IK is kinda meaningless for my particular robot (base is very mobile) without directly tying it to the planning process.

At the moment though, I'm just using a constrained sampler, which seems to work just fine for my purpose.

I still believe my PR (#2872) would be useful, since in my particular sampler, projecting and sampling really are two distinct actions; one relies on the other, actually. (It produces samples uniformly, then projects them to obtain constrained ones.)

 I *can* put all of my code in the `project`, method, but that feels kinda wrong, and I'd prefer not to depend on what appears to me to be a bug in MoveIt.

commit a5607832 (2021-09-14): Fix #2811 (#2872)

This is a PR for #2811
closed 2021-09-14 by v4hn