moveit/moveit #2812: Defining pipelines names parameter causes CHOMP pipelines to be defined

pvanlaar, 2021-08-13:
### Description

During the switch from MoveItCpp to Move Group Interface, I forgot to remove a config file, which was still loaded on startup. In this file the pipeline names were still loaded:

~~~yaml
planning_pipelines:
  pipeline_names:
    - ompl
~~~

This caused two pipelines to be loaded with CHOMP planners. This can be verified by calling the query service (`rosservice call /query_planner_interface`). This results in the following output:
~~~
planner_interfaces: 
  - 
    name: "CHOMP"
    pipeline_id: "ompl"
    planner_ids: 
      - CHOMP
  - 
    name: "CHOMP"
    pipeline_id: "pipeline_names"
    planner_ids: 
      - CHOMP
~~~

In this case, there were no planning pipelines defined after loading the config file, but if one is defined with the same name (for example "ompl"), this pipeline can be overwritten. This was weird behaviour because I only used OMPL and have not referenced CHOMP planners anywhere.

### Environment
* ROS Distro: Noetic
* OS Version: Ubuntu 20.04
* Source: detached at tag 1.1.5

### Steps to reproduce
In this [repo](https://github.com/pvanlaar/moveit_pipeline_config_issue) a minimum example can be found.
Required are the following packages (defined in the example package):
- moveit 1.1.5
- panda_moveit_config 

1. Launch the move_group node including the `planning_pipelines` configuration
2. Call the `/query_planner_interface` service to verify

### Expected behaviour
No pipelines should be defined and probably a warning should be reported. 

### Actual behaviour
One or more CHOMP planners are defined. These planners are not the default planner of MoveIt (OMPL's RTTConnect as far as I know) and do not support pose targets.

fi6, 2022-01-25:
Got the same problem, calling service responds as follow:
```
rosservice call /my_gen3/query_planner_interface                                                           yida@yida-bigai-pc  15:39:12
planner_interfaces: 
  - 
    name: "CHOMP"
    pipeline_id: "arm"
    planner_ids: 
      - CHOMP
  - 
    name: "CHOMP"
    pipeline_id: "capabilities"
    planner_ids: 
      - CHOMP
  - 
    name: "CHOMP"
    pipeline_id: "default_planner_config"
    planner_ids: 
      - CHOMP
  - 
    name: "CHOMP"
    pipeline_id: "gripper"
    planner_ids: 
      - CHOMP
  - 
    name: "CHOMP"
    pipeline_id: "joint_pos_term_info"
    planner_ids: 
      - CHOMP
  - 
    name: "CHOMP"
    pipeline_id: "planner_configs"
    planner_ids: 
      - CHOMP
  - 
    name: "CHOMP"
    pipeline_id: "planning_plugin"
    planner_ids: 
      - CHOMP
  - 
    name: "CHOMP"
    pipeline_id: "problem_info"
    planner_ids: 
      - CHOMP
  - 
    name: "CHOMP"
    pipeline_id: "request_adapters"
    planner_ids: 
      - CHOMP
  - 
    name: "CHOMP"
    pipeline_id: "start_state_max_bounds_error"
    planner_ids: 
      - CHOMP
  - 
    name: "CHOMP"
    pipeline_id: "trajopt_param"
    planner_ids: 
      - CHOMP
```

rhaschke, 2022-01-25:
Sorry that nobody responded here yet. Could you please provide the exact MoveIt version as well the robot config repository you are using? I think this is a configuration problem.

fi6, 2022-01-25:
Hi @rhaschke ,

I'm currently using moveit built from commit 7da00c19 (branch: master), the robot config repo is from https://github.com/zyjiao4728/ros_kortex (please look at the latest commit). Whatever the default planner is, there will be CHOMP in all planner_ids.

rhaschke, 2022-01-25:
@fi6, as expected, you screwed your configuration. Please find fixes in https://github.com/zyjiao4728/ros_kortex/pull/1.
The key issue was a [missing namespace definition in planning_pipeline.launch.xml](https://github.com/zyjiao4728/ros_kortex/pull/1/files#diff-75e31216a9d196bc80f8d6c17b1a30279c876aed800a1cbff34383950a58b6d0L9).

closed 2022-01-25 by rhaschke

fi6, 2022-01-26:
@rhaschke I've found out the key steps to fix the problem from your PR. Adding namespace to planning_pipeline.launch will solve the problem perfectly, and just remember to remove the `Planning Functionality` lines in `move_group.py`(which I forgot to push). Thanks a lot for the fix.